espressif/arduino-esp32 #9595: Problem with HTTPClient https HTTPC_ERROR_CONNECTION_REFUSED

mickahell, 2024-05-06:
### Board

ESP32 Watchy

### Device Description

ESP32 Watchy from SQFMI : https://watchy.sqfmi.com/docs/hardware

### Hardware Configuration

na

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

macOS 11.7

### Flash frequency

na

### PSRAM enabled

no

### Upload speed

921600

### Description

I want to trigger a GitHub actions dispatch workflow from my esp32 using HTTPClient.

It's working perfectly using curl with :

```bash
curl -L -X POST https://api.github.com/repos/mickahell/robots-data/dispatches \
    -H "Accept: application/vnd.github+json" -H "Authorization: token ${TOKEN}" \
    -d '{
        "event_type": "watchy-data",
        "client_payload":
        {
            "data-name":"steps", "date":"01-01-1970", "hour":"00", "data":"7000"
        }
    }'
```

But I'm always getting an error `-1` connection refused from the esp32.
 

### Sketch

```cpp
#include "sendData.h"

#define API_URL "https://api.github.com/repos/mickahell/robots-data/dispatches"
#define API_TOKEN "toto"

const char* test_root_ca = "-----BEGIN CERTIFICATE-----\n" \
                           "MIICjzCCAhWgAwIBAgIQXIuZxVqUxdJxVt7NiYDMJjAKBggqhkjOPQQDAzCBiDEL\n" \
                           "MAkGA1UEBhMCVVMxEzARBgNVBAgTCk5ldyBKZXJzZXkxFDASBgNVBAcTC0plcnNl\n" \
                           "eSBDaXR5MR4wHAYDVQQKExVUaGUgVVNFUlRSVVNUIE5ldHdvcmsxLjAsBgNVBAMT\n" \
                           "JVVTRVJUcnVzdCBFQ0MgQ2VydGlmaWNhdGlvbiBBdXRob3JpdHkwHhcNMTAwMjAx\n" \
                           "MDAwMDAwWhcNMzgwMTE4MjM1OTU5WjCBiDELMAkGA1UEBhMCVVMxEzARBgNVBAgT\n" \
                           "Ck5ldyBKZXJzZXkxFDASBgNVBAcTC0plcnNleSBDaXR5MR4wHAYDVQQKExVUaGUg\n" \
                           "VVNFUlRSVVNUIE5ldHdvcmsxLjAsBgNVBAMTJVVTRVJUcnVzdCBFQ0MgQ2VydGlm\n" \
                           "aWNhdGlvbiBBdXRob3JpdHkwdjAQBgcqhkjOPQIBBgUrgQQAIgNiAAQarFRaqflo\n" \
                           "I+d61SRvU8Za2EurxtW20eZzca7dnNYMYf3boIkDuAUU7FfO7l0/4iGzzvfUinng\n" \
                           "o4N+LZfQYcTxmdwlkWOrfzCjtHDix6EznPO/LlxTsV+zfTJ/ijTjeXmjQjBAMB0G\n" \
                           "A1UdDgQWBBQ64QmG1M8ZwpZ2dEl23OA1xmNjmjAOBgNVHQ8BAf8EBAMCAQYwDwYD\n" \
                           "VR0TAQH/BAUwAwEB/zAKBggqhkjOPQQDAwNoADBlAjA2Z6EWCNzklwBBHU6+4WMB\n" \
                           "zzuqQhFkoJ2UOQIReVx7Hfpkue4WQrO/isIJxOzksU0CMQDpKmFHjFJKS04YcPbW\n" \
                           "RNZu9YO6bVi9JNlWSOrvxKJGgYhqOkbRqZtNyWHa0V1Xahg=\n" \
                           "-----END CERTIFICATE-----\n";

void SendData::pushAPIData(String json) {

  HTTPClient https;
  Serial.println(API_URL);
  Serial.println(json);

  https.setConnectTimeout(10000); // 3 second max timeout
  https.begin(API_URL, test_root_ca);

  https.addHeader("Accept",  "application/vnd.github+json");
  https.addHeader("Content-Type", "application/json");
  https.addHeader("Authorization", "token " + String(API_TOKEN));
  int httpsResponseCode = https.POST(json);

  if (httpsResponseCode != 200) {
    Serial.println("- not send to Github");
    Serial.println(httpsResponseCode);
    String response = https.getString();
    Serial.println(response);
    Serial.printf("[HTTPS] GET... failed, error: %s\n", https.errorToString(httpsResponseCode).c_str());

  } else {
    Serial.println("- success send to Github");
    Serial.println(httpsResponseCode);
    String response = https.getString();
    Serial.println(response);
  }
  https.end();
}

```cpp
String day_api = (currentTime.Day < 10) ? ("0" + String(currentTime.Day)) : String(currentTime.Day);
String month_api = (currentTime.Month < 10) ? ("0" + String(currentTime.Month)) : String(currentTime.Month);
String date_api = day_api + "-" + month_api + "-" + String(currentTime.Year + 1970);

String hour_api = (currentTime.Hour < 10) ? ("0" + String(currentTime.Hour)) : String(currentTime.Hour);
String json_steps = "{\"event_type\":\"" + String(ENDPOINT_API) + "\",\"client_payload\":{\"data-name\":\"" + String(ENDPOINT_STEPS) + "\",\"date\":\"" + date_api + "\",\"hour\":\"" + hour_api + "\",\"data\":\"" + String(sensor.getCounter()) + "\"}}";

SendData::pushAPIData(json_steps);
```
```

### Debug Message

```plain
https://api.github.com/repos/mickahell/robots-data/dispatches
{"event_type":"watchy-data","client_payload":{"data-name":"steps","date":"06-05-2024","hour":"16","data":"0"}}
[    99][D][HTTPClient.cpp:303] beginInternal(): protocol: https, host: api.github.com port: 443 url: /repos/mickahell/robots-data/dispatches
[   117][D][HTTPClient.cpp:598] sendRequest(): request type: 'POST' redirCount: 0

[   124][E][WiFiGeneric.cpp:1583] hostByName(): DNS Failed for api.github.com
[   131][D][HTTPClient.cpp:1163] connect(): failed connect to api.github.com:443
[   139][W][HTTPClient.cpp:1483] returnError(): error(-1): connection refused
- not send to Github
-1
[   146][W][HTTPClient.cpp:1483] returnError(): error(-4): not connected

[HTTPS] GET... failed, error: connection refused
                                                [   154][D][HTTPClient.cpp:408] disconnect(): tcp is closed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-05-07:
Github does redirections. Don't know if this needs explicit allowed in code?

mickahell, 2024-05-14:
The problem was from my way to check if the network is on.
The code in the issue works fine

closed 2024-05-14 by mickahell