espressif/arduino-esp32 #9596: Same MAC address for different network interfaces

Avenitos, 2024-05-07:
### Board

Custom

### Device Description

Custom hardware with ESP32-S3 and W5500 ethernet

### Hardware Configuration

ESP32-S3 + W5500

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windiows 10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

After updgrade to 3.0.0-rc1 Arduino Core i saw that I was getting the same MAC address for different network interfaces.

Previously, for the Ethernet interface, the MAC address was issued with the last octet changed and there were no problems with their simultaneous operation.

How do I change the MAC address for the Ethernet interface?

### Sketch

```cpp
No need
```

### Debug Message

```plain
No need
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-07:
Can you at least show us how you were trying to get those MACs? For me they are all different

Avenitos, 2024-05-07:
`ETH.macAddress()`
and

me-no-dev, 2024-05-07:
I have 4 interfaces: AP, STA and 2 ETH. They all have different IPs and MACs. 
```cpp
ETH.begin();
Serial.println(ETH.macAddress());
WiFi.begin();
Serial.println(WiFi.macAddress());
```

me-no-dev, 2024-05-07:
Since you are already on 3.0.0, you can just `Serial.println(Network);` to print out all enabled interfaces, their MACs and IPs

Avenitos, 2024-05-08:
Yes, you are right, everything is fine. Perhaps I overlooked, because for W5500 the first octet of the MAC address changes, and for the usual RTL the last. Thank you!

closed 2024-05-08 by Avenitos