espressif/arduino-esp32 #9604: ESP going into exception while using BLE

haseebzaib, 2024-05-09:
### Board

ESP32-S3

### Device Description

Plain module

### Hardware Configuration

Nothing attached to device

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Window 10

### Flash frequency

80000000L

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello everyone. 

I have sketch in which simple BLE service is being implemented with 4 of its characteristics. 
In each characteristic a user input an string and I do operation on that. 

The issue I am facing is that, when ever I send the string from any characteristic I get hardfault 

And sometimes this hardfault occur and after trying 2 3 times it stops, but it is random for sure. 

Now I also increased stack size using this SET_LOOP_TASK_STACK_SIZE(20*1024); but this is not solved.

### Sketch

```cpp
#include <Arduino.h>
#include <SPIFFS.h>
#include <WiFi.h>
#include <ArduinoJson.h>
#include <WiFiClientSecure.h>
#include <PubSubClient.h>
#include <Preferences.h>
#include "aws_handling.h"
#include "ble_handling.h"

uint8_t status = 0;

char ssid[50];
char pass[50];

SET_LOOP_TASK_STACK_SIZE(20*1024);
void setup()
{
[... 318 lines cut ...]
         strcpy(pass, PASS_S.c_str());
        return 1;
    }
}
```

### Debug Message

```plain
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (BTC_TASK)
Core  0 register dump:
PC      : 0x40387463  PS      : 0x00060136  A0      : 0x4037fc5a  A1      : 0x3fcf6050  
A2      : 0x3fcf60c0  A3      : 0x00000000  A4      : 0x00000015  A5      : 0x00000000
A6      : 0x3fcb35e4  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x3fcb5d6c  
A10     : 0x00000042  A11     : 0x000000ba  A12     : 0xfffffd9c  A13     : 0x3fcf6a98
A14     : 0x3c126169  A15     : 0x3c1261bb  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffa

Backtrace: 0x40387460:0x3fcf6050 0x4037fc57:0x3fcf6130 0x420f9d36:0x3fcf6160 0x420ff952:0x3fcf6470 0x420ff98e:0x3fcf6500 0x4201a04a:0x3fcf6540 0x4209a356:0x3fcf6580 0x4200df9d:0x3fcf6600 0x420173ed:0x3fcf68c0 0x42012735:0x3fcf68e0 0x42013169:0x3fcf6940 0x420120be:0x3fcf6960 0x4200f8a5:0x3fcf69c0 0x420480b2:0x3fcf6a20 0x42048681:0x3fcf6a40 0x4206a705:0x3fcf6d40 0x4206c9e3:0x3fcf6d60
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

haseebzaib, 2024-05-09:
Also sometimes it works fine, but most of the time it goes into hardfault.

me-no-dev, 2024-05-09:
stack overflow in the BTC task.

lbernstone, 2024-05-09:
It is very unclear in your code where you are starting the BTC_TASK, but that stack needs to be increased.  The default size can be changed in [RTOS.h](https://github.com/espressif/arduino-esp32/blob/2.0.14/libraries/BLE/src/RTOS.h#L27).

haseebzaib, 2024-05-10:
I was looking for solutions of how to increase stack size for BLE task but I got no solution. I dont use arduino so I did not had an idea that you have to change it directly in the files and there is no way to change in your code. 

Anyways I got it working from some other ble library which I found and did the job.

haseebzaib, 2024-05-10:
> It is unclear in your code where you are starting the BTC_TASK,

I have no idea how does arduino starts this task, this was the example I found everywhere and just modified it for my case. Though I increased the "void loop stack size" but it did nothing.