espressif/arduino-esp32 #9607: Can't compile with CONFIG_COMPILER_OPTIMIZATION_PERF (-O2) option, ESP-IDF with arduino as a component

ScottyDoesKnow, 2024-05-09:
### Board

ESP32-WROOM-32

### Device Description

ESP32-2432S028 (Cheap Yellow Display)

### Hardware Configuration

No

### Version

latest development Release Candidate (RC-X)

### IDE Name

ESP-IDF Extension for Visual Studio Code

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

460800

### Description

Can't compile for performance. Not sure if this is the right place because the issues are with the the components this component uses. You can see a pull request to fix one of these issues here:

https://github.com/espressif/esp-rainmaker/pull/297

I manually made the same changes to fix this issue and another component broke with a reference to a potentially uninitialized enum variable. I fixed that and more strncpy errors popped up, at which point I gave up.

### Sketch

```cpp
In sdkconfig.defaults:

CONFIG_COMPILER_OPTIMIZATION_DEFAULT=n
CONFIG_COMPILER_OPTIMIZATION_SIZE=n
CONFIG_COMPILER_OPTIMIZATION_PERF=y
```

### Debug Message

```plain
/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_schedule.c:635:17: error: 'strncpy' output truncated before terminating nul copying
as many bytes from a string as its length [-Werror=stringop-truncation]
  635 |                 strncpy(*info, _info, strlen(_info));
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ricardoquesada, 2024-06-05:
+1. 

Triggering this issue as well.
When I was using Arduino Core v2.0.17 + ESP-IDF 4.4.7 it was compiling Ok.
When I updated it to Arduino Core v3.0.0 + ESP-IDF 5.1.4, I fails to compile with that error.

Using Arduino Core as an ESP-IDF component.

me-no-dev, 2024-06-05:
The issue is in rainmaker. Not sure what action we can take. @ricardoquesada @ScottyDoesKnow suggesting file an issue in the RainMaker repo

ricardoquesada, 2024-06-06:
done: https://github.com/espressif/esp-rainmaker/issues/315

ricardoquesada, 2024-06-06:
Filed a similar bug for arduino-core: https://github.com/espressif/arduino-esp32/issues/9784

PR #9801 (2024-06-10): change(idf): Rename component example and fix compilation warnings
## Description of Change
Fixes component compilation and workflow conflicts.
This will also fix the workflows being canceled randomly in PRs.

## Tests scenarios
Locally and CI

## Related links
Closes #9607
Closes #9784
closed 2024-06-10 by me-no-dev