espressif/arduino-esp32 #9608: Can't create a USB Composite device. Need Guidance!

iminsideyou, 2024-05-09:
### Board

esp32-s3-devkit c

### Device Description

esp32-s3-devkit c

### Hardware Configuration

Nothing else is attached to the device

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I am trying to understand how I an create a usb composite device. When I upload this sketch it just creates 1 usb input device. I want the mouse under it's own interface and the keyboard under it's own interface

### Sketch

```cpp
#include "Arduino.h"
#include "USB.h"
#include "USBHIDMouse.h"
#include "USBHIDKeyboard.h"

USBHIDMouse Mouse;
USBHIDKeyboard Keyboard;

void setup() {

  Serial.begin(115200); 
  Mouse.begin();
  Keyboard.begin();
  USB.begin();

}

void loop() {

}
```

### Debug Message

```plain
No error. I just don't understand why it isn't creating a composite device but somehow in the composite example it does.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-05-09:
This forum is for bug reports with the codebase hosted here.  Please ask general questions at https://esp32.com

iminsideyou, 2024-05-09:
> This forum is for bug reports with the codebase hosted here. Please ask general questions at https://esp32.com

Would this be a feature request then? The example only creates a composite device only when I include #include "FirmwareMSC.h"

if I remove the items related to that it creates everything under 1 input device. Is it possible to make a feature request of an example of creating a usb composite device with only a mouse and keyboard? 

Is the forum that active? I am just worried my question won't get answered.

me-no-dev, 2024-05-10:
Here it is: https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/KeyboardAndMouseControl/KeyboardAndMouseControl.ino

closed 2024-05-10 by me-no-dev

iminsideyou, 2024-05-10:
> Here it is: https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/KeyboardAndMouseControl/KeyboardAndMouseControl.ino

This literally doesn't create a usb composite device.... only 1 usb input with a mouse and keyboard

me-no-dev, 2024-05-10:
> Is it possible to make a feature request of an example of creating a usb composite device with only a mouse and keyboard?

what do you mean by "composite device" here?

iminsideyou, 2024-05-11:
> > Is it possible to make a feature request of an example of creating a usb composite device with only a mouse and keyboard?
> 
> what do you mean by "composite device" here?

[image]

this.... A usb composite device.... That sketch example does not create that.... and I need it to create that.

iminsideyou, 2024-05-14:
> > Is it possible to make a feature request of an example of creating a usb composite device with only a mouse and keyboard?
> 
> what do you mean by "composite device" here?

Any ideas?

iminsideyou, 2024-05-14:
> Sorry, but I do not use Windows and have no idea what those icons show. If you can send the USB descriptor, we could check what is exactly making your device to be "Composite device". If it requires two separate HID devices, rather then once HID device with descriptors for Keyboard and Mouse, then you are out of luck

Example of a real usb composite device
[gaming-keyboard.txt](https://github.com/espressif/arduino-esp32/files/15312010/gaming-keyboard.txt)

Example of the esp32 
[esp32.txt](https://github.com/espressif/arduino-esp32/files/15312017/esp32.txt)

me-no-dev, 2024-05-14:
So yes, as I suspected. Yours have 3 separate USB devices (each one is HID) and we can create currently only 1 such USB device in Arduino. One device can have multiple HID report descriptors (Keyboard, Mouse, etc.) and one of the devices that your gaming keyboard exports also has multiple HID report descriptors. TinyUSB itself permits such devices, but that is not implemented in Arduino (for ease of use, etc.). You could try to use TinyUSB directly