espressif/arduino-esp32 #9614: ESP32-S2-Mini-N4 Touch Values very high and do not change

hykilpikonna, 2024-05-11:
### Board

ESP32-S2-Mini-N4

### Device Description

[image]
[image]

### Hardware Configuration

Touch GPIO pins 1 - 12 are connected to 2.54 pin sockets, which directly connect to the self-capacitance touch board.

There is also a MPR121 that's connected to the same pins.

```ini
[env:control]
platform = espressif32 @ ^6.6.0
board = featheresp32-s2
;board = lolin_s2_mini
framework = arduino
upload_speed = 921600
platform_packages =
;    toolchain-xtensa32 @ 2.80400.210211
    platformio/framework-arduinoespressif32@^3.20016.0
;    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
;    platformio/framework-arduinoespressif32@^3.20003.0
build_unflags =
    -DBOARD_HAS_PSRAM
```

### Version

v2.0.14

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

The raw readings of the 12 touch pins do not change at all, and are way beyond normal value ranges. Meanwhile, the MPR121 connected to the same 12 pins functions correctly. This issue was previously observed by https://github.com/espressif/arduino-esp32/issues/6837 on the same ESP32-S2-Mini SMT package, but their solution of downgrading to Arduino IDE 1.8.19 and Arduino Core 2.0.3 did not work for me.

### Sketch

```cpp
#include "Arduino.h"
#include "driver/touch_sensor.h"

constexpr auto touches = {TOUCH_PAD_NUM1, TOUCH_PAD_NUM2, TOUCH_PAD_NUM3, TOUCH_PAD_NUM4, TOUCH_PAD_NUM5, TOUCH_PAD_NUM6,
        TOUCH_PAD_NUM7, TOUCH_PAD_NUM8, TOUCH_PAD_NUM9, TOUCH_PAD_NUM10, TOUCH_PAD_NUM11, TOUCH_PAD_NUM12};

void setup() {
    Serial.begin(115200);
    delay(1000);  // give me time to bring up the serial monitor

    touch_pad_init();
     for (const auto touch : touches) {
        touch_pad_io_init(touch);
    }
}

void loop() {
    // for (int i = 1; i < 12; ++i) {
    for (const auto i : touches) {
        // const auto touch = touchRead(i);
        uint32_t touch = 0;
        touch_pad_read_raw_data(i, &touch);
        Serial.printf("%d,", touch);
    }
    Serial.println();

    delay(80);
}
```

### Debug Message

```plain
![image](https://github.com/espressif/arduino-esp32/assets/22280294/90397273-996a-4d65-8a81-d0a293660851)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2024-06-06:
@hykilpikonna - your code is using IDF code to read the Touch pads.
For more information, please read the IDF documentation:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/touch_pad.html

You can also read the Arduino code, which is based on the IDF calls:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-touch.c

ESP32-S2 uses the IDF Touch API called `SOC_TOUCH_VERSION_2`

I have tested the Arduino Touch API with Arduino Core 3.0.1 and 2.0.14.
It works fine in both cases.

``` cpp
constexpr auto touches = {TOUCH_PAD_NUM1, TOUCH_PAD_NUM2, TOUCH_PAD_NUM3, TOUCH_PAD_NUM4, TOUCH_PAD_NUM5, TOUCH_PAD_NUM6,
        TOUCH_PAD_NUM7, TOUCH_PAD_NUM8, TOUCH_PAD_NUM9, TOUCH_PAD_NUM10, TOUCH_PAD_NUM11, TOUCH_PAD_NUM12};

void setup() {
    Serial.begin(115200);
    delay(1000);  // give me time to bring up the serial monitor
}

void loop() {
    for (const auto i : touches) {
        const auto touch = touchRead(i);
        Serial.printf("%d,", touch);
    }
    Serial.println();
    delay(500);
}
```

VojtechBartoska, 2024-09-12:
Hello, closing as wontfix.

closed 2024-09-12 by VojtechBartoska