espressif/arduino-esp32 #9615: Esp32 Cam Example

faizannazir, 2024-05-11:
### Board

AI - Thinker ESP32 CAM

### Device Description

Esp32 Ai thinker cam

### Hardware Configuration

Esp32 Cam face detection and recognition 

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Window 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Esp32 Cam face detection and recognition not working on esp32 cam module another guy tested on other board he also facing same issue

### Sketch

```cpp
Esp32 Cam webserver example
```

### Debug Message

```plain
Not debugged
```

### Other Steps to Reproduce

tested with other board 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-11:
the new face detection and recognition library is too slow on ESP32, so we have removed it and is only available on ESP32-S3. If you need the old one, just use older version ESP32 Arduino

faizannazir, 2024-05-12:
> the new face detection and recognition library is too slow on ESP32, so we have removed it and is only available on ESP32-S3. If you need the old one, just use older version ESP32 Arduino

Which version support it ? and is it possible to use face recogination without web portal

VojtechBartoska, 2024-09-12:
@me-no-dev can you please clarify the points from last comment? After that, this ticket can be closed. Thanks

me-no-dev, 2024-10-21:
ESP32 use version 1.0.6
ESP32-S3 use versions up to 3.0.x
Feature does not work starting with 3.1.0

closed 2024-10-21 by me-no-dev