espressif/arduino-esp32 #9616: packages\esp32\hardware\esp32\2.0.16\cores\esp32/core_version.h:2:32: error: too many decimal points in number

StefanL38, 2024-05-12:
### Board

all

### Device Description

all

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

1.8.19

### Operating System

Win11

### Flash frequency

all

### PSRAM enabled

no

### Upload speed

115200

### Description

If I try to use the core_version.h so that my application can print this info to the serial monitor
I get this eror for obvious reasons
packages\esp32\hardware\esp32\2.0.16\cores\esp32/core_version.h:2:32: error: too many decimal points in number

`#define ARDUINO_ESP32_GIT_VER 0x54927338
#define ARDUINO_ESP32_GIT_DESC 2.0.16
#define ARDUINO_ESP32_RELEASE_2_0_16
#define ARDUINO_ESP32_RELEASE "2_0_16"
`
if this information is stored into a C++-header-file with file-extension ".h"
it must be compilable. To me it seams somebody was a bit lazy and just pasted the dotted number

so you should change this that it does not cause a compiler-error

either
#define ARDUINO_ESP32_GIT_DESC "2.0.16"
or
#define ARDUINO_ESP32_GIT_DESC "2_0_16"

### Sketch

```cpp
any sketch
```

### Debug Message

```plain
packages\esp32\hardware\esp32\2.0.16\cores\esp32/core_version.h:2:32: error: too many decimal points in number
```

### Other Steps to Reproduce

simply add the line 

`#include <core_version.h>`

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-12:
What you ask for is already defined

StefanL38, 2024-05-12:
> What you ask for is already defined

You should read more carefully
there is line 2 of the file
`#define ARDUINO_ESP32_GIT_DESC 2.0.16`

which has two decimal points.

If I add 
`#include <core_version.h>
`

I get a compiler-error too **many decimal points.**

So can you explain to me how to keep the decimal points and to avoid the compiler-error?

me-no-dev, 2024-05-12:
I did read it carefully. You ask for either "2.0.16" or "2_0_16". The second option is already there but with a different name

me-no-dev, 2024-05-12:
Just including the file will not cause error, unless you use the define. ESP32_ARDUINO_VERSION is what you should use

me-no-dev, 2024-05-12:
Including esp_arduino_version.h is yet another way to access the core version

StefanL38, 2024-05-12:
Ok Thank you very much for explaining it

closed 2024-05-12 by StefanL38