espressif/arduino-esp32 #9619: Add specific #define for each board in variants

fredlcore, 2024-05-13:
### Related area

Board support

### Hardware specification

All models

### Is your feature request related to a problem?

No.

### Describe the solution you'd like

My project runs on several ESP32 board types (DevKit, Olimex EVB, Olimex POE, Olimex POE-ISO) and in some parts of the code it needs to compile differently based on the board type. For now, I use dirty hacks such as definitions of `RX1` etc. which are defined differently across said boards. But this is not a reliable solution in the long run. 
Therefore, I would like to suggest to add a specific, unique definement for each board variant, for example in `pins_arduino.h` as this is always included in all configurations.
For example:
`#define BOARD_MODEL AirM2M_CORE_ESP32C3`
or
`#define BOARD_MODEL AVENTEN_S3_SYNC`
or
`#define BOARD_MODEL BEE_DATA_LOGGER`
etc.

### Describe alternatives you've considered

Using other definements such as `RX1`, these work for some constellations, but not all.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-05-13:
There is a define for ARDUINO_BOARD

fredlcore, 2024-05-13:
Oh, really?  Thanks! Where is this defined exactly? And does it differentiate between each board type or just the platform (such as ARM, ESP32 etc.)?

fredlcore, 2024-05-13:
Sorry, found it myself - that's great, so you know if this is also available when using the Arduino IDE? I only use PlatformIO, and it works there, but not sure if also on Arduino IDE which most of my users use...

fredlcore, 2024-05-13:
Thanks! I'm just wondering if I made an error in thinking in my post above:
Because since you cannot compare strings with `#if defined`, there is not really a way to do conditional code compilation, am I right?
So it would rather have to be
`#define AIRM2M_CORE_ESP32C3`
etc.
Or am I missing some other way to achieve my goal?

fredlcore, 2024-05-13:
Ok, I found out that whatever is defined in `board.txt` under `.build.board` will be converted into a definement following the structure `ARDUINO_` followed by the all-caps value of the content of `.build.board`. So in my case, it's `ARDUINO_ESP32_EVB`, `ARDUINO_ESP32_POE` or `ARDUINO_ESP32_DEV`.

Thank you for the fast support, I'll close this now.

closed 2024-05-13 by fredlcore