espressif/arduino-esp32 #9630: Arduino Ethernet library not working with arduino-esp32:3.0.0-rc2

kkrygiel, 2024-05-14:
### Board

ESP32

### Device Description

esp32 devkit v1

### Hardware Configuration

Custom pcb with ESP32 and ethernet module (wiznet w5500 spi to Ethernet chip).

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Ubuntu

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I have custom pcb with esp32 and ethernet module connected with SPI interface. I use Arduino IDE and Arduino Ethernet library. It is working with esp32:3.0.0-alpha3 but it is not compatible with esp32:3.0.0-rc2. Etherent library uses IPAddress::raw_address function which is private. In previous versions IPAddress defined EthernetClass, DhcpClass and DNSClient as friend. In recent update onlu classes UDP, Client and Server are declared as friends. Because of that change I can't use Arduino Ethernet library. Can You change it back?

### Sketch

```cpp
#include <Ethernet.h>

String x = Ethernet.localIP().toString();
```

### Debug Message

```plain
/home/ABC/Arduino/libraries/Ethernet/src/Dns.cpp: In member function 'uint16_t DNSClient::ProcessResponse(uint16_t, IPAddress&)':
/home/ABC/Arduino/libraries/Ethernet/src/Dns.cpp:340:55: error: 'uint8_t* IPAddress::raw_address()' is private within this context
  340 |                         iUdp.read(aAddress.raw_address(), 4);
      |                                   ~~~~~~~~~~~~~~~~~~~~^~
In file included from /home/ABC/.arduino15/packages/esp32/hardware/esp32/3.0.0-rc2/cores/esp32/Arduino.h:196
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-14:
What is `Ethernet.h`? It doers not seem to be from our repository. We have `ETH.h`

JAndrassy, 2024-05-14:
@me-no-dev "Ethernet library" is always the Arduino Ethernet library for W5500. The 16 years old and still used first Arduino networking library which established the Arduino networking API.
https://www.arduino.cc/reference/en/libraries/ethernet/

This PR https://github.com/espressif/arduino-esp32/pull/9522 on my offered a compatibility of ESP32 Ethernet library with the Arduino library

me-no-dev, 2024-05-14:
If this is not a project that will run on non-ESP MCUs, then using our ETH library is much better offering than any other Ethernet lib. API is not all that different and is tightly integrated into ESPs network stack

kkrygiel, 2024-05-15:
Maybe You can leave those 3 lines with friends declaration so We can use old Ethernet library in case we need support for many MCUs?

me-no-dev, 2024-05-15:
@kkrygiel will add those, but I still recommend you to look into our ethernet implementation for best experience.
@JAndrassy maybe you can think of a header-only wrapper that can be included instead of Ethernet.h for people that would want that backward compatibility (not included by default and not using memory when not included)

JAndrassy, 2024-05-15:
> @JAndrassy maybe you can think of a header-only wrapper that can be included instead of Ethernet.h for people that would want that backward compatibility (not included by default and not using memory when not included)

that was my plan but it required a just little more code than I would put into a header. but the cpp in https://github.com/espressif/arduino-esp32/pull/9522/files  is simple

JAndrassy, 2024-05-15:
and the main reason for the cpp is the instance

closed 2024-05-15 by me-no-dev