espressif/arduino-esp32 #9641: arduino-esp32 as a component on SD card via SPI failure

kungfupizza, 2024-05-15:
### Board

ESP32S3 Wroom N8R2

### Device Description

SPI connected to SD Card for Wroom esp32s3

### Hardware Configuration

My pinout:
MOSI / DI	IO13
MISO / DO	IO14
CLK	                IO15
CS	                IO16

### Version

latest development Release Candidate (RC-X)

### IDE Name

ESP-IDF

### Operating System

Windows 11

### Flash frequency

240Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am using esp32-arduino as an IDF component. I have SD card on SPI, it initialises file but on reading a file, i can see it reading for a while then failing with
```
[E][sd_diskio.cpp:132] sdSelectCard(): Select Failed
[201932][E][sd_diskio.cpp:630] ff_sd_status(): Check status failed
[202440][E][sd_diskio.cpp:132] sdSelectCard(): Select Failed
[202445][E][sd_diskio.cpp:630] ff_sd_status(): Check status failed
[202951][E][sd_diskio.cpp:132] sdSelectCard(): Select Failed
[202956][E][sd_diskio.cpp:630] ff_sd_status(): Check status failed
[203464][E][sd_diskio.cpp:132] sdSelectCard(): Select Failed
```

### Sketch

```cpp
SPIClass spi = SPIClass(FSPI);
esp_err_t sd_card_init( void )
{
    // init from pins
    spi.begin(PRJ_CONFIG_SDCARD_SCK, PRJ_CONFIG_SDCARD_MISO, PRJ_CONFIG_SDCARD_MOSI, PRJ_CONFIG_SDCARD_CS);
    if(SD.begin(PRJ_CONFIG_SDCARD_CS, spi, 10000000) == false){
        ESP_LOGE(TAG, "Failure in SD card init");
        return ESP_FAIL;
    }
...
}
```
```

### Debug Message

```plain
[ 20376][E][sd_diskio.cpp:132] sdSelectCard(): Select Failed
[ 20381][E][sd_diskio.cpp:630] ff_sd_status(): Check status failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-05-15:
You need to test with a [minimal example](https://wokwi.com/projects/397980473830806529).  SD card is known to work.  If you have made changes to your sdkconfig, you need to post those changes.

kungfupizza, 2024-05-15:
`sdkconfig` attached.
[sdkconfig.txt](https://github.com/espressif/arduino-esp32/files/15326428/sdkconfig.txt)

lbernstone, 2024-05-15:
You have hundreds of differences between the arduino sdkconfig and yours.  Start from the [known good config](https://github.com/espressif/esp32-arduino-libs/blob/idf-d23b7a0361/esp32s3/sdkconfig) and make your changes to that.

VojtechBartoska, 2024-06-03:
did you solve your problem @kungfupizza?

VojtechBartoska, 2024-09-12:
Hello, closing as Expired due to no answer.

closed 2024-09-12 by VojtechBartoska