espressif/arduino-esp32 #9644: Error Partition Tables

victorMartins2024, 2024-05-16:
### Board

ESP32 38 pines NODE MCU 

### Device Description

I'm developing a project of telemetry using esp-32. I put the codes with the libraries and when i pressed the "ESP-IDF: Build `Project" looks like a mistake. Can you help me ? 

I wait the retorn. 

Follow the BUG: 

`FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/chip-debug-report.cpp.obj 
C:\Users\victor.martins\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -DESP_PLATFORM -DIDF_VER=\"v5.1.3\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/victor.martins/esp/project-name/build/config -IC:/Users/victor.martins/esp/project-name/components/arduino/variants/esp32 -IC:/Users/victor.martins/esp/project-name/components/arduino/cores/esp32 -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/ArduinoOTA/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/AsyncUDP/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/BLE/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/BluetoothSerial/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/DNSServer/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/EEPROM/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/ESP_I2S/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/ESP_NOW/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/ESP_SR/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/ESPmDNS/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/Ethernet/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/FFat/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/FS/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/HTTPClient/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/HTTPUpdate/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/Insights/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/LittleFS/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/NetBIOS/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/Network/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/PPP/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/Preferences/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/RainMaker/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/SD_MMC/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/SD/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/SimpleBLE/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/SPIFFS/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/SPI/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/Ticker/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/Update/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/USB/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/WebServer/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/NetworkClientSecure/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/WiFi/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/WiFiProv/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/Wire/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/PubSubClient/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/LiquidCrystal_I2C/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/MFRC522/src -IC:/Users/victor.martins/esp/project-name/components/arduino/libraries/INA226/src -IC:/Users/victor.martins/esp/project-name/components/arduino/cores/esp32/libb64 -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/newlib/platform_include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/freertos/esp_additions/include/freertos -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/freertos/esp_additions/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/freertos/esp_additions/arch/xtensa/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_hw_support/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_hw_support/include/soc -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_hw_support/include/soc/esp32 -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_hw_support/port/esp32/. -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_hw_support/port/esp32/private_include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/heap/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/log/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/soc/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/soc/esp32 -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/soc/esp32/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/hal/esp32/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/hal/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/hal/platform_port/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_rom/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_rom/include/esp32 -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_rom/esp32 -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_common/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_system/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_system/port/soc -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_system/port/include/private -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/xtensa/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/xtensa/esp32/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/lwip/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/lwip/include/apps -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/lwip/include/apps/sntp -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/lwip/lwip/src/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/lwip/port/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/lwip/port/freertos/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/lwip/port/esp32xx/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/lwip/port/esp32xx/include/arch -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/spi_flash/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_partition/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/mbedtls/port/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/mbedtls/mbedtls/library -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/mbedtls/esp_crt_bundle/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/wifi_provisioning/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/protocomm/include/common -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/protocomm/include/security -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/protocomm/include/transports -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/protocomm/include/crypto/srp6a -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_timer/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_wifi/include -IC:/Users/victor.martins/esp/v5.1.3/esp-idf/components/esp_wifi/wif [... cut ...]
[1133/1286] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-misc.c.obj
ninja: build stopped: subcommand failed.``
```

### Debug Message

```plain
I'm developing a project of telemetry using esp-32. I put the codes with the libraries and when i pressed the "ESP-IDF: Build `Project" looks like a mistake. Can you help me ? 

I wait the retorn.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2024-05-17:
If you want to use vanilla IDF 5.1.3, you will need to checkout an older version of arduino (maybe 3.0.0-alpha).  Otherwise, use the latest version of release/v5.1

victorMartins2024, 2024-05-29:
As soon as I go back to the Arduino 3.0.0 version, this error will disappear and you will be able to compile my code for the ESP32? It's trustable?

lbernstone, 2024-05-29:
I don't think my response had any guarantees with it :) .  You must use a version of IDF that supports all the functions arduino-esp32 requires.  This typically means the code that is in the release/v5.1 branch at the time of compile for a particular version of arduino-esp32 3.0.  If you want a more reliable versioning and the sdkconfigs that go along with arduino, use the [lib-builder](https://github.com/espressif/esp32-arduino-lib-builder).

VojtechBartoska, 2024-09-12:
Hello @victorMartins2024, is this still valid?

JooJooBee666, 2024-11-13:
I just ran in to this one myself.  I'm trying to update OTA from a URL instead of a file upload but needed to define LITLEFS as the type only to find it doesn't work.  I'm currently running version 6.9.0.  I', confused about this as I understoof LittleFS was the new version and should be used going forward.  Is that no longer the case?  Is support for LittleFS just not gettign any love or being deprecated?  This issue has been open for 6 months which makes me wonder about this...

closed 2025-01-15 by Parsaabasi