espressif/arduino-esp32 #9645: ETH_W5500_IDF_SPI.ino // It can not get dhcp ip.

sirapol, 2024-05-17:
### Board

ESP32 Dev module.

### Device Description

[image]
[image]

### Hardware Configuration

#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS 27
#define ETH_PHY_IRQ 35
#define ETH_PHY_RST 15

// SPI pins
#define ETH_SPI_SCK 18
#define ETH_SPI_MISO 19
#define ETH_SPI_MOSI 23

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino 1.8.19

### Operating System

Ubuntu

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I using example.
It can not get dhcp ip from router.
I verify same cable to pc. It can get dhcp.

How to solve it ?

### Sketch

```cpp
//-------------------------
// This code can interface with W5500. But, It not obtain DHCP IP
#include <ETH.h>
#include <SPI.h>

#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS 27
#define ETH_PHY_IRQ 35
#define ETH_PHY_RST 15

// SPI pins
#define ETH_SPI_SCK 18
#define ETH_SPI_MISO 19
#define ETH_SPI_MOSI 23

static bool eth_connected = false;

void onEvent(arduino_event_id_t event, arduino_event_info_t info)
{
    switch (event)
    {
    case ARDUINO_EVENT_ETH_START:
        Serial.println("ETH Started");
        // set eth hostname here
        ETH.setHostname("esp32-eth0");
        break;
    case ARDUINO_EVENT_ETH_CONNECTED:
        Serial.println("ETH Connected");
        ETH.
        break;
        break;
    case ARDUINO_EVENT_ETH_GOT_IP:
        Serial.printf("ETH Got IP: '%s'\n", esp_netif_get_desc(info.got_ip.esp_netif));
        Serial.println(ETH);
        eth_connected = true;
        break;
    case ARDUINO_EVENT_ETH_LOST_IP:
        Serial.println("ETH Lost IP");
        eth_connected = false;
        break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
        Serial.println("ETH Disconnected");
        eth_connected = false;
        break;
    case ARDUINO_EVENT_ETH_STOP:
        Serial.println("ETH Stopped");
        eth_connected = false;
        break;
    default:
        break;
    }
}

void testClient(const char *host, uint16_t port)
{
    Serial.print("\nconnecting to ");
    Serial.println(host);

    NetworkClient client;
    if (!client.connect(host, port))
    {
        Serial.println("connection failed");
        return;
    }
    client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
    while (client.connected() && !client.available())
        ;
    while (client.available())
    {
        Serial.write(client.read());
    }

    Serial.println("closing connection\n");
    client.stop();
}

void setup()
{
    Serial.begin(115200);
    delay(200);
    Network.onEvent(onEvent);

    SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
    ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI);
    Serial.println("Fin Setup");
}
unsigned long tUpdate;

void loop()
{
    // if (millis() - tUpdate > 2000)
    // {
    //     tUpdate = millis();
    //     Serial.println(ETH.localIP());
    //     Serial.println(ETH);
    // }
    if (eth_connected)
    {
        testClient("google.com", 80);
    }
    delay(1000);
}
```

### Debug Message

```plain
10:52:42.363 -> ets Jun  8 2016 00:22:57
10:52:42.363 -> 
10:52:42.363 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
10:52:42.396 -> configsip: 0, SPIWP:0xee
10:52:42.396 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
10:52:42.396 -> mode:DIO, clock div:1
10:52:42.396 -> load:0x3fff0030,len:1448
10:52:42.396 -> load:0x40078000,len:14844
10:52:42.396 -> ho 0 tail 12 room 4
10:52:42.396 -> load:0x40080400,len:4
10:52:42.396 -> load:0x40080404,len:3356
10:52:42.396 -> entry 0x4008059c
10:52:42.562 -> [     2][D][esp32-hal-cpu.c:264] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
10:52:42.595 -> =========== Before Setup Start ===========
10:52:42.595 -> Chip Info:
10:52:42.595 -> ------------------------------------------
10:52:42.595 ->   Model             : ESP32
10:52:42.628 ->   Package           : D0WD-Q6
10:52:42.628 ->   Revision          : 100
10:52:42.628 ->   Cores             : 2
10:52:42.628 ->   Frequency         : 240 MHz
10:52:42.628 ->   Embedded Flash    : No
10:52:42.661 ->   Embedded PSRAM    : No
10:52:42.661 ->   2.4GHz WiFi       : Yes
10:52:42.661 ->   Classic BT        : Yes
10:52:42.661 ->   BT Low Energy     : Yes
10:52:42.661 ->   IEEE 802.15.4     : No
10:52:42.661 -> ------------------------------------------
10:52:42.661 -> INTERNAL Memory Info:
10:52:42.695 -> ------------------------------------------
10:52:42.695 ->   Total Size        :   379844 B ( 370.9 KB)
10:52:42.695 ->   Free Bytes        :   349724 B ( 341.5 KB)
10:52:42.695 ->   Allocated Bytes   :    23052 B (  22.5 KB)
10:52:42.695 ->   Minimum Free Bytes:   344204 B ( 336.1 KB)
10:52:42.728 ->   Largest Free Block:   114676 B ( 112.0 KB)
10:52:42.728 -> ------------------------------------------
10:52:42.728 -> Flash Info:
10:52:42.728 -> ------------------------------------------
10:52:42.728 ->   Chip Size         :  4194304 B (4 MB)
10:52:42.728 ->   Block Size        :    65536 B (  64.0 KB)
10:52:42.761 ->   Sector Size       :     4096 B (   4.0 KB)
10:52:42.761 ->   Page Size         :      256 B (   0.2 KB)
10:52:42.761 ->   Bus Speed         : 80 MHz
10:52:42.761 ->   Bus Mode          : QIO
10:52:42.761 -> ------------------------------------------
10:52:42.794 -> Partitions Info:
10:52:42.794 -> ------------------------------------------
10:52:42.794 ->                 nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
10:52:42.794 ->             otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
10:52:42.841 ->                app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
10:52:42.841 ->                app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
10:52:42.861 ->              spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
10:52:42.861 ->            coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
10:52:42.894 -> ------------------------------------------
10:52:42.894 -> Software Info:
10:52:42.894 -> ------------------------------------------
10:52:42.894 ->   Compile Date/Time : May 17 2024 10:51:26
10:52:42.894 ->   Compile Host OS   : linux
10:52:42.927 ->   ESP-IDF Version   : v5.1.4-51-g442a798083-dirty
10:52:42.927 ->   Arduino Version   : 3.0.0
10:52:42.927 -> ------------------------------------------
10:52:42.927 -> Board Info:
10:52:42.927 -> ------------------------------------------
10:52:42.927 ->   Arduino Board     : ESP32_DEV
10:52:42.960 ->   Arduino Variant   : esp32
10:52:42.960 ->   Arduino FQBN      : esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=debug,EraseFlash=none,ZigbeeMode=default
10:52:42.960 -> ============ Before Setup End ============
10:52:43.324 -> ETH Started
10:52:43.324 -> Fin Setup
10:52:43.324 -> ETH Started
10:52:43.324 -> =========== After Setup Start ============
10:52:43.324 -> INTERNAL Memory Info:
10:52:43.324 -> ------------------------------------------
10:52:43.324 ->   Total Size        :   379844 B ( 370.9 KB)
10:52:43.324 ->   Free Bytes        :   326764 B ( 319.1 KB)
10:52:43.357 ->   Allocated Bytes   :    44548 B (  43.5 KB)
10:52:43.357 ->   Minimum Free Bytes:   321012 B ( 313.5 KB)
10:52:43.357 ->   Largest Free Block:   114676 B ( 112.0 KB)
10:52:43.357 -> ------------------------------------------
10:52:43.357 -> GPIO Info:
10:52:43.357 -> ------------------------------------------
10:52:43.357 ->   GPIO : BUS_TYPE[bus/unit][chan]
10:52:43.357 ->   --------------------------------------  
10:52:43.357 ->      1 : UART_TX[0]
10:52:43.357 ->      3 : UART_RX[0]
10:52:43.357 ->     15 : ETHERNET_SPI
10:52:43.357 ->     18 : SPI_MASTER_SCK[3]
10:52:43.357 ->     19 : SPI_MASTER_MISO[3]
10:52:43.357 ->     23 : SPI_MASTER_MOSI[3]
10:52:43.357 ->     27 : ETH_CS
10:52:43.357 ->     35 : ETHERNET_SPI
10:52:43.357 -> ============ After Setup End =============
10:52:47.296 -> ETH Connected
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-17:
I have not tested with your particular pins, but this sketch works fine on our end. I use it almost daily. Can you verify the pins? Also maybe try removing the pull-ups and capacitor from the IRQ line?

sirapol, 2024-05-17:
> I have not tested with your particular pins, but this sketch works fine on our end. I use it almost daily. Can you verify the pins? Also maybe try removing the pull-ups and capacitor from the IRQ line?

This sch, I copy from another product ( Deploy more 5,000pcs ) , But it using IDF.

me-no-dev, 2024-05-17:
You are using IDF now too. Just abstracted to Arduino

sirapol, 2024-05-17:
> You are using IDF now too. Just abstracted to Arduino

I not responsible about that project idf.
And this project prefer Arduino IDE

me-no-dev, 2024-05-17:
What I am trying to say is that Arduino is also using IDF so the code should work and it does work for us. We have tested W5500 quite a bit. This is from the latest 3.0.0-RC3:
```
[  1005][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 1 - ETH_START
ETH Started
[  2990][V][ETH.cpp:94] _onEthEvent(): eth0 Connected
[  2994][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 3 - ETH_CONNECTED
ETH Connected
[  7077][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 5 - ETH_GOT_IP
ETH Got IP[2]: 'eth0' (en1)
*eth0: <UP,100M,FULL_DUPLEX,AUTO,ADDR:0x1> (DHCPC,GARP,IP_MOD)
      ether 32:C6:F7:F4:2F:8C
      inet 192.168.52.161 netmask 255.255.255.0 broadcast 192.168.52.255
      gateway 192.168.52.1 dns 8.8.8.8

```

sirapol, 2024-05-17:
> What I am trying to say is that Arduino is also using IDF so the code should work and it does work for us. We have tested W5500 quite a bit. This is from the latest 3.0.0-RC3:
>
> ```
> [ [...]

What is your hardware ?

felmue, 2024-05-17:
Hello @sirapol 

I tried your code verbatim and for me it works too - the W5500 gets an IP address just fine.

I noticed that in your schematic TOCAP (W5500 pin 20) is connected to C13 which is a 10 nF. According to the specifications (see [here](https://cdn.sparkfun.com/datasheets/Dev/Arduino/Shields/W5500_datasheet_v1.0.2_1.pdf)) it should be a 4.7 uF capacitor.

_20 TOCAP - AO External Reference Capacitor
This pin must be connected to a 4.7uF capacitor.
The trace length to capacitor should be short to
stabilize the internal signals_

Thanks
Felix

Jason2866, 2024-05-18:
> This sch, I copy from another product ( Deploy more 5,000pcs ) , But it using IDF.

Wrong circuit design does not get corrected by amount of build pieces ;-)
The wrong choosen `AO External Reference Capacitor` does harm.

TD-er, 2024-05-21:
Also another design tip for future designs...
You should separate the power nets for the Ethernet chip and the rest. In your schematic snippets they seem to be using the same nets. (GND and 3V3)
So have a **single location** where both GNDs are connected using for example a beefy 0 Ohm resistor to connect both GNDs and use a **ferrite bead** to separate both 3V3 nets.

It will most likely not cause these issues (wrong capacitor is more likely as pointed out by others), but it will make it more likely the CE testing will pass without issues.

Also please look into how the metal shield of the RJ45 connector is connected to the rest of the nets.
There is lots of info online about this. Just to be sure there won't be any issue when someone is using Ethernet cables with shielding.

sirapol, 2024-05-21:
After i change C13 to 4.7uF. I error more.

```text
18:05:30.785 -> E (330) w5500.mac: W5500 version mismatched, expected 0x04, got 0x00
18:05:30.785 -> E (330) w5500.mac: emac_w5500_init(825): vefiry chip ID failed
18:05:30.785 -> E (330) esp_eth: esp_eth_driver_install(228): init mac failed
18:05:30.785 -> Fin Setup

```

TD-er, 2024-05-21:
Maybe a closer inspection of the W5500 chip to double check it isn't a W6100 or W5100 or some clone?

Jason2866, 2024-05-23:
Maybe using the [4 GPIO mode](https://github.com/espressif/arduino-esp32/pull/9659) stabilizies your setup.
Just connected a dirt cheap diy W5500 module to an ESP32-C3 (via Dupont wires!!) and it is working rock solid this way.

sirapol, 2024-06-05:
I investigate with W5500 module it same problem.
I try with esp-idf last stable version. It work with W5500 module.

I think my hardware is fault.

I will take time and report again. Can i still open this case ?

closed 2025-01-15 by Parsaabasi