espressif/arduino-esp32 #9646: ESP32-C3 Arduino IDE Upload ERROR when USB CDC on, No serial data received Error occurs

SooDragon, 2024-05-17:
### Board

ESP32 C3 dev board

### Device Description

[image]

### Hardware Configuration

ESP32-C3FH4
USB_N to GPIO18
USB_P to GPIO19
[Schematic.pdf](https://github.com/espressif/arduino-esp32/files/15344030/Schematic.pdf)

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

MAC OS Sonoma 14.4.1

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

After Updated to 2.0.16, Error occurs and says
"
A fatal error occurred: No serial data received.
Failed uploading: uploading error: exit status 2
"
[image]

It happens when I activate USB CDC On Boot function.
if I turn it Off, it is fine
I also tried Upload Speed to 115200, but it also didn't work.
I tried remove Arduino IDE and Arduino15 folder and re-installed, but it didn't work either

These are the scenarios worked for me
1. Flash dummy firmware through VS Code, Espressif IDF extension
it will allow Arduino IDE to flash code for once.
2. Or when I installed Version 2.0.14, It worked just fine.
[image]

### Sketch

```cpp
void setup() {
  Serial.begin(115200);
}

void loop() {
  ;
}
```

### Debug Message

```plain
Launching gdb-server: /Users/soodragon/Library/Arduino15/packages/esp32/tools/openocd-esp32/v0.12.0-esp32-20230921/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /private/var/folders/m3/ld9yhl7169918_sky0gbzdc40000gn/T/.arduinoIDE-unsaved2024417-6673-1v5zpmg.wsluj/sketch_may17a -f "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f board/esp32c3-builtin.cfg -d3
    Please check TERMINAL tab (gdb-server) for output from /Users/soodragon/Library/Arduino15/packages/esp32/tools/openocd-esp32/v0.12.0-esp32-20230921/bin/openocd
Finished reading symbols from objdump: Time: 6 ms
Finished reading symbols from nm: Time: 6 ms
Reading symbols from /private/var/folders/m3/ld9yhl7169918_sky0gbzdc40000gn/T/arduino/sketches/C7AE222C6875CF7EE355EEE9DBCA5A6D/sketch_may17a.ino.elf...
GNU gdb (esp-gdb) 11.2_20220823
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later 
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=aarch64-apple-darwin21.1 --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
.
Find the GDB manual and other documentation resources online at:
    .
For help, type "help".
Type "apropos word" to search for commands related to "word".
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.
0x40001de6 in ?? ()
Program stopped, probably due to a reset and/or halt issued by debugger
JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
[esp32c3] Reset cause (3) - (Software core reset)
Hardware assisted breakpoint 1 at 0x42000020: file /private/var/folders/m3/ld9yhl7169918_sky0gbzdc40000gn/T/.arduinoIDE-unsaved2024417-6673-1v5zpmg.wsluj/sketch_may17a/sketch_may17a.ino, line 2.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

spilz87, 2024-05-17:
I have the same issue. 

i still didn’t fine the solution 

any help is welcome :)

me-no-dev, 2024-05-17:
This is fixed in RC3, but for first flash you will need to manually enter download mode (hold Boot while resetting)

spilz87, 2024-05-17:
@me-no-dev thanks for your reply 

For people who use stable release 2.0.X 
It will be fixed in next one too ?

me-no-dev, 2024-05-17:
@spilz87 yes. Fix will be ported to 2.x and new 2.0.17 will probably be released

spilz87, 2024-05-17:
Nice 👍

Do you know what file(s) we have to modify to solve it in 2.0.16 release ? (If it’s not too much)

me-no-dev, 2024-05-17:
There will be a pull request with all changes. You can copy then the changes to your current install. I think only HWCDC.cpp will be modified, but some small edits might need to be done in other places too.

SooDragon, 2024-05-17:
Thank you for reply!

SooDragon, 2024-05-17:
thank you for reply😆 it’s now clear, I’ll wait for next update.
should I close this issue? or leave it for Todo - core project roadmap?

me-no-dev, 2024-05-17:
let's wait for the backport to 2.x first, then can be closed
cc @SuGlider

spilz87, 2024-05-17:
> There will be a pull request with all changes. You can copy then the changes to your current install. I think only HWCDC.cpp will be modified, but some small edits might need to be done in other places too.

I dont have this issue when I go back on v2.0.15 do I trued to get HWCDC.cpp and put it on V2.0.16 but doesn’t solve it

ColdFireHunter, 2024-05-20:
Good day,
the same problem also happens with the ESP32-S3.
With v2.0.14 it works with v2.0.16 not but with RC3 it does again.

VojtechBartoska, 2024-05-20:
@ColdFireHunter as mentioned above, it is fixed in 3.0.0-RC3 but it awaits a backport to 2.X version.

spilz87, 2024-05-20:
If there is a way to know how to quickly fix it on 2.0.x (not official way but still a way) it would help a lot. I have to unmount everything everytime I want to upload a new software 🤯

thanks in advance

SuGlider, 2024-05-20:
The 2.0.x core version shall be fixed and merged in about 2 weeks.

SuGlider, 2024-05-22:
@SooDragon - PR #9660 is merged into `release/2.x` branch. This PR solves this HW USB CDC issue.

SuGlider, 2024-05-22:
Feel free to confirm it before we close this issue.

spilz87, 2024-05-22:
> @SooDragon - PR #9660 is merged into `release/2.x` branch. This PR solves this HW USB CDC issue.

applying this merge solve the issue for me, thanks a lot

PR #9660 (2024-05-22): feat(hwcdc): Fixes HWCDC fw uploading
## Description of Change

This PR is a backport of #9628

- Fixes HW CDC uploading issue using the HWSerial/JTAG port.
- Changes SOF detection from ISR to Timer routine instead 
- Uses correct logging functions in order to separate UART from USB CDC, always closing one or the other whenever one of them is selected using `setDebugOutput(bool)` method.
- Fixes a warning about `%ld` versus `%d` in the `HWCDC_Events.ino` example. 

## Tests scenarios
Tested for ESP32-S3 and ESP32-C3 using the `HWCDC_Events.ino` example.
Tested also for ESP32-S3, ESP32-C3 and E [... cut]
closed 2024-05-23 by VojtechBartoska