espressif/arduino-esp32 #9648: ESP32-S3 OTA crashes with W5500 phy module

MeisterQ, 2024-05-17:
### Board

ESP32-S3 WROOM

### Device Description

I created a board with ESP32-S3 WROOM and W5500 Ethernet module connected via SPI

### Hardware Configuration

CS -> GPIO10
MOSI -> GPIO11
SCLK -> GPIO12
MISO -> GPIO13

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

QIO 80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

After uploading the sketch, the ESP crashes when reaching OTA settings in setup.

Sketch is BasicOTA Example combine with Basic Ethernet Sketch.

If i comment out OTA part, everything boots.

### Sketch

```cpp
void OTAsettings()
{
  // Port defaults to 3232
  // ArduinoOTA.setPort(3232);

  // Hostname defaults to esp3232-[MAC]
  // ArduinoOTA.setHostname("myesp32");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA
  .onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  })
  .onEnd([]() {
    Serial.println("\nEnd");
  })
  .onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  })
  .onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });

  ArduinoOTA.begin();
}
```

### Debug Message

```plain
assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace: 0x403777aa:0x3fcebb00 0x4037ac11:0x3fcebb20 0x403807f9:0x3fcebb40 0x4202356d:0x3fcebc70 0x4202ec69:0x3fcebca0 0x4202ecba:0x3fcebcc0 0x420232c9:0x3fcebd10 0x42008181:0x3fcebd30 0x4200820d:0x3fcebd70 0x42006c99:0x3fcebda0 0x42002eb5:0x3fcebdf0 0x42003d4b:0x3fcebe60 0x4201428a:0x3fcebeb0

ELF file SHA256: da7f8bbbf2f931a1

Rebooting...
▒ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42052fc2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-17:
Are even starting ETH? Please provide a minimal code to reproduce. In your case, you are only starting OTA, but I do not see anywhere any network to be brought up. That is probably the reason you get the error

MeisterQ, 2024-05-17:
Sorry.

`#include <SPI.h>
#include <Ethernet.h>
#include <ArduinoOTA.h>

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

void setup()
{
  Serial.begin(115200);
  ethconfig();
  OTAsettings();

}

void loop()
{
  ArduinoOTA.handle();
}

void OTAsettings()
{
  // Port defaults to 3232
  // ArduinoOTA.setPort(3232);

  // Hostname defaults to esp3232-[MAC]
  // ArduinoOTA.setHostname("myesp32");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA
  .onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  })
  .onEnd([]() {
    Serial.println("\nEnd");
  })
  .onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  })
  .onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });

  ArduinoOTA.begin();
}

void ethconfig()
{
  Ethernet.begin(mac);

  delay(2000);
  Serial.println("connecting...");
  Serial.println(Ethernet.localIP());
}`

Thats the whole code.

Debug Output after flashing:
`
ELF file SHA256: 3f80884acff2c094

Rebooting...
▒ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4203eda6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
connecting...
192.168.10.97

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace: 0x403777aa:0x3fcebb20 0x4037ac11:0x3fcebb40 0x403807f9:0x3fcebb60 0x420157e1:0x3fcebc90 0x42020edd:0x3fcebcc0 0x42020f2e:0x3fcebce0 0x4201553d:0x3fcebd30 0x42005b19:0x3fcebd50 0x42005ba5:0x3fcebd90 0x42004621:0x3fcebdc0 0x420025c5:0x3fcebe10 0x42002669:0x3fcebe80 0x420082f6:0x3fcebeb0

ELF file SHA256: 3f80884acff2c094

Rebooting...
▒ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4203eda6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
`

me-no-dev, 2024-05-17:
oh... if you are on current master as you say, you actually need this: https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_W5500_Arduino_SPI/ETH_W5500_Arduino_SPI.ino

`Ethernet.h` is external library that does not plug into our network system.

closed 2024-05-17 by me-no-dev

MeisterQ, 2024-05-17:
Thank you. But its not working.

```
C:\Users\a-pod\AppData\Local\Temp\.arduinoIDE-unsaved2024417-13008-z29tt8.ikbhp\sketch_may17a\sketch_may17a.ino: In function 'void onEvent(arduino_event_id_t, arduino_event_info_t)':
C:\Users\a-pod\AppData\Local\Temp\.arduinoIDE-unsaved2024417-13008-z29tt8.ikbhp\sketch_may17a\sketch_may17a.ino:47:134: error: no matching function for call to 'HardwareSerial::println(ETHClass&)'
     case ARDUINO_EVENT_ETH_GOT_IP: Serial.printf("ETH Got IP: '%s'\n", esp_netif_get_desc(info.got_ip.esp_netif)); Serial.println(ETH);
                                                                                                                                      ^
In file included from C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Stream.h:26,
                 from C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Arduino.h:189,
                 from C:\Users\a-pod\AppData\Local\Temp\arduino\sketches\18FFF0D5DC11F92BEF5927A68417BFFD\sketch\sketch_may17a.ino.cpp:1:
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:96:12: note: candidate: 'size_t Print::println(const __FlashStringHelper*)'
     size_t println(const __FlashStringHelper *ifsh) { return println(reinterpret_cast<const char *>(ifsh)); }
            ^~~~~~~
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:96:12: note:   no known conversion for argument 1 from 'ETHClass' to 'const __FlashStringHelper*'
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:97:12: note: candidate: 'size_t Print::println(const String&)'
     size_t println(const String &s);
            ^~~~~~~
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:97:12: note:   no known conversion for argument 1 from 'ETHClass' to 'const String&'
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:98:12: note: candidate: 'size_t Print::println(const char*)'
     size_t println(const char[]);
            ^~~~~~~
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:98:12: note:   no known conversion for argument 1 from 'ETHClass' to 'const char*'
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:99:12: note: candidate: 'size_t Print::println(char)'
     size_t println(char);
            ^~~~~~~
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:99:12: note:   no known conversion for argument 1 from 'ETHClass' to 'char'
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:100:12: note: candidate: 'size_t Print::println(unsigned char, int)'
     size_t println(unsigned char, int = DEC);
            ^~~~~~~
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:100:12: note:   no known conversion for argument 1 from 'ETHClass' to 'unsigned char'
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:101:12: note: candidate: 'size_t Print::println(int, int)'
     size_t println(int, int = DEC);
            ^~~~~~~
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:101:12: note:   no known conversion for argument 1 from 'ETHClass' to 'int'
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:102:12: note: candidate: 'size_t Print::println(unsigned int, int)'
     size_t println(unsigned int, int = DEC);
            ^~~~~~~
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:102:12: note:   no known conversion for argument 1 from 'ETHClass' to 'unsigned int'
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:103:12: note: candidate: 'size_t Print::println(long int, int)'
     size_t println(long, int = DEC);
            ^~~~~~~
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:103:12: note:   no known conversion for argument 1 from 'ETHClass' to 'long int'
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:104:12: note: candidate: 'size_t Print::println(long unsigned int, int)'
     size_t println(unsigned long, int = DEC);
            ^~~~~~~
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:104:12: note:   no known conversion for argument 1 from 'ETHClass' to 'long unsigned int'
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:105:12: note: candidate: 'size_t Print::println(long long int, int)'
     size_t println(long long, int = DEC);
            ^~~~~~~
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:105:12: note:   no known conversion for argument 1 from 'ETHClass' to 'long long int'
C:\Users\a-pod\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.16\cores\esp32/Print.h:106:12: note: candidate: 'size_t Print::println(long long unsigned int, int)'
     size_t println(unsigned long long, int = DEC);
            ^~~~~~~
[... cut ...]
C:\Users\a-pod\AppData\Local\Temp\.arduinoIDE-unsaved2024417-13008-z29tt8.ikbhp\sketch_may17a\sketch_may17a.ino:94:66: note: suggested alternative: 'ETH_PHY_MAX'
   ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI);
                                                                  ^~~~~~~~~~~
                                                                  ETH_PHY_MAX

exit status 1

Compilation error: no matching function for call to 'HardwareSerial::println(ETHClass&)'
````

My boards are on current version.
Arduino IDE is on 2.3.2

But it gives me this issues... no idea

[image]

me-no-dev, 2024-05-17:
you need to be on 3.0.0-RC3 (or master as you have stated)
Board manager link: https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json

MeisterQ, 2024-05-17:
> you need to be on 3.0.0-RC3 (or master as you have stated) Board manager link: https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json

With this one its compiling. Ive set the SPI pins to my pins, but it stops here:

```

Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c72
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
ETH Started
ETH Started
ETH Connected
``` 

I dont know if i need IRQ and RST pins. W5500 usually works without them being connected to GPIOs and my board layout got them pulled up to VCC.

Any deeper documentation about this?

me-no-dev, 2024-05-17:
it's best to use them, though we do support it without it. Looks like you did not get IP from DHCP. The rest worked. Try adding the IRQ and RST pins

MeisterQ, 2024-05-17:
> it's best to use them, though we do support it without it. Looks like you did not get IP from DHCP. The rest worked. Try adding the IRQ and RST pins

Well. The main problem here is it is wired hard to VCC.

See the schematic of my board.

[image]

If you look on most of the boards, its not connected.

Is there a way to init Ethernet without this 2 pins?

JAndrassy, 2024-05-17:
> The main problem here is it is wired hard to VCC.

that are just the pull-up resistors

MeisterQ, 2024-05-17:
> > The main problem here is it is wired hard to VCC.
> 
> that are just the pull-up resistors

Sure. But its a premade board.

Theres a way. I could solder it. But the chip is working without this 2 pins.

So.. Its a way to init it without this 2?

[image]

MeisterQ, 2024-05-18:
> it's best to use them, though we do support it without it. Looks like you did not get IP from DHCP. The rest worked. Try adding the IRQ and RST pins

Ive seen its supported only by IDF, right?

me-no-dev, 2024-05-19:
> Ive seen its supported only by IDF, right?

We do use IDF (in case of 3.0.0, it's 5.1.x)

MeisterQ, 2024-05-19:
> > Ive seen its supported only by IDF, right?
> 
> We do use IDF (in case of 3.0.0, it's 5.1.x)

Can you give me an example how i have to use it?

If i put this on top #define ETH_SPI_SUPPORTS_NO_IRQ 1 it dosnt change anything.

If i do this

  ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, -1, -1, SPI);
  ETH.begin();

it also dosnt help.

I there is like no explanation how to use it without IRQ and RST.