espressif/arduino-esp32 #9652: Kendi kartımın pin.h dosyasını nasıl eklerim buraya

hsyngndz85, 2024-05-18:
### Related area

Kendi kartımı. Pin h dosyasını ekletmek istiyorum

### Hardware specification

Kartımın dosyalarını nasıl eklerim buraya

### Is your feature request related to a problem?

Kartımın dosyalarını nasıl eklerim

### Describe the solution you'd like

Yeni kart dosyaları nasıl ekelerim

### Describe alternatives you've considered

Yeni kart ekleme

### Additional context

Yenı kart ekleme

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

hsyngndz85, 2024-05-18:
Ayrıca kütüphane ve pin.h dosyasında değişiklik yapınca nasıl ekleyebilirim buraya

me-no-dev, 2024-05-18:
we support communication only in English. Please re-submit in a language that we can understand

closed 2024-05-18 by me-no-dev