espressif/arduino-esp32 #9654: fatal error: bits/c++config.h: No such file or directory

Song-aff, 2024-05-19:
### Board

ESP32C3 Dev Module

### Device Description

I think this is not related to the device, but rather an environmental issue

### Hardware Configuration

none

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows11

### Flash frequency

None

### PSRAM enabled

yes

### Upload speed

None

### Description

\appdata\local\arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\include\c++\8.4.0\cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

**I have downloaded multiple versions of xtensa-esp32 elf, but none of them can find the c++config.h file**

### Sketch

```cpp
#define LED1_BUILTIN 12
#define LED2_BUILTIN 13
int LED=1;
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED1_BUILTIN, OUTPUT);
  pinMode(LED2_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED1_BUILTIN, LED);   
  digitalWrite(LED2_BUILTIN, !LED); 
  delay(100);
  LED = !LED;
}
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-05-19:
Installation of latest master was not successful.

Song-aff, 2024-05-19:
> Installation of latest master was not successful.
I installed esp32 by espressf Systems in the Board manager of Arduino IDE. I only tried versions 2.0.16 and 2.0.10, but I feel that the other versions are the same.

I have downloaded multiple versions of xtensa-esp32 elf, but none of them can find the c++config.h file

me-no-dev, 2024-05-19:
we usually get this when the username on Windows is too long (there is a path limit that get's overflowed with some toolchains). We have done our best for v3.0.0 to resolve this by shortening the RV toolchain package name, but very long username can still trigger it

Song-aff, 2024-05-20:
> we usually get this when the username on Windows is too long (there is a path limit that get's overflowed with some toolchains). We have done our best for v3.0.0 to resolve this by shortening the RV toolchain package name, but very long username can still trigger it

you are right.

closed 2024-05-20 by Song-aff

rhousedorf, 2024-10-31:
So How would I go about fixing this?
Do I need to get xtensa-esp32 elf myself and compile locally?
if so, which repo?

This is a missing file from 
https://github.com/SolderedElectronics/Inkplate-Arduino-library

I'm not a C++ person, .NET, using this for Arduino.
I'll also comment in the inkplate repo.

Can I just create this file as an empty file?
I tried one, and there are, of course, more missing probably also due to the path length issue you mention.