espressif/arduino-esp32 #9655: ETH.end() crashes

JAndrassy, 2024-05-19:
### Board

ESP32 Dev Module

### Device Description

Espressif Dev module

### Hardware Configuration

W5500 wired to default SPI pins 

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Linix Mint

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

ETH.end() crashes

### Sketch

```cpp
#define ETH_PHY_TYPE        ETH_PHY_W5500
#define ETH_PHY_ADDR        1
#define ETH_PHY_CS          5
#define ETH_PHY_IRQ         26
#define ETH_PHY_RST         -1
#define ETH_PHY_SPI         SPI

#include <ETH.h>

void setup() {

  Serial.begin(115200);
  delay(500);

  SPI.begin();

  ETH.begin();
  while (!ETH.hasIP()) {
    Serial.print('.');
    delay(1000);
  }

  ETH.end();

  Serial.println("END");
}

void loop() {
  delay(1);
}
```

### Debug Message

```plain
CORRUPT HEAP: Bad tail at 0x3ffb9a00. Expected 0xbaad5678 got 0x00000000
assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)
```

stack trace:
```
0x400827ad: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:466
0x40088709: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x4008d99a: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:81
0x4008cb25: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:276 (discriminator 1)
0x40083792: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:388
0x4008d9d9: free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c:39
0x4011dddd: post_instance_delete at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_event/esp_event.c:439
 (inlined by) esp_event_loop_run at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_event/esp_event.c:615
0x4011de40: esp_event_loop_run_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_event/esp_event.c:107 (discriminator 15)
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

sirapol, 2024-05-19:
I same as this.
I using WT32-ETH01

If reset, It can not obtain IP from router.

Serial display
`05:38:19.583 -> ETH Connected
05:38:21.581 -> ETH Disconnected
05:38:27.580 -> ETH Connected
05:38:29.567 -> ETH Disconnected
05:38:37.564 -> ETH Connected
05:38:39.558 -> ETH Disconnected
05:38:47.560 -> ETH Connected
05:38:49.587 -> ETH Disconnected
05:38:57.567 -> ETH Connected
05:38:59.559 -> ETH Disconnected
05:39:07.567 -> ETH Connected
05:39:09.584 -> ETH Disconnected
05:39:17.557 -> ETH Connected
05:39:19.577 -> ETH Disconnected
05:39:27.563 -> ETH Connected
05:39:29.581 -> ETH Disconnected
05:39:37.585 -> ETH Connected
05:39:39.569 -> ETH Disconnected
05:39:47.564 -> ETH Connected
05:39:49.560 -> ETH Disconnected
05:39:57.583 -> ETH Connected
05:39:59.570 -> ETH Disconnected
05:40:07.573 -> ETH Connected
05:40:09.570 -> ETH Disconnected
05:40:17.565 -> ETH Connected
05:40:19.584 -> ETH Disconnected
`

I try ETH.end() and ETH.begin(). It error same.
How to solve ?

JAndrassy, 2024-05-20:
@sirapol please do not hijack my issue

sirapol, 2024-05-20:
> @sirapol please do not hijack my issue. your problem is different.

Ok sorry

closed 2024-05-22 by me-no-dev

me-no-dev, 2024-05-22:
@JAndrassy please open separate issue for that. ETH is different than WiFi, so it might need different approach.

me-no-dev, 2024-05-22:
Also in what case would it make sense to restart the network and keep the same settings, as opposed to just leave it as-is and unplug and plug wherever is necessary

TD-er, 2024-05-22:
One example when one might want to call `ETH.end()` and later start over with exactly the same settings:

Ethernet does consume quite a lot of energy.
Typically 50 mA for the Ethernet chip alone and another 50 mA when the link is 'up'.

So it does make sense to power the chip down when it gets unplugged and periodically do a quick check to see if it is plugged in again.

me-no-dev, 2024-05-23:
Because of our network model, implementing this feature will require specific workarounds, that I do not think warrant this 1 in a million use case. You have to be on ETH cable, without PoE or wall power, decide to not just pull RST of ETH and go to deep sleep and also be too bothered to add one line. BTW if you really shut down all WiFi it should also lose the address, because both WiFi netifs will be destroyed. IMHO its not worth it to implement this instead of calling `config()` after `begin()`

TD-er, 2024-05-23:
If you can simply restore it all with a single line of code then I totally agree with not fixing it.

Just for the sake of argument, the use case I mentioned is a real use case where there is a 4g router in the field with an ESP connected to it. If the 4g signal is no longer present, the router is configured to turn off the network LAN ports.
Still this is specific enough to put in some extra effort to make sure the ETH device is configured correctly after the port is 'up' again.