espressif/arduino-esp32 #9662: Reboot loop upon using Serial0

zul00, 2024-05-21:
### Board

Arduino Nano ESP32

### Device Description

Default devkit from Arduino

### Hardware Configuration

The UART0 bus is connected to an FTDI serial adapter

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Ubuntu 22.04

### Flash frequency

240Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

My platform was on espressif32@6.7.0 which uses arduino@2.0.16.
When using the setup above I got a reboot loop upon using Serial0, I expect there is no reboot loop.

Below is the message that was captured from UART0 port
```bash
22:42:31.537 > ESP-ROM:esp32s3-20210327
22:42:31.887 > Build:Mar 27 2021
22:42:31.887 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
22:42:31.887 > Saved PC:0x4202a9b2
22:42:31.887 > SPIWP:0xee
22:42:31.887 > mode:DIO, clock div:1
22:42:31.887 > load:0x3fce3808,len:0x44c
22:42:31.887 > load:0x403c9700,len:0xbe4
22:42:31.887 > load:0x403cc700,len:0x2a68
22:42:31.887 > entry 0x403c98d4
22:42:32.849 > ESP-ROM:esp32s3-20210327
22:42:33.205 > Build:Mar 27 2021
22:42:33.205 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
22:42:33.205 > Saved PC:0x4037ef97
22:42:33.205 > SPIWP:0xee
22:42:33.205 > mode:DIO, clock div:1
22:42:33.205 > load:0x3fce3808,len:0x44c
22:42:33.205 > load:0x403c9700,len:0xbe4
22:42:33.205 > load:0x403cc700,len:0x2a68
22:42:33.205 > entry 0x403c98d4
22:42:34.164 > ESP-ROM:esp32s3-20210327
22:42:34.515 > Build:Mar 27 2021
22:42:34.515 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
22:42:34.515 > Saved PC:0x4037efb4
22:42:34.515 > SPIWP:0xee
22:42:34.515 > mode:DIO, clock div:1
22:42:34.515 > load:0x3fce3808,len:0x44c
22:42:34.515 > load:0x403c9700,len:0xbe4
22:42:34.515 > load:0x403cc700,len:0x2a68
22:42:34.515 > entry 0x403c98d4
```

However, if I downgraded the platform to espressif32@6.6.0 which uses arduino@2.0.14, it works as expected.

```bash
22:44:07.107 > [  1053][E][main.cpp:11] loop(): Hi
22:44:07.107 >
22:44:08.097 > [  2053][E][main.cpp:11] loop(): Hi
22:44:08.113 >
22:44:09.103 > [  3053][E][main.cpp:11] loop(): Hi
22:44:09.103 >
22:44:10.109 > [  4053][E][main.cpp:11] loop(): Hi
22:44:10.109 >
22:44:11.099 > [  5053][E][main.cpp:11] loop(): Hi
22:44:11.115 >
22:44:12.105 > [  6053][E][main.cpp:11] loop(): Hi
```

I am new to both Arduino and PlatformIO, so let me know if I posted in the correct repository.

Thanks,
Zul

### Sketch

```cpp
Sketch

#include <Arduino.h>

void setup() {
    Serial0.begin(115200);
}

void loop() {
    log_e("Hi\r\n");

    delay(1000);  // Delay to throttle loop execution, adjust as necessary
}

PlatformIO configuration
```ini
[env:arduino_nano_esp32]
; NOTE: We got an issue with Serial0 on v6.7.0
; espressif v6.7.0 uses arduino v2.0.16; idf v4.4.7
; device went into reboot loop
platform = espressif32@6.7.0
framework = arduino
board = arduino_nano_esp32

```
```

### Debug Message

```plain
22:42:31.537 > ESP-ROM:esp32s3-20210327
22:42:31.887 > Build:Mar 27 2021
22:42:31.887 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
22:42:31.887 > Saved PC:0x4202a9b2
22:42:31.887 > SPIWP:0xee
22:42:31.887 > mode:DIO, clock div:1
22:42:31.887 > load:0x3fce3808,len:0x44c
22:42:31.887 > load:0x403c9700,len:0xbe4
22:42:31.887 > load:0x403cc700,len:0x2a68
22:42:31.887 > entry 0x403c98d4
22:42:32.849 > ESP-ROM:esp32s3-20210327
22:42:33.205 > Build:Mar 27 2021
22:42:33.205 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
22:42:33.205 > Saved PC:0x4037ef97
22:42:33.205 > SPIWP:0xee
22:42:33.205 > mode:DIO, clock div:1
22:42:33.205 > load:0x3fce3808,len:0x44c
22:42:33.205 > load:0x403c9700,len:0xbe4
22:42:33.205 > load:0x403cc700,len:0x2a68
22:42:33.205 > entry 0x403c98d4
22:42:34.164 > ESP-ROM:esp32s3-20210327
22:42:34.515 > Build:Mar 27 2021
22:42:34.515 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
22:42:34.515 > Saved PC:0x4037efb4
22:42:34.515 > SPIWP:0xee
22:42:34.515 > mode:DIO, clock div:1
22:42:34.515 > load:0x3fce3808,len:0x44c
22:42:34.515 > load:0x403c9700,len:0xbe4
22:42:34.515 > load:0x403cc700,len:0x2a68
22:42:34.515 > entry 0x403c98d4
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

zul00, 2024-05-22:
I forgot to mention that there is no problem on `espressif32@6.7.0`  if I use `USBCDC` as the streamer, i.e. `Serial`

lbernstone, 2024-05-22:
Do you have this issue if you use the vanilla esp32 variant rather than the Nano?

zul00, 2024-05-22:
I have an access to an ESP32-S3-EYE, I will test it later

zul00, 2024-05-23:
I tested the configuration above, but I still got a reboot loop with a report on UART0 is not installed. Below is the snapshot

```bash
10:40:19.139 > ESP-ROM:esp32s3-20210327
10:40:19.442 > Build:Mar 27 2021
10:40:19.442 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
10:40:19.474 > Saved PC:0x420248d6
10:40:19.474 > SPIWP:0xee
10:40:19.474 > mode:DIO, clock div:1
10:40:19.474 > load:0x3fce3808,len:0x44c
10:40:19.474 > load:0x403c9700,len:0xbe4
10:40:19.474 > load:0x403cc700,len:0x2a68
10:40:19.474 > entry 0x403c98d4
10:40:19.561 > [    92][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 46 - Using Channel 7, Value = 183
10:40:19.561 > [   100][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 0 - Using Channel 6, Value = 255
10:40:19.561 > [   108][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 45 - Using Channel 5, Value = 71
...
10:40:22.444 > [  3006][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 45 - Using Channel 5, Value = 255
10:40:22.495 > [  3036][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
10:40:22.495 > [  3046][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(115) txPin(0)
10:40:22.495 > [  3055][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
10:40:22.495 > ESP-ROM:esp32s3-20210327
10:40:22.828 > Build:Mar 27 2021
10:40:22.828 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
10:40:22.828 > Saved PC:0x420248d6
10:40:22.828 > SPIWP:0xee
10:40:22.828 > mode:DIO, clock div:1
10:40:22.828 > load:0x3fce3808,len:0x44c
10:40:22.828 > load:0x403c9700,len:0xbe4
10:40:22.828 > load:0x403cc700,len:0x2a68
10:40:22.828 > entry 0x403c98d4
```

What is weird is that it mentioned about initializing the uart0 on an incorrect pin.

I adjusted the setup to
```cpp
void setup() {
    Serial0.begin(115200, SERIAL_8N1, SOC_RX0, SOC_TX0); // or
    //     Serial0.begin(115200, SERIAL_8N1, 44, 43);
}
```

But the result is still the same. I am guessing that is happening before it went into the `setup` from the sketch, right?

SuGlider, 2024-05-23:
@zul00 - Arduino ESP32 Nano board has an exclusive [pin remapping scheme]
(https://docs.arduino.cc/resources/pinouts/ABX00083-full-pinout.pdf)

Please use the pins as defined in the pinout pdf file or/and in this header file:
https://github.com/espressif/arduino-esp32/blob/master/variants/arduino_nano_nora/pins_arduino.h

``` cpp
void setup() {
    Serial0.begin(115200, SERIAL_8N1, D0, D1); // or
    //  Serial0.begin(115200, SERIAL_8N1, RX, TX);  // this also works for the ESP32 Nano
    //     Serial0.begin(115200, SERIAL_8N1, 44, 43);  <-- remapped
}
```

zul00, 2024-05-24:
Thanks @SuGlider !

Now I understand that there is a remap there. So this one also works :)

```cpp
    Serial0.begin(115200, SERIAL_8N1, D0, 1);
```

zul00, 2024-05-24:
Should there be a fix? Maybe HardwareSerial can have an `#ifndef` for `SOC_*? or `#ifdef` for `BOARD_HAS_PIN_REMAP`?

SuGlider, 2024-05-24:
There are a few ways to fix it.

The `SOC_*` pins could be defined in the `pins_arduino.h` ESP32 Nano board file.

Or the mapping function shall check if the requested pin does exist and return `-1`, which means that it shouldn't be changed.

The main problem is that the mapping function defined in the variant folder of this board returns a value from an array, but the array max number of items is shorter than 43 or 44 (Rx/Tx real pins), causing it to return trash values for the "pins" out of the array space.

VojtechBartoska, 2024-06-03:
Looks this ticket was already answered, closing. If needed, you can reopen it.

closed 2024-06-03 by VojtechBartoska