espressif/arduino-esp32 #9668: ESP32-4848S040 Not supported in 3.0.x

jesetek, 2024-05-22:
### Board

ESP32S3 Dev Module

### Device Description

ESP32-4848S040
Driver: ST7701

### Hardware Configuration

4 inch display with capacitive touch

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Win10

### Flash frequency

80Hz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I am trying to run ESP32-4848S040 display with the latest 3.0.0-rc3, but I keep getting the same error.
I am using Arduino_GFX_Library.

If I use the latest release (2.0.16), the issue is gone, but I don't like changing between versions all the time.

### Sketch

```cpp
#include <Arduino_GFX_Library.h>
#include "lv_test.h"

#define GFX_BL 38

Arduino_DataBus *bus = new Arduino_SWSPI(
  GFX_NOT_DEFINED /* DC */, 39 /* CS */, 48 /* SCK */, 47 /* MOSI */, GFX_NOT_DEFINED /* MISO */);

// Uncomment for 4" rect display
Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
  18 /* DE */, 17 /* VSYNC */, 16 /* HSYNC */, 21 /* PCLK */,
  /*R*/ 11 /* R0 */, 12 /* R1 */, 13 /* R2 */, 14 /* R3 */, 0 /* R4 */,
  /*G*/ 8 /* G0 */, 20 /* G1 */, 3 /* G2 */, 46 /* G3 */, 9 /* G4 */, 10 /* G5 */,
  /*B*/ 4 /* B0 */, 5 /* B1 */, 6 /* B2 */, 7 /* B3 */, 15 /* B4 */,

  1 /* hsync_polarity */, 10 /* hsync_front_porch */, 8 /* hsync_pulse_width */, 50 /* hsync_back_porch */,
  1 /* vsync_polarity */, 10 /* vsync_front_porch */, 8 /* vsync_pulse_width */, 20 /* vsync_back_porch */);
Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
  480 /* width */, 480 /* height */, rgbpanel, 0 /* rotation */, true /* auto_flush */,
  bus, GFX_NOT_DEFINED /* RST */, st7701_type8_init_operations, sizeof(st7701_type8_init_operations));

/*******************************************************************************
 * Please config the touch panel in touch.h
 ******************************************************************************/
#include "touch.h"

/* Change to your screen resolution */
static uint32_t screenWidth;
static uint32_t screenHeight;
static lv_disp_draw_buf_t draw_buf;
static lv_color_t *disp_draw_buf;
static lv_disp_drv_t disp_drv;

/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p) {
  uint32_t w = (area->x2 - area->x1 + 1);
  uint32_t h = (area->y2 - area->y1 + 1);

#if (LV_COLOR_16_SWAP != 0)
  gfx->draw16bitBeRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
#else
  gfx->draw16bitRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
#endif

  lv_disp_flush_ready(disp);
}

void my_touchpad_read(lv_indev_drv_t *indev_driver, lv_indev_data_t *data) {
  if (touch_has_signal()) {
    if (touch_touched()) {
      data->state = LV_INDEV_STATE_PR;

      /*Set the coordinates*/
      data->point.x = touch_last_x;
      data->point.y = touch_last_y;
    } else if (touch_released()) {
      data->state = LV_INDEV_STATE_REL;
    }
  } else {
    data->state = LV_INDEV_STATE_REL;
  }
}

void setup() {
  Serial.begin(115200);
  // while (!Serial);
  Serial.println("LVGL Widgets Demo");

  delay(1000);

  // Init touch device
  touch_init();

  // Init Display
  //gfx->begin();

  gfx->begin(16000000); /* specify data bus speed */

  gfx->fillScreen(BLACK);

#ifdef GFX_BL
  pinMode(GFX_BL, OUTPUT);
  digitalWrite(GFX_BL, HIGH);
#endif

  lv_init();

  screenWidth = gfx->width();
  screenHeight = gfx->height();
#ifdef ESP32
  // Allocate buffer with some diagnostics
  size_t buf_size = sizeof(lv_color_t) * screenWidth * 200;
  disp_draw_buf = (lv_color_t *)heap_caps_malloc(buf_size, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
#else
  disp_draw_buf = (lv_color_t *)malloc(sizeof(lv_color_t) * screenWidth * 200);
#endif
  if (!disp_draw_buf) {
    Serial.println("LVGL disp_draw_buf allocate failed!");
  } else {
    lv_disp_draw_buf_init(&draw_buf, disp_draw_buf, NULL, screenWidth * 200);

    /* Initialize the display */
    lv_disp_drv_init(&disp_drv);
    /* Change the following line to your display resolution */
    disp_drv.hor_res = screenWidth;
    disp_drv.ver_res = screenHeight;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.draw_buf = &draw_buf;
    lv_disp_drv_register(&disp_drv);

    /* Initialize the (dummy) input device driver */
    static lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_touchpad_read;
    lv_indev_drv_register(&indev_drv);

    lv_test();

    Serial.println("Setup done");
  }
}

void loop() {
  lv_timer_handler(); /* let the GUI do its work */
  delay(5);
}
```

### Debug Message

```plain
J:\RAMEK\ESP\4.0inch_ESP32-4848S040\AndrazTest\AndrazTest.ino:10:1: error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?
   10 | Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(
      | ^~~~~~~~~~~~~~~~~~~~~
      | Arduino_ESP32S2PAR8Q
J:\RAMEK\ESP\4.0inch_ESP32-4848S040\AndrazTest\AndrazTest.ino:18:1: error: 'Arduino_RGB_Display' does not name a type
   18 | Arduino_RGB_Display *gfx = new Arduino_RGB_Display(
      | ^~~~~~~~~~~~~~~~~~~
In file included from J:\RAMEK\ESP\4.0inch_ESP32-4848S040\AndrazTest\AndrazTest.ino:26:
J:\RAMEK\ESP\4.0inch_ESP32-4848S040\AndrazTest\touch.h: In function 'bool touch_touched()':
J:\RAMEK\ESP\4.0inch_ESP32-4848S040\AndrazTest\touch.h:164:71: error: 'gfx' was not declared in this scope
  164 |     touch_last_x = map(ts.points[0].x, TOUCH_MAP_X1, TOUCH_MAP_X2, 0, gfx->width() - 1);
      |                                                                       ^~~
J:\RAMEK\ESP\4.0inch_ESP32-4848S040\AndrazTest\AndrazTest.ino: In function 'void my_disp_flush(lv_disp_drv_t*, const lv_area_t*, lv_color_t*)':
J:\RAMEK\ESP\4.0inch_ESP32-4848S040\AndrazTest\AndrazTest.ino:52:3: error: 'gfx' was not declared in this scope
   52 |   gfx->draw16bitRGBBitmap(area->x1, area->y1, (uint16_t *)&color_p->full, w, h);
      |   ^~~
J:\RAMEK\ESP\4.0inch_ESP32-4848S040\AndrazTest\AndrazTest.ino: In function 'void setup()':
J:\RAMEK\ESP\4.0inch_ESP32-4848S040\AndrazTest\AndrazTest.ino:87:3: error: 'gfx' was not declared in this scope
   87 |   gfx->begin(16000000); /* specify data bus speed */
      |   ^~~

exit status 1

Compilation error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-05-22:
The used Arduino_GFX_Library is not espressif32 Arduino core 3.0.0 compatible.
You should ask the maintainer of the Arduino_GFX_Library to update to make it work with core 3.0.0

jesetek, 2024-05-22:
He told me that 3.0.0 version does not yet support RGBPanel and to wait for 3.0.0 version to update and start supporting it

Jason2866, 2024-05-22:
Why dont you write exactly this in the headline? `Feature Request: Support for RGB LC Displays`

jesetek, 2024-05-22:
Because I wanted to solve this myself, but I need some advice in which files to look since I couldn't find anything

TigoTas, 2024-05-28:
So...

No workarounds other than waiting for either Espressif or @moononournation?

Jason2866, 2024-05-28:
@TigoTas  espressif does not have a task here. You can use the mentioned Display with Arduino. Just not with the mentioned 3rd party library.

VojtechBartoska, 2024-06-03:
As mentioned above, here is not much we can do, closing the ticket. Thanks for understanding.

closed 2024-06-03 by VojtechBartoska