espressif/arduino-esp32 #9673: ETH static IP configuration doesn't work

JAndrassy, 2024-05-23:
### Board

ESP32 Dev module

### Device Description

Espressif dev module

### Hardware Configuration

W5500 wired on default SPI pins

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Linux 

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

any

### Description

ETH.config doesn't set static IP configuration, because of 
```
  if (_esp_netif == NULL) {
    return false;
  }
```
in `NetworkInterface.config`

### Sketch

```cpp
#define ETH_PHY_TYPE        ETH_PHY_W5500
#define ETH_PHY_ADDR        -1
#define ETH_PHY_CS           5
#define ETH_PHY_IRQ         -1
#define ETH_PHY_RST         -1
#define ETH_PHY_SPI         SPI

#include <ETH.h>

void setup() {

  Serial.begin(115200);
  delay(500);

  SPI.begin();

  IPAddress ip(192, 168, 1, 177);
  IPAddress gw(192, 168, 1, 1);
  IPAddress dns(192, 168, 1, 1);
  IPAddress mask(255, 255, 255, 0);

  ETH.config(ip, dns, gw, mask);
  ETH.begin();
  while (!ETH.hasIP()) {
    Serial.print('.');
    delay(1000);
  }
  if (ip != ETH.localIP()) {
    Serial.println("ERROR: Static IP was not used.");
  }

 }

void loop() {

}
```

### Debug Message

```plain
ERROR: Static IP was not used.
```

### Other Steps to Reproduce

https://github.com/espressif/arduino-esp32/issues/9435

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JAndrassy, 2024-05-23:
https://github.com/espressif/arduino-esp32/issues/9435

me-no-dev, 2024-05-23:
AFTER begin ;) there is nothing to set the address to. WiFi works differently in this regard

closed 2024-05-23 by me-no-dev

JAndrassy, 2024-05-26:
so static IP can be used only if begin() with DHCP is executed first?

JAndrassy, 2024-05-26:
it doesn't work
```
#define ETH_PHY_TYPE        ETH_PHY_W5500
#define ETH_PHY_ADDR        -1
#define ETH_PHY_CS           5
#define ETH_PHY_IRQ         -1
#define ETH_PHY_RST         -1
#define ETH_PHY_SPI         SPI

#include <ETH.h>

void setup() {

  Serial.begin(115200);
  delay(500);

  SPI.begin();

  IPAddress ip(192, 168, 1, 177);
  IPAddress gw(192, 168, 1, 1);
  IPAddress dns(192, 168, 1, 1);
  IPAddress mask(255, 255, 255, 0);

  ETH.begin();
  ETH.config(ip, dns, gw, mask);

  Serial.print("Attempt to connect to port 80 on ");
  Serial.println(gw);
  NetworkClient client;
  if (client.connect(gw, 80)) {
    Serial.println("\t...success");
  } else {
    Serial.println("\t...ERROR");
  }
  client.stop();
  Serial.println();

 }

void loop() {
}
```
```
Attempt to connect to port 80 on 192.168.1.1
[   629][E][NetworkClient.cpp:242] connect(): connect on fd 48, errno: 118, "Host is unreachable"
	...ERROR
```

me-no-dev, 2024-05-26:
Wrong order of arguments: https://github.com/espressif/arduino-esp32/blob/master/libraries/Network/src/NetworkInterface.h#L39

JAndrassy, 2024-05-26:
sorry I don't know from where did I copy that line into the minimal test, but I had the error in the big test sketch so fixing the ordering doesn't help.
```
IP Address: 192.168.1.177
gateway IP Address: 192.168.1.1
subnet IP mask: 255.255.255.0

Attempt to connect to port 80 on 192.168.1.1
[  9618][E][NetworkClient.cpp:242] connect(): connect on fd 48, errno: 118, "Host is unreachable"
	...ERROR
```

me-no-dev, 2024-05-27:
```cpp
  IPAddress ip(192, 168, 254, 8);
  IPAddress gw(192, 168, 254, 1);
  IPAddress mask(255, 255, 255, 0);
  IPAddress dns(192, 168, 254, 3);
  SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
  ETH.begin(ETH_TYPE, ETH_ADDR, ETH_CS, ETH_IRQ, ETH_RST, SPI);
  ETH.config(ip, gw, mask, dns);
```

```
============
Network Info
============
*eth0: <UP,100M,FULL_DUPLEX,AUTO,ADDR:0x1> (DHCPC_OFF,GARP,IP_MOD)
      ether 32:C6:F7:F4:2F:8C
      inet 192.168.254.8 netmask 255.255.255.0 broadcast 192.168.254.255
      gateway 192.168.254.1 dns 192.168.254.3

============
[  3230][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
[ 10236][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'google.com' with error '-54'
[ 13246][I][NetworkClient.cpp:253] connect(): select returned due to timeout 3000 ms for fd 48
Connection Failed
[ 33269][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 142.250.187.142
Connection Success
```

IP was correctly assigned and was able to connect to google, though the first attempt failed with DNS error, but second attempt was success. 
Can you try a second connect?

JAndrassy, 2024-05-27:
I found this https://github.com/espressif/arduino-esp32/issues/5733 in ETH.cpp mentioned at the end of `begin` for internal MAC

JAndrassy, 2024-05-27:
this works
```
  ETH.begin();
  ETH.config(ip, gw, mask, dns);
  while (!ETH.linkUp()) {
    Serial.print('.');
    delay(1000);
  }
  Serial.println();
```

me-no-dev, 2024-05-27:
waiting in libs should be set to minimum. I'll try to find another solution

kapyaar, 2025-07-22:
@me-no-dev Has there been any updates on this? better solutions?

JAndrassy, 2025-07-23:
@kapyaar I recommend my EthernetESP32 library