espressif/arduino-esp32 #9674: ETH.end clears static IP config

JAndrassy, 2024-05-23:
### Board

Esp32 Dev module

### Device Description

Espressif Dev module

### Hardware Configuration

Wiz850io  wired on default SPI pins

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Linux

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

any

### Description

In other Ethernet libraries with 'config` method (esp8266 and RP2040 core), `end' doesn't clear the static IP configuration set by 'config' and In WiFi libraries `disconnect` doesn't clear the static IP configuration.

### Sketch

```cpp
#define ETH_PHY_TYPE        ETH_PHY_W5500
#define ETH_PHY_ADDR        -1
#define ETH_PHY_CS           5
#define ETH_PHY_IRQ         -1
#define ETH_PHY_RST         -1
#define ETH_PHY_SPI         SPI

#include <ETH.h>

void setup() {

  Serial.begin(115200);
  delay(500);

  SPI.begin();

  // first begin with DHCP as workaround for issue #9673
  ETH.begin();
  while (!ETH.hasIP()) {
    Serial.print('.');
    delay(1000);
  }

  IPAddress ip(192, 168, 1, 177);
  IPAddress gw(192, 168, 1, 1);
  IPAddress dns(192, 168, 1, 1);
  IPAddress mask(255, 255, 255, 0);

  ETH.config(ip, dns, gw, mask);
  ETH.begin();
  while (!ETH.hasIP()) {
    Serial.print('.');
    delay(1000);
  }
  if (ip != ETH.localIP()) {
    Serial.println("ERROR: Static IP was not used.");
  }

  ETH.end();

  ETH.begin();
  while (!ETH.hasIP()) {
    Serial.print('.');
    delay(1000);
  }
  if (ip != ETH.localIP()) {
    Serial.println("ERROR: Static IP was not remembered.");
  }

 }

void loop() {

}
```

### Debug Message

```plain
ERROR: Static IP was not remembered.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-23:
I really think it's up to the application to decide what to store and what not to store. In majority of cases, the app already has the IPs stored, so why not just call `config()` instead of storing class objects twice?

Jason2866, 2024-05-23:
For me it is more logical as it is currently. When doing `ETH.end();` and going to do `ETH.begin();` again there should no old config be reused.
There is a reason for calling `ETH.begin();` again, probably because settings have changed.

JAndrassy, 2024-05-23:
@Jason2866
> In other Ethernet libraries with config method (esp8266 and RP2040 core), end doesn't clear the static IP configuration set by config and In WiFi libraries disconnect doesn't clear the static IP configuration.

Jason2866, 2024-05-23:
esp32 is very different in many aspects, no need to be compatible to anything existing in the world. Such stuff bloats and increases the core unnecessary.

JAndrassy, 2024-05-23:
yes but maybe this issue will be solved with the solution for https://github.com/espressif/arduino-esp32/issues/9673

me-no-dev, 2024-05-24:
@JAndrassy begin then config makes total sense. The fact that it works the other way round for WiFi is just luck, because we can init the WiFi netifs earlier (no pin configs or anything else necessary). That is not the case for ETH, as we need to know what ETH, which pins and so on. As for saving the IP, I already talked about that. It's unnecessary complicating the network libraries and diverging ETH from the rest of the interfaces. Just call to `config` again will solve this otherwise tedious task to lug around class instances (of IPAddress) that you already store in your sketch anyway

JAndrassy, 2024-05-24:
https://github.com/espressif/arduino-esp32/issues/9673 is the important one

closed 2024-06-03 by me-no-dev