espressif/arduino-esp32 #9676: Continuous ADC does not function correctly with ADC1_CH8 and ADC1_CH9

victorfleischauer, 2024-05-23:
### Board

ESP32-S3-DevKitC-1

### Device Description

ESP32-S3-DevKitC-1 v1.0, plain module on breadboard

### Hardware Configuration

All pins left floating.
Jumper wire used only to pull up or pull down individual pins being tested.

### Version

latest development Release Candidate (RC-X)

### IDE Name

PlatformIO

### Operating System

macOS 14.4

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

460800

### Description

The ADC continuous driver does not work with channels 8 and 9 of ADC1 (GPIOs 9 and 10).

The issue seems to be with [this line](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-adc.c?plain=1#L443), "` & 0x7 `" appears to change channel 8 to channel 0 and channel 9 to 1.
By changing the line to `adc_pattern[i].channel = channel[i];` the sketch appears to function as expected.

Is this limitation intended? 

### Sketch

```cpp
#include <Arduino.h>
/**
 * ADC continuous example found in:
 * https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/adc.html
 * With the following alterations
 * - fixed spelling of adc_continuos_data_t to adc_continuous_data_t
 * - changed the pins to be used in the example from {1, 2, 3, 4} to {1, 2, 9, 10}
*/

// Define how many conversion per pin will happen and reading the data will be and average of all conversions
#define CONVERSIONS_PER_PIN 5

// Declare array of ADC pins that will be used for ADC Continuous mode - ONLY ADC1 pins are supported
// Number of selected pins can be from 1 to ALL ADC1 pins.
#ifdef CONFIG_IDF_TARGET_ESP32
uint8_t adc_pins[] = {36, 39, 34, 35}; //some of ADC1 pins for ESP32
#else
uint8_t adc_pins[] = {1, 2, 9, 10}; //ADC1 common pins for ESP32S2/S3 + ESP32C3/C6 + ESP32H2
#endif

// Calculate how many pins are declared in the array - needed as input for the setup function of ADC Continuous
uint8_t adc_pins_count = sizeof(adc_pins) / sizeof(uint8_t);

// Flag which will be set in ISR when conversion is done
volatile bool adc_coversion_done = false;

// Result structure for ADC Continuous reading
adc_continuous_data_t * result = NULL;

// ISR Function that will be triggered when ADC conversion is done
void ARDUINO_ISR_ATTR adcComplete() {
  adc_coversion_done = true;
}

void setup() {
    // Initialize serial communication at 115200 bits per second:
    Serial.begin(115200);

    // Optional for ESP32: Set the resolution to 9-12 bits (default is 12 bits)
    analogContinuousSetWidth(12);

    // Optional: Set different attenaution (default is ADC_11db)
    analogContinuousSetAtten(ADC_11db);

    // Setup ADC Continuous with following input:
    // array of pins, count of the pins, how many conversions per pin in one cycle will happen, sampling frequency, callback function
    analogContinuous(adc_pins, adc_pins_count, CONVERSIONS_PER_PIN, 20000, &adcComplete);

    // Start ADC Continuous conversions
    analogContinuousStart();
}

void loop() {
    // Check if conversion is done and try to read data
    if (adc_coversion_done == true) {
        // Set ISR flag back to false
        adc_coversion_done = false;
        // Read data from ADC
        if (analogContinuousRead(&result, 0)) {

            // Optional: Stop ADC Continuous conversions to have more time to process (print) the data
            analogContinuousStop();

            for (int i = 0; i < adc_pins_count; i++) {
            Serial.printf("\nADC PIN %d data:", result[i].pin);
            Serial.printf("\n   Avg raw value = %d", result[i].avg_read_raw);
            Serial.printf("\n   Avg milivolts value = %d", result[i].avg_read_mvolts);
            }

            // Delay for better readability of ADC data
            delay(1000);

            // Optional: If ADC was stopped, start ADC conversions and wait for callback function to set adc_coversion_done flag to true
            analogContinuousStart();
        }
        else {
            Serial.println("Error occured during reading data. Set Core Debug Level to error or lower for more informations.");
        }
    }
}
```

### Debug Message

```plain
ADC PIN 1 data:
   Avg raw value = 316
   Avg milivolts value = 277
ADC PIN 2 data:
   Avg raw value = 123
   Avg milivolts value = 109
ADC PIN 9 data:
   Avg raw value = 597132472
   Avg milivolts value = -2047405039
ADC PIN 10 data:
   Avg raw value = 301595872
   Avg milivolts value = -160639543
```

### Other Steps to Reproduce

platformio.ini:
```
[env:esp32-s3-devkitc-1]
platform = https://github.com/platformio/platform-espressif32.git
platform_packages =
	platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.0-rc3
	platformio/framework-arduinoespressif32-libs @ https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-05-24:
Probably a left over from the times when only the esp32 existed. The esp32 has 8 channels.

victorfleischauer, 2024-05-24:
> Probably a left over from the times when only the esp32 existed. The esp32 has 8 channels.

Right, both the S2 and S3 have more than eight channels, so perhaps it could be changed to a check like:
```
if(channel[i] >= SOC_ADC_MAX_CHANNEL_NUM || channel[i] < 0)
{
  log_e("Channel %d is not a valid channel for ADC UNIT 1", channel[i]);
  return ESP_FAIL;
}
adc_pattern[i].channel = channel[i];
```
?

P-R-O-C-H-Y, 2024-05-27:
@victorfleischauer Thank you for reporting the issue. Can you open a PR with the change to remove `& 0x7` from the `adc_pattern[i].channel = channel[I] & 0x7;`? The check as you commented is not needed, as the check for valid pins is done in `analogContinuous` function before calling `__analogContinuousInit`. 

I can create the PR with the fix if you prefer also, so let us know and if you open the PR, you can ping me there :) Thanks

PR #9692 (2024-05-27): fix(adc): Remove masking for ADC channel number
## Description of Change
This PR fixes ADC Continuous channel number limit for S2/S3, where there are 10 channels per ADC peripheral.
Credits to @victorfleischauer

## Tests scenarios
Tested on esp32-s3

## Related links
Closes #9676