espressif/arduino-esp32 #9678: Problem with Wifi arduino Version 5.1.4

victorMartins2024, 2024-05-24:
### Board

ESP32

### Device Description

"I'm developing a project through VS Code and I'm using Wifi.h on Arduino version 5.1.4. I'm able to flash my code, but when it comes to the Wi-Fi, my project doesn't connect. Could you please provide me with this support?" 

### Hardware Configuration

Connect I2C

### Version

latest master (checkout manually)

### IDE Name

VSCode

### Operating System

Windows 10

### Flash frequency

20Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

"I'm developing a project through VS Code and I'm using Wifi.h on Arduino version 5.1.4. I'm able to flash my code, but when it comes to the Wi-Fi, my project doesn't connect. Could you please provide me with this support?" 

I want to give flash do meu code no esp32 and automatically or esp32 connect no wifi . 

I do have to use the version 5.1.3 of arduino??

### Sketch

```cpp
// ----------------------------------------------------------------- 
// ---connection infos--
const char *ssid    =    "xxxxxxxxxxxxxxxx";              // wifi name      
const char *pass    =    "xxxxxxxxxxxxxxxx";                       // wifi password
const char *mqtt    =    "xxx.xxx.xx.xxx";                        // broker IP
const char *user    =    "xxxxxxxxx";                             // broker user      
const char *passwd  =    "xxxxxxxxx";                          // broker password     
int port  =  1883;
// -----------------------------------------------------------------

// -----------------------------------------------------------------  
// -----Topics----- 
const char *topic_user    =    "sinal/sim/rfid";
const char *topic_CAD     =    "sinal/sim/Cadastro";
const char *topic_TEC     =    "sinal/sim/manutenção";  
const char *topic_V       =    "sinal/sim/checklist/P1:";        // broker topic Vazamento           
const char *topic_G       =    "sinal/sim/checklist/P2:";        // broker topic Garfo              
const char *topic_E       =    "sinal/sim/checklist/P3:";        // broker topic Emergencia       
const char *topic_F       =    "sinal/sim/checklist/P4:";        // broker topic Frente e Ré     
const char *topic_B       =    "sinal/sim/checklist/P5:";        // broker topic Bateria, cabos, conectores
// -----------------------------------------------------------------
```

### Debug Message

```plain
PS C:\Users\engenharia.automacao\project-name> set IDF_PATH=C:\Users\engenharia.automacao\esp\v5.1.4\esp-idf
PS C:\Users\engenharia.automacao\project-name> C:\Users\engenharia.automacao\.espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:\Users\engenharia.automacao\esp\v5.1.4\esp-idf\tools\idf_monitor.py -p COM4 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 c:\Users\engenharia.automacao\project-name\build\main.elf
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM4 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7112
load:0x40078000,len:14544
load:0x40080400,len:4
0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:3812
entry 0x4008064c
I (31) boot: ESP-IDF v5.1.4 2nd stage bootloader
I (31) boot: compile time May 23 2024 14:18:46
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=28458h (164952) map
I (138) esp_image: segment 1: paddr=00038480 vaddr=3ffb0000 size=03d60h ( 15712) load
I (142) esp_image: segment 2: paddr=0003c1e8 vaddr=40080000 size=03e30h ( 15920) load
I (148) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=9eb24h (650020) map
I (286) esp_image: segment 4: paddr=000deb4c vaddr=40083e30 size=13904h ( 80132) load
I (319) boot: Loaded app from partition at offset 0x10000
I (319) boot: Disabling RNG early entropy source...
I (331) cpu_start: Multicore app
I (331) cpu_start: Pro cpu up.
I (331) cpu_start: Starting app cpu, entry point is 0x400814a8
0x400814a8: call_start_cpu1 at C:/Users/engenharia.automacao/esp/v5.1.4/esp-idf/components/esp_system/port/cpu_start.c:159

I (0) cpu_start: App cpu up.
I (349) cpu_start: Pro cpu start user code
I (349) cpu_start: cpu freq: 160000000 Hz
I (349) cpu_start: Application information:
I (354) cpu_start: Project name:     main
I (358) cpu_start: App version:      1
I (363) cpu_start: Compile time:     May 23 2024 14:18:02
I (369) cpu_start: ELF file SHA256:  4e73942ccd6bca54...
I (375) cpu_start: ESP-IDF:          v5.1.4
I (380) cpu_start: Min chip rev:     v0.0
I (385) cpu_start: Max chip rev:     v3.99 
I (389) cpu_start: Chip rev:         v3.1
I (394) heap_init: Initializing. RAM available for dynamic allocation:
I (401) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (407) heap_init: At 3FFB8F30 len 000270D0 (156 KiB): DRAM
I (414) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (420) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (426) heap_init: At 40097734 len 000088CC (34 KiB): IRAM
I (434) spi_flash: detected chip: generic
I (437) spi_flash: flash io: dio
I (443) app_start: Starting scheduler on CPU0
I (446) app_start: Starting scheduler on CPU1
I (446) main_taskI (465) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (469) wifi:wifi osi_nvs_open fail ret=4353

W (470) wifi_init: Failed to unregister Rx callbacks
E (473) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
E (479) wifi_init: Failed to deinit Wi-Fi (0x3001)
[    52][E][WiFiGeneric.cpp:248] wifiLowLevelInit(): esp_wifi_init 4353
[    70][E][STA.cpp:298] begin(): STA enable failed!
I (507) gpio: GPIO[18]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (508) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (515) gpio: GPIO[23]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (525) gpio: GPIO[5]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (534) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
[  1236][E][Preferences.cpp:47] begin(): nvs_open failed: NOT_INITIALIZED
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2024-05-24:
Espressif Arduino framework has the complete wifi lib included. Do NOT use any other one.

VojtechBartoska, 2024-06-03:
Closing due to no response, if needed you can reopen.

closed 2024-06-03 by VojtechBartoska