espressif/arduino-esp32 #9685: `ETH.end()` doesn't allow `begin` again if IDF SPI host is used

JAndrassy, 2024-05-26:
### Board

ESP32 Dev Module

### Device Description

Espressif Dev module

### Hardware Configuration

Wiz850io wired to default SPI pins

### Version

latest master (checkout manually)

### IDE Name

Sloeber

### Operating System

Linux Mint

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

921600

### Description

when IDF SPI host is used for ETH, new `ETH.begin` fails if `ETH.end` was invoked before.

### Sketch

```cpp
#define ETH_PHY_TYPE        ETH_PHY_W5500
#define ETH_PHY_ADDR        1
#define ETH_PHY_CS          5
#define ETH_PHY_IRQ         -1
#define ETH_PHY_RST         -1

#define ETH_PHY_SPI_HOST    SPI2_HOST
#define ETH_PHY_SPI_SCK     18
#define ETH_PHY_SPI_MISO    19
#define ETH_PHY_SPI_MOSI    23

#include <ETH.h>

void setup() {

  Serial.begin(115200);
  delay(500);

  ETH.begin();
  while (!ETH.hasIP()) {
    Serial.print('.');
    delay(1000);
  }

  Serial.println();
  Serial.println("end");
  ETH.end();

  ETH.begin();
  while (!ETH.hasIP()) {
    Serial.print('.');
    delay(1000);
  }
}

void loop() {
}
```

### Debug Message

```plain
E (4670) spi: spi_bus_initialize(768): SPI bus already initialized.
[  3632][E][ETH.cpp:566] beginSPI(): SPI bus initialize failed: 259
```

### Other Steps to Reproduce

I don't have a setup for ESP32 MAC with a PHY module yet so I can't test that

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

closed 2024-05-27 by me-no-dev

JAndrassy, 2024-05-28:
@me-no-dev I spotted one more `end` problem. it removes the even handler without checking if other instances are still active

me-no-dev, 2024-05-28:
can you elaborate a bit more? what other instances you mean?

JAndrassy, 2024-05-28:
`static ETHClass *_ethernets[3] = {NULL, NULL, NULL};`

me-no-dev, 2024-05-28:
ok, I have a board with two ETH on it. Will give it a shot.

PR #9693 (2024-05-27): fix(eth): Fix ETH.stop() with IDF SPI
Fixes: #9685