espressif/arduino-esp32 #9689: The esp32-arduino core should bundle pyserial, so esptool works

egnor, 2024-05-26:
### Related area

platform support for esptool

### Hardware specification

(relevant to all)

### Is your feature request related to a problem?

If pyserial is not installed on the base system, esptool can't run, and because of https://github.com/espressif/esptool/issues/981 you just get a stack trace and "No module named 'serial'" which isn't the most helpful, particularly for people using the Arduino IDE not conversant with Python.

This also makes the [install-on-Linux instructions](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#linux) a lot more complicated than they have to be. Should just be a matter of adding the board manager URL.

### Describe the solution you'd like

Somehow, when installing the arduino-esp32 platform, pyserial should get installed alongside other things.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Jason2866, 2024-05-27:
That needs major changes. You can't install just py serial. You need a Pyenv for Arduino to have all rights to install and don't disturb the system installed python. So a check for pyenv is needed and when not there it needs to be installed. To do so python for all systems needs to be provided.

egnor, 2024-05-27:
(I assume you mean venv, not pyenv?)

You don't actually need a venv to add a bundled module, you can just edit `$PYTHONPATH` to point to the bundled pyserial (or its `.whl`). Even if you are using a venv, you can still use the system Python, though not relying on that would be preferred!

For making portable self contained Python based executables (that bundle Python itself and any needed modules), I hear good things about [PyOxidizer](https://gregoryszorc.com/docs/pyoxidizer/main/index.html); I don't know if it would be compatible with pyserial.

Jason2866, 2024-05-27:
Been there. Believe me only way that will work is to have a extra venv when requirements needs to be installed. Every other try will not work on a platform. Arduino esp8266 is doing that and Platformio (for every framework) too.

egnor, 2024-05-27:
Just seems like there should be some better way. We do bundle a full Python installation for Windows?

Jason2866, 2024-05-27:
What's wrong to provide a small python package. It is around 7MB.

egnor, 2024-05-27:
You mean the Python binary itself (along with whatever libraries are needed)? Sounds good to me!

closed 2025-01-15 by Parsaabasi