espressif/arduino-esp32 #9690: USB vid and pid missing from many boards.txt entries

egnor, 2024-05-27:
### Board

Many many boards

### Device Description

(not relevant)

### Hardware Configuration

(not relevant)

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE, arduino-cli, anything else that uses boards.txt

### Operating System

Any OS

### Flash frequency

Any frequency

### PSRAM enabled

yes

### Upload speed

Any speed

### Description

In the arduino-esp32 `boards.txt` file, many boards that have integrated USB don't have their vid/pid listed, meaning they can't be autodetected by the IDE (or other tools) and the user has to guess which "COM" port might be the board. Maybe there's some reluctance to list generic vid/pid of onboard serial converters, but I think even that is helpful.

The board I happen to have in front of me is the `adafruit_feather_esp32_v2`, but the list of boards that absolutely definitely have USB ports on them but no vid/pid listed is very very long.

If it's worth actually tracking down and fixing this (they're not missing for a reason) then we can try to fix at least the more popular ones?

### Sketch

```cpp
(not relevant)
```

### Debug Message

```plain
(not relevant)
```

### Other Steps to Reproduce

```
arduino-cli core install esp32:esp32
arduino-cli board list  # with one of the affected boards connected
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-27:
be default those boards use the integrated USB CDC/JTAG that actually have hardcoded in hardware VID/PID and is all the same (and not changeable). So it does not make sense to add VID and PID to those. Some boards opt to use TinyUSB instead and they define their VID/PID

egnor, 2024-05-27:
Bah. I guess it's considered poor form to associate a "generic" vid/pid with a board, since it results in a bad experience in the Arduino IDE.

egnor, 2024-05-28:
Closing this out, pending some clarification from our overlords

closed 2024-05-28 by egnor