espressif/arduino-esp32 #9695: missing usb plugged event

Red-Owl, 2024-05-27:
### Board

ESP32 wt32-s3-wrover (wt32sc01 plus)

### Device Description

LIB: docs/en/api/usb_msc.rst
missing usb plugged event

static void usbEventCallback(void* arg, esp_event_base_t event_base, int32_t event_id, void* event_data){
  
    Serial.println(event_id); 
  Serial.println(event_base);
// show all event

  
  if(event_base == ARDUINO_USB_EVENTS){
    arduino_usb_event_data_t * data = (arduino_usb_event_data_t*)event_data;
    switch (event_id){
      case ARDUINO_USB_STARTED_EVENT:
        Serial.println("USB PLUGGED");
        break;
      case ARDUINO_USB_STOPPED_EVENT:
        Serial.println("USB UNPLUGGED");
        break;
      case ARDUINO_USB_SUSPEND_EVENT:
        Serial.printf("USB SUSPENDED: remote_wakeup_en: %u\n", data->suspend.remote_wakeup_en);
        break;
      case ARDUINO_USB_RESUME_EVENT:
        Serial.println("USB RESUMED");
        break;
      
      default:
        break;
    }
  }
}

missing USB PLUGGED events

**ARDUINO_USB_EVENTS
USB UNPLUGGED
1**

### Hardware Configuration

esp32s3-20210327

### Version

latest master (checkout manually)

### IDE Name

platform io

### Operating System

w11

### Flash frequency

40mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

missing parameter

### Sketch

```cpp
static void usbEventCallback(void* arg, esp_event_base_t event_base, int32_t event_id, void* event_data){
  
    Serial.println(event_id);
  Serial.println(event_base);

  
  if(event_base == ARDUINO_USB_EVENTS){
    arduino_usb_event_data_t * data = (arduino_usb_event_data_t*)event_data;
    switch (event_id){
      case ARDUINO_USB_STARTED_EVENT:
        Serial.println("USB PLUGGED");
        break;
      case ARDUINO_USB_STOPPED_EVENT:
        Serial.println("USB UNPLUGGED");
        break;
      case ARDUINO_USB_SUSPEND_EVENT:
        Serial.printf("USB SUSPENDED: remote_wakeup_en: %u\n", data->suspend.remote_wakeup_en);
        break;
      case ARDUINO_USB_RESUME_EVENT:
        Serial.println("USB RESUMED");
        break;
      
      default:
        break;
    }
  }
}

void setup_usb(){
  USB.onEvent(usbEventCallback);
  MSC.vendorID(HARDWARE.US_VID);//max 8 chars
  MSC.productID(HARDWARE.US_PID);//max 16 chars
  MSC.productRevision(HARDWARE.US_PRID);//max 4 chars
  MSC.onStartStop(onStartStop);
  MSC.onRead(onRead);
  MSC.onWrite(onWrite);
  MSC.mediaPresent(POLLO_SDCARD_PRESENT);
  MSC.begin(SD.numSectors(), SD.sectorSize());
  
  
  USBCOM.onEvent(usb_serial_callback);
  USBCOM.begin(115200);
  USB.begin();
}
```

### Debug Message

```plain
none (no error)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2024-06-03:
@SuGlider Can you please help with triage? Thanks

SuGlider, 2024-06-03:
@Red-Owl - please provide more information about the issue.
I've tested it and I can see the events when I plug and unplug the USB - using master branch arduino core code.

The output is based on this example: https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/USBSerial/USBSerial.ino

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
USB PLUGGED
CDC LINE CODING: bit_rate: 115200, data_bits: 8, stop_bits: 0, parity: 0
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB PLUGGED
```

SuGlider, 2024-06-06:
@Red-Owl - please check the IDE Board configuration for the ESP32-S3:
In order to test USB-OTG and make this example sketch to work, `USB Mode` must be `USB-OTG (TinyUSB)`, otherwise it will compile an empty sketch that has no output.

[image]

SuGlider, 2024-06-06:
Closing this issue because there is no problem detected. Please feel free to open it again, if necessary.

closed 2024-06-06 by SuGlider

Red-Owl, 2024-06-25:
void setup_usb(){
  /*
  if(usbcom==true){
    USBCOM.onEvent(usb_serial_callback);
    USBCOM.begin(115200);  
  }else{
    MSC.vendorID(HARDWARE.US_VID);//max 8 chars
    MSC.productID(HARDWARE.US_PID);//max 16 chars
    MSC.productRevision(HARDWARE.US_PRID);//max 4 chars
    MSC.onStartStop(onStartStop);
    MSC.onRead(onRead);
    MSC.onWrite(onWrite);
      if(SD_CARD_INFO[0]==0){
        MSC.mediaPresent(false);
      }else{
        MSC.mediaPresent(true);
      }
    MSC.begin(SD.numSectors(), SD.sectorSize());
  }
  */

  USBD.onEvent(usbEventCallback);
  USBD.begin();
}

USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
USB UNPLUGGED
...
an plugged stop report and NOT write plugged (only usb no other device enabled)