espressif/arduino-esp32 #9702: There are many non-unique VID/PID pairs in `boards.txt`

egnor, 2024-05-28:
### Board

Many many boards

### Device Description

Many many boards

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

All

### Operating System

All

### Flash frequency

N/A

### PSRAM enabled

yes

### Upload speed

N/A

### Description

Per the discussion in https://github.com/espressif/arduino-esp32/issues/9690, and per [the "USB VID/PID" section of the Arduino Platform specification](https://arduino.github.io/arduino-cli/dev/platform-specification/#board-vidpid),
> USB vendor IDs (VID) and product IDs (PID) identify USB devices to the computer. If the board uses a unique VID/PID pair, it may be defined in boards.txt

This suggests that `boards.txt` should not include VID/PID pairs that are "generic" from a USB/serial converter, or from the fixed VID/PID of a chip's Serial/JTAG unit, or from the default VID/PID of a chip's USB OTG module. Otherwise, many different boards might be confused with each other, and the Arduino IDE tends to pick a random one of those choices, and it's a confusing user experience.

However, many entries in `boards.txt` currently have nonunique Espressif-generic VID/PID entries, which should probably be removed?

- 303a:1001 - esp32s3, esp32c3, esp32s3box, esp32s3usbotg, esp32s3camlcd, tinys2, lolin_c3_mini, dfrobot_beetle_esp32c3, adafruit_qtpy_esp32c3, AirM2M_CORE_ESP32C3
- 303a:0002 - esp32s2, deneyapmini

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-28:
the reason is that they all use the same peripheral that has hardware set VID and PID and are not changeable by software.

closed 2024-05-28 by me-no-dev

egnor, 2024-05-28:
But @me-no-dev in #9690 you said peripheral-hardcoded VID/PID should NOT be in `boards.txt`. Which is it? I am confused surely

me-no-dev, 2024-05-28:
wether they are there or not, does not really make a difference. Or does it for you? Either way the board can not be properly identified. Exception of course is when they use TinyUSB (like Adafruit boards, etc.). Still this option is available only on S2 and S3 currently (that have a full USB peripheral).

egnor, 2024-05-28:
It does for me...!

Backing up, my use case (which I think is common!!) is not "here's a serial port, what board is attached?" but instead "this is the board I'm using, which of these serial ports is it connected to?", because lots of random peripherals (my webcam, for example) show up as serial ports. So without some sort of VID/PID matching I'm left to guess like "it's probably the highest numbered one because I connected it last" or "let me plug and unplug it and see which one goes away" or "let's just try all of them".

So it would be helpful (to me!) to have the expected VID/PID even if it's not unique.

BUT I was led to believe from the other bug (and documentation) that this is actively harmful to the user experience with the IDE, so I was like "bah" and then I was like "ok if it's actively harmful to have nonunique VID/PIDs in boards.txt, we should get rid of them, right?" and so here we are.

me-no-dev, 2024-05-28:
I have to check the current state, but last the IDE2 was showing always some random board that matches, which causes port to not be recognized on reboot and so on mess. If that is all fixed and instead the first board that matches is displayed, we can add a "default" board with the vid and pid and name it something like "ESP32 Family MCU".