espressif/arduino-esp32 #9704: -DUSE_ESP_IDF_LOG breaks build on CORE 2 and CORE 3

everslick, 2024-05-28:
### Board

n.a.

### Device Description

n.a.

### Hardware Configuration

n.a.

### Version

latest master (checkout manually)

### IDE Name

n.a.

### Operating System

n.a.

### Flash frequency

n.a.

### PSRAM enabled

yes

### Upload speed

n.a.

### Description

Defining `USE_ESP_IDF_LOG` to make 

`esp_log_set_vprintf(esp_log_vprintf);`

work, breaks the build.

### Sketch

```cpp
Don't have ArduinoIDE. :-(
```

### Debug Message

```plain
excerpt from log:

In file included from /home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/cores/esp32/esp32-hal-log.h:191,
                 from /home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/libraries/Update/src/HttpsOTAUpdate.cpp:16:
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/libraries/Update/src/HttpsOTAUpdate.cpp: In member function 'void HttpsOTAUpdateClass::begin(const char*, const char*, bool)':
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/cores/esp32/esp32-hal-log.h:165:67: error: 'TAG' was not declared in this scope
  165 | #define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
      |                                                                   ^~~
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/tools/esp32-arduino-libs/esp32/include/log/include/esp_log.h:392:81: note: in definition of macro 'ESP_LOG_LEVEL'
  392 |         if (level==ESP_LOG_ERROR )          { esp_log_write(ESP_LOG_ERROR,      tag, LOG_FORMAT(E, format), esp_log_timestamp(), tag __VA_OPT__(,) __VA_ARGS__); } \
      |                                                                                 ^~~
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/cores/esp32/esp32-hal-log.h:165:32: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  165 | #define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
      |                                ^~~~~~~~~~~~~~~~~~~
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/libraries/Update/src/HttpsOTAUpdate.cpp:104:13: note: in expansion of macro 'log_e'
  104 |             log_e("OTA Event Group Create Failed");
      |             ^~~~~
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/cores/esp32/esp32-hal-log.h:165:67: error: 'TAG' was not declared in this scope
  165 | #define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
      |                                                                   ^~~
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/tools/esp32-arduino-libs/esp32/include/log/include/esp_log.h:393:81: note: in definition of macro 'ESP_LOG_LEVEL'
  393 |         else if (level==ESP_LOG_WARN )      { esp_log_write(ESP_LOG_WARN,       tag, LOG_FORMAT(W, format), esp_log_timestamp(), tag __VA_OPT__(,) __VA_ARGS__); } \
      |                                                                                 ^~~
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/cores/esp32/esp32-hal-log.h:165:32: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
  165 | #define log_e(format, ...) do {ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR, TAG, format, ##__VA_ARGS__);}while(0)
      |                                ^~~~~~~~~~~~~~~~~~~
/home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/libraries/Update/src/HttpsOTAUpdate.cpp:104:13: note: in expansion of macro 'log_e'
  104 |             log_e("OTA Event Group Create Failed");
      |             ^~~~~
```

Errors are reported for many files and libs, not only HttpsOTAUpdate.
```

### Other Steps to Reproduce

https://github.com/espressif/arduino-esp32/issues/1380

https://stackoverflow.com/questions/60442350/arduinos-esp-log-set-vprintf-does-not-work-on-esp32

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-28:
https://github.com/espressif/arduino-esp32/commit/d94b32ed20abf33796e4deca3be3fa5bf99b7d60

everslick, 2024-05-28:
Thanks, yes, just realized, that the default TAG name was missing. Unfortunately defining it, will now break NimBLE, because they use a `const char *TAG` symbol. I will report there.

I close this. Thanks!

closed 2024-05-28 by everslick