espressif/arduino-esp32 #9707: ESP32Servo.h doesnt work with v3.0.0 arduino ide

JameSlunAr, 2024-05-28:
### Board

ESP32 DEV Module

### Device Description

ESP32 DEV Kit C

### Hardware Configuration

nothing connect, just upload code

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Cant upload code, the error is
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp: In destructor 'virtual ESP32PWM::~ESP32PWM()':
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp:53:17: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
   53 |                 ledcDetachPin(pin);
      |                 ^~~~~~~~~~~~~
      |                 ledcDetach
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp: In static member function 'static double ESP32PWM::_ledcSetupTimerFreq(uint8_t, double, uint8_t)':
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp:60:16: error: 'ledcSetup' was not declared in this scope
   60 |         return ledcSetup(chan, freq, bit_num);
      |                ^~~~~~~~~
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp: In member function 'double ESP32PWM::setup(double, uint8_t)':
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp:150:17: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
  150 |                 ledcDetachPin(pin);
      |                 ^~~~~~~~~~~~~
      |                 ledcDetach
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp:151:30: error: 'ledcSetup' was not declared in this scope
  151 |                 double val = ledcSetup(getChannel(), freq, resolution_bits);
      |                              ^~~~~~~~~
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp:155:16: error: 'ledcSetup' was not declared in this scope
  155 |         return ledcSetup(getChannel(), freq, resolution_bits);
      |                ^~~~~~~~~
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp: In member function 'void ESP32PWM::adjustFrequencyLocal(double, double)':
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp:172:17: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
  172 |                 ledcDetachPin(pin);
      |                 ^~~~~~~~~~~~~
      |                 ledcDetach
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp:176:17: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  176 |                 ledcAttachPin(pin, getChannel()); // re-attach the pin after frequency adjust
      |                 ^~~~~~~~~~~~~
      |                 ledcAttach
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp: In member function 'void ESP32PWM::attachPin(uint8_t)':
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp:237:17: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
  237 |                 ledcAttachPin(pin, getChannel());
      |                 ^~~~~~~~~~~~~
      |                 ledcAttach
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp: In member function 'void ESP32PWM::detachPin(int)':
d:\Arduino\libraries\ESP32Servo\src\ESP32PWM.cpp:264:9: error: 'ledcDetachPin' was not declared in this scope; did you mean 'ledcDetach'?
  264 |         ledcDetachPin(pin);
      |         ^~~~~~~~~~~~~
      |         ledcDetach

exit status 1

Compilation error: exit status 1

### Sketch

```cpp
ESC.attach(32,1000,2000)
```

### Debug Message

```plain
nothing
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-28:
That is an external library that needs to be updated to support 3.0.0. Please file an issue there

closed 2024-05-28 by me-no-dev

JameSlunAr, 2024-05-28:
yah, i know it is an external library that needs to be updated to adapt with 3.0.0. so, if the library's owner wont do any update then the library cant be used anymore, right? thank you so much for replying me

me-no-dev, 2024-05-28:
You can use earlier versions of the core to keep functionality, but you will miss on the new stuff. There are ways to detect the core version and the lib maintainer can use it to make the lib work for all versions. That is, as you said, if they want to.