espressif/arduino-esp32 #9712: WIFI can no longer use hostname in HTTP URL

thammr, 2024-05-28:
### Board

ESP32S2 Development Board

### Device Description

ESP32S2 Development Board

### Hardware Configuration

Two VL53L1Xs

### Version

v3.0.0

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

OTA

### PSRAM enabled

no

### Upload speed

N/A

### Description

My ESP32S2 POSTs a data packet to InfluxDb every 30-40 seconds. After recompiling my code today, now POSTing the data fails after about 5-7 packets. Wireshark shows that the ESP32 starts trying to send packets to 0.0.0.0. However, if I replace hostname with the IP, everything works fine. While the ESP32S2 is failing to POST data to InfluxDb, all the other networking stuff works OK- MQTT, OTA, ping.

The board manager shows version 3.0.0-rc3. I tried downgrading to rc2 - same result, Downgraded to 2.0.16 - works OK now.

I have at least 6 ESP32s sending data to InfluxDb without problem. All use hostname in the URL. 

   uri = String("http://") + host + ":8086/api/v2/write?org=" + org + "&bucket=" + bucket + "&precision=s";

The following code running on a Tiny Pico starts returning -1 after several minutes, Some time later, it will start returning a valid status (405 in my case). This pattern repeats. 

### Sketch

```cpp
#include <WiFi.h>
#include <HTTPClient.h>
#include "arduino_secrets.h"

String postData = "Hello";
String url = "http://myhost:8080/";

void setup() {
  Serial.begin(115200);
  delay(2000);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(1000);
  }
}

void loop() {

  WiFiClient client;
  HTTPClient http;

  http.begin(client, url);

  http.addHeader("Content-Type", "text/plain; charset=utf-8");
  http.addHeader("Content-Length", String(postData.length()));
  int statusCode = http.POST(postData);
  
  http.end();
  Serial.println(statusCode);
  delay(1000);
}
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-29:
Works fine here. tested daily with `http://google.com`. Please update to 3.0.0 final and enable core debug to Verbose, then post the output here

thammr, 2024-05-31:
Using 3.0.0 final and 30 second loop time, fails after 3 to 12 iterations.

Following is debug output:

21:51:23.421 -> [167419][V][HTTPClient.cpp:230] beginInternal(): url: http://hector:8080/
21:51:23.421 -> [167426][D][HTTPClient.cpp:281] beginInternal(): protocol: http, host: hector port: 8080 url: /
21:51:23.421 -> [167435][D][HTTPClient.cpp:552] sendRequest(): request type: 'POST' redirCount: 0
21:51:23.421 -> 
21:51:30.400 -> [174442][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'hector' with error '-54'
21:51:35.418 -> [179453][I][NetworkClient.cpp:253] connect(): select returned due to timeout 5000 ms for fd 48
21:51:35.418 -> [179462][D][HTTPClient.cpp:1083] connect(): failed connect to hector:8080
21:51:35.464 -> [179469][W][HTTPClient.cpp:1399] returnError(): error(-1): connection refused
21:51:35.464 -> [179476][D][HTTPClient.cpp:379] disconnect(): tcp is closed
21:51:35.464 -> 
21:51:35.464 -> -1

Haven't been able to workout what -54 means.

thammr, 2024-06-01:
Explicitly called getaddrinfo. This returns 202 when above error occurs.

lbernstone, 2024-06-01:
How is that getting resolved in the network? mdns or does the dns server have a default domain?

thammr, 2024-06-01:
The router provides DNS. It does have a default domain. Using the FQDN makes no difference.
I have been running several ESP32s and arduinos on my network for several years. Only had problems with version 3.0.0.*

thammr, 2024-06-10:
DNS fails because the software start trying to send the DNS request to the routers IPV6 address. I don't see any trace of this packet using Wireshark

Before DNS starts to fail, dns_set_server() get called twice, both times with an IPV6 address. After that, DNS fails. Some time later, dns_set_server() is called with an IPV4 address and DNS starts working again.

I suspect that the software is seeing some DHCPV6 packet that it should ignore but I haven't verified this.

I do see a DHCPv6 advertise packet at about the same time as the bogus dns_set_server calls.

me-no-dev, 2024-06-11:
Have you tried enabling IPv6?

thammr, 2024-06-12:
Doesn't seem to make any difference. Not sure I've done it correctly.

void WiFiEvent(WiFiEvent_t event) {
  switch (event) {
    case ARDUINO_EVENT_WIFI_STA_CONNECTED:
      Serial.println("Enabling IPv6");
      WiFi.enableIPv6();
      break;
    default:
      break;
  }
}

Appears dchp6.c is unrelated as it doesn't seem to be calling dns_set_server. 

Can you tell me what version of lwip is used in the current board? None of the releases of lwip that I downloaded appear to match.

me-no-dev, 2024-06-12:
```cpp
WiFi.enableIPv6(true);
WiFi.begin(ssid, pass);
```

thammr, 2024-06-13:
That’s weird. Mostly this works but my software regularly switches between using IPv4 and IPv6 for DNS.

After boot, if the first DNS request works, then DNS keeps succeeding. If the first DNS fails, then DNS keeps failing.
On boot, it mostly works but it fails about 20% of the time.

Strangely, if I connect to my PCs hotspot rather than to my Wifi Access Point, DNS uses Ipv4 exclusively.

SMi1984DE, 2024-06-24:
I had at least a very similar problem with my ongoing ESP32 project. After a certain period, no more data was regularly transmitted to an external server because the hostname could no longer be resolved. Sometimes the problem and the following output appeared after just half an hour, sometimes only after one or two hours:

`[3248820][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'xxxxxxxxx' with error '-54'`
`[3253832][I][ssl_client.cpp:129] start_ssl_client(): select returned due to timeout 5000 ms for fd 48`
`[3253842][I][NetworkClientSecure.cpp:153] connect(): Actual TLS start postponed.`
`[3253849][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1`
`[3253857][W][HTTPClient.cpp:1399] returnError(): error(-1): connection refused`

As user @thammr in https://github.com/espressif/arduino-esp32/issues/9712#issuecomment-2159386661_ pointed out that this issue is related to IPv6, and because user @me-no-dev in https://github.com/espressif/arduino-esp32/issues/9712#issuecomment-2159856618_ suggested enabling IPv6, I explicitly enabled IPv6. My project is working with WifiMulti, where the corresponding option can or must be set directly to true in **WiFiMulti.cpp**:

`32: WiFiMulti::WiFiMulti() {`
`33:    ipv6_support = true;`
`34: }`

The happy result of this is that since then, no connection interruptions have occurred. In my view, it is still a strange behavior when it was possible to resolve the hostname with IPv4 for half an hour before, and then suddenly, no connection can be made because apparently only IPv6 is to be used to reach the DNS server. At least my router has IPv6 enabled, which is probably why this solution works well for me. I do not know if this solution will also work if only an IPv4 network is available. Perhaps my contribution will help solve this case for someone.

Best regards
SMi

me-no-dev, 2024-06-24:
maybe we need to check WiFiMulti, but in general when you switch to IPv6 (having global address that can reach internet) the DNS cache will be purged and new request would be made to get the possible new IPv6 address of the host. This is true since 3.0.0 for using the regular WiFi interfaces. Did not check Multi though

SMi1984DE, 2024-06-24:
> maybe we need to check WiFiMulti, but in general when you switch to IPv6 (having global address that can reach internet) the DNS cache will be purged and new request would be made to get the [...]

Thank you very much for the troubleshooting and your work here on GitHub, @me-no-dev. Just a quick note: I hadn't looked into WiFiMulti in detail yet, but I just read in the description that it is mainly used when multiple Access Points are available and when a connection to the strongest available Wi-Fi network should be established. In my project, I don't really need this function, but WiFiMulti was used in the example project "BasicHttpClient", which I initially chose as the basis for my new project. I can imagine that other users might also use WiFiMulti only for this reason, and that they might then encounter the same issue. Best regards.

thammr, 2024-06-25:
After much investigation, I am convinced that the DNS problem is not due to any deficiency in my LAN appliances. 

Just to summarize my findings.

If there is no ICMPv6 traffic on the LAN, the software uses IPv4 without problem. If I connect to the hotspot on my Windows 11 PC, I see no ICMPv6 packets and IPv4 works perfectly. So there shouldn't be any problem on an IPv4 only network.

However, if there is ICMPv6 traffic on the LAN, the ESP32 decides to join the fun. Every 400 seconds, it sends an ICMPv6 Router Solicitation and then provides the resultant IPv6 address to the dns module. After that, no DNS requests are sent and consequently DNS just fails. At some stage, the dns module is told to use the routers IPv4 address again and DNS starts working until the next 400 second time slice expires and DNS start failing again. While DNS is failing, MQTT and HTPP continue to work fine.

If I enable IPv6, DNS uses IPv6 but HTTP continues to use IPv4. This solution would be viable apart from the fact that DNS fails completely on every 4th or 5th reboot. It appears that something is not being initialized correctly. When DNS fails, no DNS requests are sent.

Furthermore, I can find no documentation on enabling IPv6 and in my humble opinion, IPv6 needs to be optional anyway.

Consequently, I have reverted to 2.0.17.

me-no-dev, 2024-06-25:
> IPv6 needs to be optional anyway

It is optional. You need to call `WiFi.enableIPv6()` to make it work. Do you have this happen without calling it?

thammr, 2024-06-25:
That’s the point. The software tries to use IPv6 for DNS without calling WiFi.enableIPv6() or even if I call WiFi.enableIPv6(false).

I don’t think the software should be sending ICMPv6 messages unless WiFi.enableIPv6() is called.

me-no-dev, 2024-06-28:
> The software tries to use IPv6 for DNS without calling WiFi.enableIPv6() or even if I call WiFi.enableIPv6(false).

unfortunately we have no control, other than `WiFi.enableIPv6()`. Since we can not change the IDF build configuration on the fly and everything is static, we have to enable support for IPv6 when compiling the libs for Arduino. I will ask my colleagues if there is any way that we can prevent all IPv6 until `WiFi.enableIPv6(true)` is called.

thammr, 2024-06-28:
Wireshark begs to differ. In the following figure, packet 10623 is sent by the ESP32. The Wireshark filter is the ESP32s MAC address.

***@***.***

The packets source MAC address is my ESP32s MAC Address.

***@***.***

Following shows the same transaction with the filter off. This shows the router responding to the Router Solicitation.

***@***.***

After the above transaction, DNS fails.

Either the IDF team is wrong or I have something on my network spoofing the MAC address of any ESP32 whose software is built using version 3.

This was captured using 3.0.2 of the board. The code definitely is not calling enableIPv6.

Sent from Mail<https://go.microsoft.com/fwlink/?LinkId=550986> for Windows

SMi1984DE, 2024-06-29:
> The official statement of the IDF team is that there is no way for the ESP to respond to any IPv6 traffic, unless a Link Local address is assigned to the interface, which can also only happen when `enableIPv6(true)` is called.

I reckon this issue might have already been covered in [espressif/esp-idf](https://github.com/espressif/esp-idf/issues/14072), too. ;) I suspect that the error described in the above post stems from the same behavior. But of course, this is just an initial suspicion.

thammr, 2024-07-05:
Well. so much for being able to respond by email. That removed all my attachments. 

[image]

[image]

[image]

Hope this bloody works!

peff74, 2024-07-05:
I am facing a similar problem.
I am sending data to an InfluxDB via the Networkclient.

`NetworkClient client;`
`client.connect("influx.xyz.xyz", 8086);`

 I have always used DNS names there. Since switching to 3.02 this works for the first time, after that DNS is no longer resolved. The script does not use IPv6 either. But IPv6 packets are travelling in the connected network.
 I use Ethernet, so it doesn't seem to be just a WiFi issue.

me-no-dev, 2024-07-05:
@peff74 did you try `ETH.enableIPv6();`?

cziter15, 2024-07-05:
@me-no-dev I've had similar issue in my system. Added `WiFi.enableIPv6()` and it resolved the issue.

Context:
- I have my own smarthome system based on esp32/esp8266 based on [ksiotframeworklib](https://github.com/cziter15/ksIotFrameworkLib). It relies on MQTT in always-connected manner. Shared core logic for both chips.
- My new flat is connected to my parents house. They have 4 devices (3x esp8266, 1x esp32).
- I'm using dyndns - I have `ext IP`, but it changes once a week or so.
- ESP32 is unable to reconnect to the MQTT server using domain name, while 3 other boards based on esp8266 worked. It eventually comes back online after ~4 hours.

I've added some other fixes along with `WiFi.enableIPv6()` and it seems to resolve the issue. The device comes back as fast as esp8266 based collegues.

peff74, 2024-07-05:
> @peff74 did you try `ETH.enableIPv6();`?

`ETH.enableIPv6();` fixes the problem.
But I have to admit I don't know why I have to enable IPv6 on the ESP for DNS to work properly, especially because it works for a while without IPv6.
Sometimes it works for 350 sometimes 500 seconds. 
Then suddenly the `client.connect` needs a little more than 14 seconds, probably the DNS query runs into a timeout.

```
#include <ETH.h>

#define ETH_PHY_TYPE ETH_PHY_LAN8720
#define ETH_PHY_ADDR 1
#define ETH_PHY_POWER 16
#define ETH_PHY_MDC 23
#define ETH_PHY_MDIO 18
#define ETH_CLK_MODE ETH_CLOCK_GPIO0_IN

NetworkClient client;
static bool eth_connected = false;
int count;
int err_count;
unsigned long previousMillis = 0;
const long interval = (1000);

void ETHevent(arduino_event_id_t event) {
  switch (event) {
    case ARDUINO_EVENT_ETH_START:
      Serial.println("ETH Started");
      // The hostname must be set after the interface is started, but needs
      // to be set before DHCP, so set it from the event handler thread.
      ETH.setHostname("WT32-ETH01-TEST");
      break;
    case ARDUINO_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case ARDUINO_EVENT_ETH_GOT_IP:
      Serial.println("ETH Got IP");
      Serial.println(ETH);
      eth_connected = true;
      break;
    case ARDUINO_EVENT_ETH_LOST_IP:
      Serial.println("ETH Lost IP");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default: break;
  }
}

void connect_DNS() {
  if (client.connect("ioborker.fritz.box", 8086, 30)) {
    Serial.println("Connected to InfluxDB --> DNS");
  } else {
    Serial.println("Failed to connect to InfluxDB --> DNS");
    err_count++;
  }
  delay(20);
  client.stop();
}

void connect_IP() {
  if (client.connect("192.168.0.13", 8086, 30)) {
    Serial.println("Connected to InfluxDB --> IP");
  } else {
    Serial.println("Failed to connect to InfluxDB --> IP");
    err_count++;
  }
  delay(20);
  client.stop();
}

void setup() {
  Serial.begin(115200);
  Serial.println("hello");
  //  ETH.enableIPv6(true);
  Network.onEvent(ETHevent);
  ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_MDC, ETH_PHY_MDIO, ETH_PHY_POWER, ETH_CLK_MODE);
}

void loop() {

  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;
    count++;
    Serial.print("Loop  count: ");
    Serial.println(count);
    Serial.print("Error count: ");
    Serial.println(err_count);
    Serial.println();

    if (count % 5 == 0 && eth_connected) {
      Serial.println(ETH.localIP());
      long int t1 = millis();
      connect_DNS();
      long int t2 = millis();
      connect_IP();
      long int t3 = millis();
      Serial.print("Millis_DNS: ");
      Serial.println(t2 - t1);
      Serial.print("Millis_IP:  ");
      Serial.println(t3 - t2);
      Serial.println();
    }

    if (err_count >= 5) {
      ESP.restart();
    }
  }
}
```

Here the debug

That's how it should be
```
192.168.0.72
Connected to InfluxDB --> DNS
Connected to InfluxDB --> IP
Millis_DNS: 24
Millis_IP:  22

Loop  count: 361
Error count: 0
```

Here with error
```
192.168.0.72
Failed to connect to InfluxDB --> DNS
Connected to InfluxDB --> IP
Millis_DNS: 14051
Millis_IP:  21

Loop  count: 366
Error count: 1
```

cziter15, 2024-07-05:
@peff74 check out @thammr posts in this thread, seems to be a bug, looks like in IDF/lwip.

peff74, 2024-07-08:
@me-no-dev 
However, errors still occur regularly despite `ETH.enableIPv6(true);`.

1 loop() == 1 Sec.

```
Loop  count: 5820
Error count: 1

192.168.0.72
[5828662][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
[5828702][I][NetworkClient.cpp:253] connect(): select returned due to timeout 30 ms for fd 49
FConnected to InfluxDB --> IP
Millis_DNS: 70
Millis_IP:  57

Loop  count: 5821
Error count: 2
```

```
Loop  count: 13435
Error count: 2

192.168.0.72
[13443665][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
[13443704][I][NetworkClient.cpp:253] connect(): select returned due to timeout 30 ms for fd 49
Connected to InfluxDB --> IP
Millis_DNS: 73
Millis_IP:  58

Loop  count: 13436
Error count: 3
```

```
Loop  count: 13805
Error count: 3

192.168.0.72
[13813664][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
[13813703][I][NetworkClient.cpp:253] connect(): select returned Connected to InfluxDB --> IP
Millis_DNS: 72
Millis_IP:  61

Loop  count: 13806
Error count: 4
```

me-no-dev, 2024-07-09:
@peff74 that seems to be a different issue. Your DNS is working well

peff74, 2024-07-09:
@me-no-dev 
Unfortunately, this cannot be.
Because  all devices under the old Core 2.x.x have no problems, same as devices with other operating systems.

me-no-dev, 2024-07-09:
This thread is about DNS issue. In your log, DNS is working fine, therefore is a different problem. Your DNS issue was "fixed" by enabling V6, now you have another one

cziter15, 2024-07-09:
@me-no-dev any ideas anyway, what can cause the dns issue?

peff74, 2024-07-09:
@me-no-dev 
Sorry if I have to disagree.
It is a Core 3.x.x. DNS problem.
First of all, DNS only works if IPv6 is enabled.
Second, it does not work in a reliable way.
I do not have any of these problems on Core 2.x.x.

I want to be clear that this is not a criticism of your work, which is great, but there is a problem here with DNS in Core 3.x.x.

me-no-dev, 2024-07-10:
@peff74 we are tracking the DNS problem, but your last log shows DNS working fine, then something else not working. Please differentiate between DNS not working and then your client not connecting. Two separate things.

@cziter15 trying to pursue IDF's network team to accept that there is such issue and look at it. It would have been great if I can reproduce it, but I can't. It definitely needs some particular environment case that I can not simulate here. Work continues...

cziter15, 2024-07-10:
@me-no-dev, when it comes to the environment I wasn't able to reproduce it on ESP32S3 and my home router (sagemcom cs 50001 - with IPv6 enabled on the router). In my parant's house there's D-LINK DWR921 with OpenWRT flashed.

The issue occured in my parent's house. There's a device I've built to monitor their central heating kettle. I'm hosting MQTT server for all the devices. Reproduction was easy - just reboot the router in my house - it triggers no-ip to update the IP and then these devices should retry to reconnect. ESP32 was unable to reconnect. W/o worharound it eventually comes back online but after 4-5h. Other ESP8266 devices come back almost instantly.

s60sc, 2024-07-10:
I've been trying to diagnose why my app was losing connection to a telegram bot due to DNS failure when compiled with core v3.x when it worked fine with v2.x. Having read this thread and added `WiFi.enableIPv6(true)` it seems to have 'fixed' the problem. Telegram bot requires polling so the app makes repeated connections before failing which may be a way to reproduce the problem.

```
// recovery after connection error occurs with WiFi.enableIPv6(true) :

13:19:18.648 -> [###### DBG @ telegram.cpp:211] connected
13:19:23.633 -> [###### DBG @ telegram.cpp:232] NOT connected
13:19:23.633 -> [916216][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:19:23.678 -> [916228][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:19:23.678 -> [916236][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
13:19:23.678 -> [916256][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2001:67c:4e8:f004::9
13:19:29.029 -> [###### DBG @ telegram.cpp:211] connected

13:20:35.159 -> [###### DBG @ telegram.cpp:211] connected
13:20:35.159 -> [987749][E][ssl_client.cpp:36] _handle_error(): [data_to_read():401]: (-76) UNKNOWN ERROR CODE (004C)
13:20:35.205 -> [987761][E][NetworkClientSecure.cpp:309] available(): Closing connection on failed available check
13:20:40.188 -> [992771][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:20:40.188 -> [###### DBG @ telegram.cpp:232] NOT connected
13:20:40.234 -> [992780][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:20:40.234 -> [992792][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:20:40.280 -> [992828][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2001:67c:4e8:f004::9
13:20:45.641 -> [###### DBG @ telegram.cpp:211] connected

// eventual failed recovery after connection error occurs with WiFi.enableIPv6(false) :

13:32:33.151 -> [###### DBG @ telegram.cpp:211] connected
13:32:38.218 -> [ 20535][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:32:38.218 -> [###### DBG @ telegram.cpp:232] NOT connected
13:32:38.218 -> [ 20544][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:32:38.218 -> [ 20556][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:32:38.218 -> [ 20564][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 149.154.167.220
13:32:43.600 -> [###### DBG @ telegram.cpp:211] connected

13:33:33.581 -> [###### DBG @ telegram.cpp:211] connected
13:33:38.641 -> [ 80979][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:33:38.641 -> [###### DBG @ telegram.cpp:232] NOT connected
13:33:38.641 -> [ 80988][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:33:38.641 -> [ 81000][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:33:44.732 -> [ 87066][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'api.telegram.org' with error '-54'
13:34:03.229 -> [105564][E][ssl_client.cpp:146] start_ssl_client(): socket error on fd 53, errno: 113, "Software caused connection abort"
13:34:03.229 -> [105575][I][NetworkClientSecure.cpp:153] connect(): Actual TLS start postponed.
13:34:03.229 -> [105582][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
13:34:03.229 -> [105590][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:34:03.274 -> [12:34:03.296 WARN remoteServerConnect] Timed out connecting to server: api.telegram.org, Err: -1, ERROR - Generic error 
13:34:08.273 -> [110598][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:34:08.273 -> [###### DBG @ telegram.cpp:232] NOT connected
13:34:08.273 -> [110607][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:34:08.273 -> [110619][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:34:14.735 -> [117066][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'api.telegram.org' with error '-54'
13:34:33.234 -> [135564][E][ssl_client.cpp:146] start_ssl_client(): socket error on fd 53, errno: 113, "Software caused connection abort"
13:34:33.234 -> [135575][I][NetworkClientSecure.cpp:153] connect(): Actual TLS start postponed.
13:34:33.234 -> [135582][E][NetworkClientSecure.cpp:159] connect(): start_ssl_client: connect failed: -1
13:34:33.234 -> [135590][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:34:33.280 -> [12:34:33.296 WARN remoteServerConnect] Timed out connecting to server: api.telegram.org, Err: -1, ERROR - Generic error 
13:34:38.279 -> [140598][E][NetworkClient.cpp:315] setSocketOption(): fail on 0, errno: 9, "Bad file number"
13:34:38.279 -> [###### DBG @ telegram.cpp:232] NOT connected
```
<repeats until rebooted>

peff74, 2024-07-12:
> @peff74 we are tracking the DNS problem, but your last log shows DNS working fine, then something else not working. Please differentiate between DNS not working and then your client not connecting. Two separate things.
> 

You're right, so I only tested DNS.

```
void resolve_Google() {
  IPAddress remoteIP;
  char server[] = "www.google.de";

  Serial.print("Resolving IP for: ");
  Serial.println(server);

  if (Network.hostByName(server, remoteIP)) {
    Serial.print("Google is: ");
    Serial.println(remoteIP);
  } else {
    Serial.println("DNS lookup failed.");
    err_count++;
  }
}
void resolve_ioBroker() {
  IPAddress remoteIP;
  char server[] = "ioborker.fritz.box";

  Serial.print("Resolving IP for: ");
  Serial.println(server);

  if (Network.hostByName(server, remoteIP)) {
    Serial.print("ioBroker is: ");
    Serial.println(remoteIP);
  } else {
    Serial.println("DNS lookup failed.");
    err_count++;
  }
}

```
I have resolved both IPv6 and IPv4.
It worked for hours.
I started at 10:45am and the next morning around 4:49am problems began.
IPv4 resolution suddenly always took 7 or 14 seconds.
```
12.07.24, 04:48:23 MESZ	DNS Server: fd00::464e:6dff:feb7:3a63
12.07.24, 04:48:23 MESZ	Local IPv4 192.168.0.72
12.07.24, 04:48:23 MESZ	Local IPv6 fe80::a2a3:b3ff:fe76:cd67
12.07.24, 04:48:23 MESZ	Resolving IP for: www.google.de
12.07.24, 04:48:23 MESZ	[65080043][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2a00:1450:4001:82a::2003
12.07.24, 04:48:23 MESZ	Google is: 2a00:1450:4001:82a::2003
12.07.24, 04:48:23 MESZ	Resolving IP for: ioborker.fritz.box
12.07.24, 04:48:23 MESZ	[65080058][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
12.07.24, 04:48:23 MESZ	ioBroker is: 192.168.0.13
12.07.24, 04:48:23 MESZ	Millis_Google: 23
12.07.24, 04:48:23 MESZ	Millis_ioBroker:  13
12.07.24, 04:48:23 MESZ	
12.07.24, 04:48:24 MESZ	Loop  count: 65069
12.07.24, 04:48:24 MESZ	Error count: 0
12.07.24, 04:48:24 MESZ	
12.07.24, 04:48:24 MESZ	DNS Server: fd00::464e:6dff:feb7:3a63
12.07.24, 04:48:24 MESZ	Local IPv4 192.168.0.72
12.07.24, 04:48:24 MESZ	Local IPv6 fe80::a2a3:b3ff:fe76:cd67
12.07.24, 04:48:24 MESZ	Resolving IP for: www.google.de
12.07.24, 04:48:24 MESZ	[65081043][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2a00:1450:4001:82a::2003
12.07.24, 04:48:24 MESZ	Google is: 2a00:1450:4001:82a::2003
12.07.24, 04:48:24 MESZ	Resolving IP for: ioborker.fritz.box
12.07.24, 04:48:24 MESZ	[65081058][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
12.07.24, 04:48:24 MESZ	ioBroker is: 192.168.0.13
12.07.24, 04:48:24 MESZ	Millis_Google: 23
12.07.24, 04:48:24 MESZ	Millis_ioBroker:  13
12.07.24, 04:48:24 MESZ	
12.07.24, 04:48:25 MESZ	Loop  count: 65070
12.07.24, 04:48:25 MESZ	Error count: 0
12.07.24, 04:48:25 MESZ	
12.07.24, 04:48:25 MESZ	DNS Server: fd00::464e:6dff:feb7:3a63
12.07.24, 04:48:25 MESZ	Local IPv4 192.168.0.72
12.07.24, 04:48:25 MESZ	Local IPv6 fe80::a2a3:b3ff:fe76:cd67
12.07.24, 04:48:25 MESZ	Resolving IP for: www.google.de
12.07.24, 04:48:25 MESZ	[65082043][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2a00:1450:4001:82a::2003
12.07.24, 04:48:25 MESZ	Google is: 2a00:1450:4001:82a::2003
12.07.24, 04:48:25 MESZ	Resolving IP for: ioborker.fritz.box
12.07.24, 04:48:25 MESZ	[65082059][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
12.07.24, 04:48:25 MESZ	ioBroker is: 192.168.0.13
12.07.24, 04:48:25 MESZ	Millis_Google: 23
12.07.24, 04:48:25 MESZ	Millis_ioBroker:  13
12.07.24, 04:48:25 MESZ	
12.07.24, 04:48:26 MESZ	Loop  count: 65071
12.07.24, 04:48:26 MESZ	Error count: 0
12.07.24, 04:48:26 MESZ	
12.07.24, 04:48:26 MESZ	DNS Server: fd00::464e:6dff:feb7:3a63
12.07.24, 04:48:26 MESZ	Local IPv4 192.168.0.72
12.07.24, 04:48:26 MESZ	Local IPv6 fe80::a2a3:b3ff:fe76:cd67
12.07.24, 04:48:26 MESZ	Resolving IP for: www.google.de
12.07.24, 04:48:26 MESZ	[65083043][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2a00:1450:4001:82a::2003
12.07.24, 04:48:26 MESZ	Google is: 2a00:1450:4001:82a::2003
12.07.24, 04:48:26 MESZ	Resolving IP for: ioborker.fritz.box
12.07.24, 04:48:40 MESZ	[65097057][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
12.07.24, 04:48:40 MESZ	ioBroker is: 192.168.0.13
12.07.24, 04:48:40 MESZ	Millis_Google: 23
12.07.24, 04:48:40 MESZ	Millis_ioBroker:  14009
12.07.24, 04:48:40 MESZ	
12.07.24, 04:48:40 MESZ	Loop  count: 65072
12.07.24, 04:48:40 MESZ	Error count: 0
12.07.24, 04:48:40 MESZ	
12.07.24, 04:48:40 MESZ	DNS Server: fd00::464e:6dff:feb7:3a63
12.07.24, 04:48:40 MESZ	Local IPv4 192.168.0.72
12.07.24, 04:48:40 MESZ	Local IPv6 fe80::a2a3:b3ff:fe76:cd67
12.07.24, 04:48:40 MESZ	Resolving IP for: www.google.de
12.07.24, 04:48:40 MESZ	[65097077][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2a00:1450:4001:82a::2003
12.07.24, 04:48:40 MESZ	Google is: 2a00:1450:4001:82a::2003
12.07.24, 04:48:40 MESZ	Resolving IP for: ioborker.fritz.box
12.07.24, 04:48:47 MESZ	[65104057][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
12.07.24, 04:48:47 MESZ	ioBroker is: 192.168.0.13
12.07.24, 04:48:47 MESZ	Millis_Google: 19
12.07.24, 04:48:47 MESZ	Millis_ioBroker:  6969
12.07.24, 04:48:47 MESZ	
12.07.24, 04:48:47 MESZ	Loop  count: 65073
12.07.24, 04:48:47 MESZ	Error count: 0
```
And if DNS takes too long... of course does not work connect(), too.
Or am I wrong?

me-no-dev, 2024-07-12:
Does it continuously take multiple seconds, or does it "fix" itself after some time? Maybe TTL of the record is about to expire in the cache and it does not know which DNS to ask, so query fails and it returns the cached result.

peff74, 2024-07-12:
No, as soon as the error starts, it persists endless.
Only rebooting the ESP or resetting the network connection (unplugging and plugging the LAN cable) makes it work normally again.

here is the LOG how the DNS works again by reconnect

```
12.07.24, 11:33:34 MESZ	DNS Server: fd00::464e:6dff:feb7:3a63
12.07.24, 11:33:34 MESZ	Local IPv4 192.168.0.72
12.07.24, 11:33:34 MESZ	Local IPv6 fe80::a2a3:b3ff:fe76:cd67
12.07.24, 11:33:34 MESZ	Resolving IP for: www.google.de
12.07.24, 11:33:34 MESZ	[89390416][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2a00:1450:4001:80e::2003
12.07.24, 11:33:34 MESZ	Google is: 2a00:1450:4001:80e::2003
12.07.24, 11:33:34 MESZ	Resolving IP for: ioborker.fritz.box
12.07.24, 11:33:42 MESZ	[89397079][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
12.07.24, 11:33:42 MESZ	ioBroker is: 192.168.0.13
12.07.24, 11:33:42 MESZ	Millis_Google: 19
12.07.24, 11:33:42 MESZ	Millis_ioBroker:  6651
12.07.24, 11:33:42 MESZ	
12.07.24, 11:33:42 MESZ	Loop  count: 67273
12.07.24, 11:33:42 MESZ	Error count: 0
12.07.24, 11:33:42 MESZ	
12.07.24, 11:33:42 MESZ	DNS Server: fd00::464e:6dff:feb7:3a63
12.07.24, 11:33:42 MESZ	Local IPv4 192.168.0.72
12.07.24, 11:33:42 MESZ	Local IPv6 fe80::a2a3:b3ff:fe76:cd67
12.07.24, 11:33:42 MESZ	Resolving IP for: www.google.de
12.07.24, 11:33:42 MESZ	[89397098][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2a00:1450:4001:80e::2003
12.07.24, 11:33:42 MESZ	Google is: 2a00:1450:4001:80e::2003
12.07.24, 11:33:42 MESZ	Resolving IP for: ioborker.fritz.box
12.07.24, 11:33:44 MESZ	[89400609][V][ETH.cpp:100] _onEthEvent(): eth0 Disconnected
12.07.24, 11:33:44 MESZ	[89400614][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 4 - ETH_DISCONNECTED
12.07.24, 11:33:44 MESZ	ETH Disconnected
12.07.24, 11:34:02 MESZ	[89418609][V][ETH.cpp:95] _onEthEvent(): eth0 Connected
12.07.24, 11:34:02 MESZ	[89418614][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 3 - ETH_CONNECTED
12.07.24, 11:34:02 MESZ	[89418617][V][NetworkInterface.cpp:66] _onIpEvent(): eth0 Got Same IP: 192.168.0.72 MASK: 255.255.255.0 GW: 192.168.0.1
12.07.24, 11:34:02 MESZ	[89418632][V][ETH.cpp:80] onEthConnected(): Enabled IPv6 Link Local on eth0
12.07.24, 11:34:02 MESZ	ETH Connected
12.07.24, 11:34:02 MESZ	[89418639][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 5 - ETH_GOT_IP
12.07.24, 11:34:02 MESZ	ETH Got IP
12.07.24, 11:34:02 MESZ	*eth0: <UP,100M,FULL_DUPLEX,AUTO,ADDR:0x1> (DHCPC,GARP,IP_MOD)
12.07.24, 11:34:02 MESZ	      ether A0:A3:B3:76:CD:67
12.07.24, 11:34:02 MESZ	      inet 192.168.0.72 netmask 255.255.255.0 broadcast 192.168.0.255
12.07.24, 11:34:02 MESZ	      gateway 192.168.0.1 dns 192.168.0.1
12.07.24, 11:34:02 MESZ	      inet6 fe80::a2a3:b3ff:fe76:cd67%en1 type LINK_LOCAL
12.07.24, 11:34:02 MESZ	
12.07.24, 11:34:03 MESZ	[89419077][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'ioborker.fritz.box' with error '-54'
12.07.24, 11:34:03 MESZ	ioBroker is: 0.0.0.0
12.07.24, 11:34:03 MESZ	Millis_Google: 18
12.07.24, 11:34:03 MESZ	Millis_ioBroker:  21971
12.07.24, 11:34:03 MESZ	
12.07.24, 11:34:03 MESZ	Loop  count: 67274
12.07.24, 11:34:03 MESZ	Error count: 0
12.07.24, 11:34:03 MESZ	
12.07.24, 11:34:03 MESZ	DNS Server: 192.168.0.1
12.07.24, 11:34:03 MESZ	Local IPv4 192.168.0.72
12.07.24, 11:34:03 MESZ	Local IPv6 ::
12.07.24, 11:34:03 MESZ	Resolving IP for: www.google.de
12.07.24, 11:34:03 MESZ	[89419098][D][NetworkManager.cpp:83] hostByName(): Clearing DNS cache
12.07.24, 11:34:03 MESZ	[89419122][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 216.58.212.163
12.07.24, 11:34:03 MESZ	Google is: 216.58.212.163
12.07.24, 11:34:03 MESZ	Resolving IP for: ioborker.fritz.box
12.07.24, 11:34:03 MESZ	[89419131][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
12.07.24, 11:34:03 MESZ	ioBroker is: 192.168.0.13
12.07.24, 11:34:03 MESZ	Millis_Google: 32
12.07.24, 11:34:03 MESZ	Millis_ioBroker:  12
12.07.24, 11:34:03 MESZ	
12.07.24, 11:34:03 MESZ	[89419875][V][NetworkInterface.cpp:111] _onIpEvent(): IF eth0 Got IPv6: Interface: 3, IP Index: 0, Type: LINK_LOCAL, Zone: 2 (en1), Address: fe80:0000:0000:0000:a2a3:b3ff:fe76:cd67
12.07.24, 11:34:03 MESZ	[89419892][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 7 - ETH_GOT_IP6
12.07.24, 11:34:04 MESZ	Loop  count: 67275
12.07.24, 11:34:04 MESZ	Error count: 0
12.07.24, 11:34:04 MESZ	
[... 67 lines cut ...]
12.07.24, 11:34:08 MESZ	[89424111][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2a00:1450:4001:80e::2003
12.07.24, 11:34:08 MESZ	Google is: 2a00:1450:4001:80e::2003
12.07.24, 11:34:08 MESZ	Resolving IP for: ioborker.fritz.box
12.07.24, 11:34:08 MESZ	[89424124][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
12.07.24, 11:34:08 MESZ	ioBroker is: 192.168.0.13
12.07.24, 11:34:08 MESZ	Millis_Google: 32
12.07.24, 11:34:08 MESZ	Millis_ioBroker:  14
12.07.24, 11:34:08 MESZ	
12.07.24, 11:34:09 MESZ	Loop  count: 67280
12.07.24, 11:34:09 MESZ	Error count: 0
12.07.24, 11:34:09 MESZ	
12.07.24, 11:34:09 MESZ	DNS Server: fd00::464e:6dff:feb7:3a63
12.07.24, 11:34:09 MESZ	Local IPv4 192.168.0.72
12.07.24, 11:34:09 MESZ	Local IPv6 fe80::a2a3:b3ff:fe76:cd67
12.07.24, 11:34:09 MESZ	Resolving IP for: www.google.de
12.07.24, 11:34:09 MESZ	[89425098][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2a00:1450:4001:80e::2003
12.07.24, 11:34:09 MESZ	Google is: 2a00:1450:4001:80e::2003
12.07.24, 11:34:09 MESZ	Resolving IP for: ioborker.fritz.box
12.07.24, 11:34:09 MESZ	[89425113][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 192.168.0.13
12.07.24, 11:34:09 MESZ	ioBroker is: 192.168.0.13
12.07.24, 11:34:09 MESZ	Millis_Google: 23
12.07.24, 11:34:09 MESZ	Millis_ioBroker:  13
12.07.24, 11:34:09 MESZ	

```

peff74, 2024-07-13:
I have now also performed a test with a public host that is only accessible via IPv4.
The result is the same.

```
13.07.24, 09:12:21 MESZ --> Loop  count: 63223
13.07.24, 09:12:21 MESZ --> Error count: 0
13.07.24, 09:12:21 MESZ --> 
13.07.24, 09:12:21 MESZ --> DNS Server: fd00::464e:6dff:feb7:3a63
13.07.24, 09:12:21 MESZ --> Local IPv4 192.168.0.72
13.07.24, 09:12:21 MESZ --> Local IPv6 fe80::a2a3:b3ff:fe76:cd67
13.07.24, 09:12:21 MESZ --> Resolving IP for: www.google.de
13.07.24, 09:12:21 MESZ --> [76971033][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first 2a00:1450:4001:803::2003
13.07.24, 09:12:21 MESZ --> Google is: 2a00:1450:4001:803::2003
13.07.24, 09:12:21 MESZ --> Resolving IP for: de5.portmap64.net
13.07.24, 09:12:28 MESZ --> [76978014][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 185.248.148.15
13.07.24, 09:12:28 MESZ --> portmap is: 185.248.148.15
13.07.24, 09:12:28 MESZ --> Millis_Google: 19
13.07.24, 09:12:28 MESZ --> Millis_portmap:  6969
```

me-no-dev, 2024-07-15:
I have been able to replicate the issue (V4 DNS fails, when V6 is advertised)! Stay tuned!

me-no-dev, 2024-07-16:
Fix is on the way! Will be fixed in 3.0.3 (hopefully tomorrow)

closed 2024-07-16 by me-no-dev

kenjones007, 2024-07-17:
Hi all, thanks a lot! Just wanted to confirm, that this fix works! Good job! I had this exact issue on all my ESP32 (ESP32-D0WDQ6) while IPV6-DualStack also enabled on my router (Swisscom Internet Box 3 / Arcadyan VRV951A FW 14.20.24). Only way to get around this issue was to disable IPV6 on my router. Today i recompiled my sources with 3.0.3 and enabled IPV6 on my router (not in my arduino source), and i no longer get a IPV6 DNS, it stays at IPV4.

peff74, 2024-07-18:
I ran my DNS IPv4 / IPv6 script with Core 3.0.3 tonight, with the ETH.enableIPv6(true);. 
This is still running into problems after a few hours.
DNS resolution from IPv4 then suddenly always takes over 6 seconds. IPv6 continues to work quickly.

So there still seems to be a problem.
```
18.07.24, 06:09:00 MESZ	DNS Server: fd00::464e:6dff:feb7:3a63
18.07.24, 06:09:00 MESZ	Local IPv4 192.168.0.72
18.07.24, 06:09:00 MESZ	Local IPv6 fe80::a2a3:b3ff:fe76:cd67
18.07.24, 06:09:00 MESZ	Resolving IP for: www.google.de
18.07.24, 06:09:00 MESZ	[31779104][D][NetworkManager.cpp:103] hostByName(): DNS found IPv6 first  2a00:1450:4001:808::2003
18.07.24, 06:09:00 MESZ	Google is: 2a00:1450:4001:808::2003
18.07.24, 06:09:00 MESZ	Resolving IP for: de5.portmap64.net
18.07.24, 06:09:07 MESZ	[31786085][D][NetworkManager.cpp:123] hostByName(): DNS found IPv4 185.248.148.15
18.07.24, 06:09:07 MESZ	portmap is: 185.248.148.15
18.07.24, 06:09:07 MESZ	Millis_Google: 18
18.07.24, 06:09:07 MESZ	Millis_portmap:  6971
```

Now I'll test client.connect(); with DNS and IPv4 only without ETH.enableIPv6(true).
That's what I'm really using.

me-no-dev, 2024-07-18:
> Now I'll test client.connect(); with DNS and IPv4 only without ETH.enableIPv6(true).

this is what was actually fixed. Why your queries start to take more time in a sudden is not yet clear

peff74, 2024-07-19:
I can confirm that now client.connect() works perfectly with DNS and only IPv4 (i.e. without ETH.enableIPv6(true)).
My Script has been running for 24 hours without any problems.

Are you still working on the other topic, or should another ticket be opened for it?
I personally don't use it, but others might have problems with it

me-no-dev, 2024-07-19:
what other topic?

peff74, 2024-07-19:
well, the issue that when IPv6 is activated, the queries for IPv4 take an extremely long time after a few hours.

peff74, 2024-07-19:
Are you trying internal or external hosts to resolve?
I tested it on two different networks and the results were comparable on both.
After 6-8h IPv4 always took more than 6s.

One network was connected with Fritzbox as router/DNS and FTH.
The other network was connected via a Juniper router using company fiber and a local W2k19 DNS server.

me-no-dev, 2024-07-20:
My router does not provide local DNS and I use MDNS to access internal hosts, so no way to test the same way as you. You sure your local DNS is not limiting you? What if you don't bombard it with requests every second? And why is cache not working? What is the TTL of the records that your local DNS is providing? Anyway... I will not fault the ESP for that. As you see your external hosts are fine, but at the same time, the ESP does not care or know that it's not a remote host, so also no way to act different based on it.

schreibfaul1, 2024-07-28:
It looks as if the DNS is "forgotten" after a short time.  The example code runs for several minutes until 
````
[277128][I][WiFiconnect.ino.ino:39] loop(): 276s   host2: success
[282144][I][WiFiconnect.ino.ino:34] loop(): 281s   host1: success
[287364][I][WiFiconnect.ino.ino:39] loop(): 287s   host2: success
[306206][E][NetworkManager.cpp:130] hostByName(): DNS Failed for 'as-hls-ww-live.akamaized.net' with error '-54'
[309219][I][NetworkClient.cpp:253] connect(): select returned due to timeout 3000 ms for fd 48
[309228][I][WiFiconnect.ino.ino:35] loop(): host1: connection failed
````

Activating `// WiFi.setDNS(dns1, dns2);` refreshes the DNS entries and counteracts "dementia". 

````c++
#include "Arduino.h"
#include "WiFi.h"
#include "WiFiClient.h"

WiFiClient client;

char ssid[] = "*****";
char pwd[]  = "*****";

char host1[] = "as-hls-ww-live.akamaized.net";
char host2[] = "stream.revma.ihrhls.com";

IPAddress dns1(8, 8, 8, 8);
IPAddress dns2(8, 8, 4, 4); 
uint32_t timer;
void setup() {
    WiFi.begin(ssid, pwd);
    while(WiFi.status() != WL_CONNECTED){;} 
    log_i("WiFi connected");
    timer = millis();
}

void loop(){
    int res;
    
    // WiFi.setDNS(dns1, dns2);

    vTaskDelay(5000 / portTICK_PERIOD_MS);
    res = client.connect(host1, 80);
    if(res)  log_i("%is   host1: success", (millis() - timer) / 1000);
    if(!res) log_i("host1: connection failed");

    vTaskDelay(5000 / portTICK_PERIOD_MS);
    res = client.connect(host2, 80);
    if(res)  log_i("%is   host2: success", (millis() - timer) / 1000);
    if(!res) log_i("host2: connection failed");   

}

````