espressif/arduino-esp32 #9714: Compilation error: 'system_event_t' has not been declared

reltkaine, 2024-05-29:
### Board

wemos lolin esp32 lite

### Device Description

wemos lolin esp32 lite

### Hardware Configuration

na

### Version

v3.0.0

### IDE Name

arduino ide 2

### Operating System

win10

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

11520

### Description

Compilation error: 'system_event_t' has not been declared

### Sketch

```cpp
esp_err_t event_handler(void* ctx, system_event_t* event) {
  return ESP_OK;
}
```

### Debug Message

```plain
na
```

### Other Steps to Reproduce

v2.0.16 works!!!!

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-29:
this is way not enough to help you. Please post complete minimal sketch to see how and where you are using this.

reltkaine, 2024-05-29:
cant compile this -->>>> https://github.com/G4lile0/ESP32-WiFi-Hash-Monster

me-no-dev, 2024-05-30:
there is something wrong in the environment. Either this sketch or ESP32-Chimera-Core is missing an include to resolve `ESP_IDF_VERSION_MAJOR` and use the proper code. `ESP_IDF_VERSION_MAJOR` is 5 in this case and should not go through the `system_event_t ` route

closed 2024-05-30 by me-no-dev