espressif/arduino-esp32 #9715: error: 'NetworkServer' does not name a type

Hehao666, 2024-05-29:
### Board

ESP32 WROOM

### Device Description

两个板子，ESP32 wroom和AI think ESP32-CAM

### Hardware Configuration

自制开发板和安信可的32CAM

### Version

v3.0.0

### IDE Name

Arduino IDE

### Operating System

windows 11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

esp32版本从2.0.14升级到3.0.0，编译时错误
In file included from D:\esp32SDcard\arduino����\ESP32_SDweb_mmc/Server.h:2,
                 from C:\Users\hehao\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\Network\src/NetworkServer.h:22,
                 from C:\Users\hehao\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\Network\src/Network.h:13,
                 from C:\Users\hehao\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\WiFi\src/WiFiGeneric.h:39,
                 from C:\Users\hehao\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\WiFi\src/WiFiSTA.h:29,
                 from C:\Users\hehao\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\WiFi\src/WiFi.h:33,
                 from D:\esp32SDcard\arduino程序\ESP32_SDweb_mmc\server.h:1,
                 from D:\esp32SDcard\arduino程序\ESP32_SDweb_mmc\server.cpp:1:
C:\Users\hehao\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\WebServer\src/WebServer.h:257:3: error: 'NetworkServer' does not name a type
  257 |   NetworkServer _server;
      |   ^~~~~~~~~~~~~

### Sketch

```cpp
这是我的项目地址https://github.com/Hehao666/esp32SDCard，2.0.14版本开发，升级到3.0.0想迁移到C6，结果编译ESP32 Dev Module和AI Think时不通过
```

### Debug Message

```plain
无
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-29:
You have a file named `Server.h` in your sketch folder that conflicts with Arduino's `Server.h`. Rename your file to something else and try again

me-no-dev, 2024-05-29:
Also we do accept reports only in English

closed 2024-05-29 by Hehao666