espressif/arduino-esp32 #9716: ESP32C6 can't run Zigbee sample application, keeps rebooting

limengdu, 2024-05-29:
### Board

ESP32-C6

### Device Description

I'm using the XIAO ESP32C6.

### Hardware Configuration

The development board running the example Zigbee_light_Bulb is not hooked up to anything. The development board running the example Zigbee_Light_Switch has a button module connected to GPIO1.

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I ran the provided examples, Zigbee_light_Bulb and Zigbee_Light_Switch, using two different ESP32-C6's, respectively, with no modifications, the programs compiled and passed, and after uploading them, the device serial prompts kept rebooting.

### Sketch

```cpp
#ifndef ZIGBEE_MODE_ZCZR
#error "Zigbee coordinator mode is not selected in Tools->Zigbee mode"
#endif

#include "esp_zigbee_core.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ha/esp_zigbee_ha_standard.h"

/* Switch configuration */
#define GPIO_INPUT_IO_TOGGLE_SWITCH GPIO_NUM_9
#define PAIR_SIZE(TYPE_STR_PAIR)    (sizeof(TYPE_STR_PAIR) / sizeof(TYPE_STR_PAIR[0]))

typedef enum {
  SWITCH_ON_CONTROL,
  SWITCH_OFF_CONTROL,
  SWITCH_ONOFF_TOGGLE_CONTROL,
  SWITCH_LEVEL_UP_CONTROL,
  SWITCH_LEVEL_DOWN_CONTROL,
  SWITCH_LEVEL_CYCLE_CONTROL,
  SWITCH_COLOR_CONTROL,
} switch_func_t;

typedef struct {
  uint8_t pin;
  switch_func_t func;
} switch_func_pair_t;
[... 274 lines cut ...]
40819470: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xe5000000 0xbaad5678
40819490: 0x00000020 0xabba1234 0x00000014 0xd3d2d1d0 0xd7d6d5d4 0xdbdad9d8 0xdfdedddc 0x00000000
408194b0: 0xbaad5678 0x00000020 0xabba1234 0x00000014 0x4267695a 0x6c416565 0x6e61696c 0x39306563
408194d0: 0x00000000 0xbaad5678 0x0000334c 0xabba1234 0x00003340 0x00000000 0x00000000 0x00000000
408194f0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40819510: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 4eabfb2626f008ec

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x1f (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xc64
load:0x4086e610,len:0x2730
load:0x40875728,len:0x594
entry 0x4086c410
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2024-05-29:
Hi @limengdu, the reason it keeps rebooting is that you did not select proper Partition scheme for Zigbee.
Please read the `README.md` of the examples, everything is explained there.

Steps from README file:
```
* Before Compile/Verify, select the correct board: `Tools -> Board`.
* Select the Coordinator Zigbee mode: `Tools -> Zigbee mode: Zigbee ZCZR (coordinator)`.
* Select Partition Scheme for Zigbee: `Tools -> Partition Scheme: Zigbee 4MB with spiffs`.
* Select the COM port: `Tools -> Port: xxx where the `xxx` is the detected COM port.
* Optional: Set debug level to info to see logs from Zigbee stack: `Tools -> Core Debug Level: Info`.
```

Please let me know if that worked out :)

rnovacek, 2024-05-29:
I'm using platformio and had the same problem. I added this into my `platformio.ini` and it has resolved the problem for me:

```
board_build.partitions = zigbee.csv
```

Thanks @P-R-O-C-H-Y for the tip, it would've taken me ages to figure this out!

limengdu, 2024-05-30:
[image]

P-R-O-C-H-Y, 2024-05-30:
Hi @limengdu, I will go through all C6/H2 boards and add all necessary options for Zigbee. You can follow this PR #9700.
I am closing this as solved/duplicate for the PRI linked.