espressif/arduino-esp32 #9718: error "This header should only be included when building for ESP32 (IDFGH-12916)

Lems123456789, 2024-05-29:
In file included from C:\Users\Kingsley\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32c3/include/esp_rom/include/esp32/rom/rtc.h:8,
                 from D:\PRIYANKA-WIFI_P2P\la\lak\frigo100wifi_23082023\frigo100wifi_23082023.ino:58:
C:\Users\Kingsley\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32c3/include/esp_rom/include/esp32/rom/ets_sys.h:15:2: error: #error "This header should only be included when building for ESP32"
   15 | #error "This header should only be included when building for ESP32"
      |  ^~~~~

exit status 1

Compilation error: exit status 1

When I'm using esp32 C3 mini 
[image]
Not finding header file.

me-no-dev, 2024-05-29:
and what do you have on line 58 in your sketch?

Lems123456789, 2024-05-29:
D:\PRIYANKA-WIFI_P2P\la\lak\frigo100wifi_23082023\frigo100wifi_23082023.ino: In function 'void setup()':
D:\PRIYANKA-WIFI_P2P\la\lak\frigo100wifi_23082023\frigo100wifi_23082023.ino:844:34: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  844 |         // timerAlarmWrite(My_SystemTimer, 1000000, true);///1sec
      |                        ~~~~~~~~~~^~~~~~~~~~~~~
In file included from C:\Users\Kingsley\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/esp32-hal.h:84,
                 from C:\Users\Kingsley\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/Arduino.h:36,
                 from C:\Users\Kingsley\AppData\Local\Temp\arduino\sketches\9FC6A7B71B6370FEFA30089A4EAF4179\sketch\frigo100wifi_23082023.ino.cpp:1:
C:\Users\Kingsley\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
D:\PRIYANKA-WIFI_P2P\la\lak\frigo100wifi_23082023\frigo100wifi_23082023.ino:846:9: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  846 | 
      |         ^              
      |         timerWrite
D:\PRIYANKA-WIFI_P2P\la\lak\frigo100wifi_23082023\frigo100wifi_23082023.ino:847:9: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  847 | 
      |         ^               
      |         timerAlarm

exit status 1

Compilation error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'