espressif/arduino-esp32 #9721: Add support for Low Power core of the ESP32C6

rrrus, 2024-05-29:
### Related area

Low power core

### Hardware specification

ESP32C6

### Is your feature request related to a problem?

I am unable to program the low power core of the ESP32C6 with the Arduino IDE.

### Describe the solution you'd like

Enable authoring, compiling, and running of code on both the high power and low power cores of the ESP32C6 in C & assembly at the very least, preferably C++ from within the Arduino IDE.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2024-05-29:
I have requested that this be added to the compiled sdk.  Note that the LP-core is a RISC-V arch, and will is expected to be programmed in C, not assembly.  I will test later today or tomorrow to see if this is even possible using the S3.  IIRC, there is a 2-pass compile, and in esp-idf the ulp code gets compiled down and included as an asm blob.  If that is the case, the arduino compiler probably will not be able to program the ulp properly.

me-no-dev, 2024-05-29:
It's not yet possible, but we will revisit it later. For now only the FSM ULP can be used (where available)

SuGlider, 2024-05-30:
@rrrus @lbernstone 
This is possible by using Arduino as IDF component.
The IDF tools and building process will be available, making it feasible.

SuGlider, 2024-05-30:
@sudeep-mohanty - Let talk about how to move forward with C6 LP support for Arduino IDE.

morcibacsi, 2024-10-20:
Can we have an update on this? Is it planned? Is there some workaround?

WereCatf, 2024-11-03:
> It's not yet possible, but we will revisit it later. For now only the FSM ULP can be used (where available)

Absolutely not a single person likes the FSM ULP core. It really should be a priority to figure some way of supporting the RISC-V core under Arduino.

lbernstone, 2024-11-03:
The compiler is essentially doing a two pass compile. Perhaps this could be a IDE plugin that you execute before your main compile that will build out the artifacts that IDF expects? Maybe it could be a specific filename that calls a pre-compile script like build_opt.h does? I assume the P4 will be using the same mechanism, so this is going to become a common request.

Parsaabasi, 2025-01-15:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

rrrus, 2025-01-16:
I would like to reopen this issue as it still persists as a missing feature.

me-no-dev, 2025-01-16:
We do not think that this will be possible with the Arduino build system. But I'll reopen it

Uspizig, 2026-10-20:
Oh boy.. No updates. So sad