espressif/arduino-esp32 #9723: new version 3.0.0

Paulo1963, 2024-05-29:
### Board

doit esp32 devkit v1

### Device Description

devki v1

### Hardware Configuration

onewire = gpio 14
i2c = gpio 21 & 22

### Version

v3.0.0

### IDE Name

arduino ide 2.3.2

### Operating System

windows 11

### Flash frequency

80 mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

after update to version 3.0.0 no longer compile the sketch.
after go back to version 2.0.17 every thing works fine.

### Sketch

```cpp
/*
-*Ligação display a ESP32 30 pinos*
-*Pino LCD* -> *Pino ESP32*
- VCC  - 01 -> 5V
- GND  - 02 -> 0V
- CS   - 03 -> D15
- RESET- 04 -> D12
- DC   - 05 -> D04
- MOSI - 06 -> D23 (MOSI VSPI_MOSI)#
- SCK  - 07 -> D18 (SCK VSPI_CLK)#
- LED  - 08 -> NC 3.3V
- MISO - 09 -> NC
- T_CLK- 10 -> D18 (SCK VSPI_CLK)#
- T_CS - 11 -> D05 (VSPI_CS)
- T_DIN- 12 -> D23 (MOSI VSPI_MOSI)#
- T_DO - 13 -> D19 (MISO VSPI_MISO)
- T_IRQ- 14 -> D13
------------------------------------

*/
#include <TFT_eSPI.h>
#include <TFT_eWidget.h>  
#include <SPI.h>
#include <esp_now.h>
#include "INA3221.h"
[... 1589 lines cut ...]
      |             ^~~~
d:\fc\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:189:13: error: 'GPIO' was not declared in this scope
  189 |             GPIO.enable1_w1tc.val = ((uint32_t)1 << (pin - 32));
      |             ^~~~
d:\fc\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'void directModeOutput(uint32_t)':
d:\fc\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:213:13: error: 'GPIO' was not declared in this scope
  213 |             GPIO.enable_w1ts = ((uint32_t)1 << pin);
      |             ^~~~
d:\fc\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:215:13: error: 'GPIO' was not declared in this scope
  215 |             GPIO.enable1_w1ts.val = ((uint32_t)1 << (pin - 32));
      |             ^~~~

exit status 1

Compilation error: exit status 1
```

### Other Steps to Reproduce

deleted and reinstaled onewire library
same result

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-29:
It's an issue with the OneWire library. Check if there is an update or file an issue there so they add support for v3.0.0

closed 2024-05-29 by me-no-dev

Jason2866, 2024-05-29:
The used OneWire library is not compatible to Arduino core 3.0.0.
Not an Arduino core 3.0.0 issue

Paulo1963, 2024-06-02:
> Hi,
> I suspect you don't have the newest version of the OneWire library. Look here: PaulStoffregen/OneWire#134<https://github.com/PaulStoffregen/OneWire/pull/134>

Hello,
Everything last version, and working.
Thank alot
Best regards
Paulo