espressif/arduino-esp32 #9729: collect2.exe error with Arduino Nano ESP32 board

ednieuw, 2024-05-30:
### Board

Arduino Nano ESP32

### Device Description

Arduino Nano ESP32

### Hardware Configuration

NB Just compiling the Blink.ino sample

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 1.0 and 2.0

### Operating System

Windows 11

### Flash frequency

NB Arduino settings in board

### PSRAM enabled

yes

### Upload speed

NB Arduino setting in board

### Description

I upgraded the  esp32 board to Version 3.0.0. I still use the Arduino ESP32 board for developing. That compiles fine but still uses V2.0. 
To get the software ready for 3.0.0 and tried the ESP32 Nano in the ESP32 board with V3.0.0
But it does not even compile the Blink.Ino and get a: 
collect2.exe: error: ld returned 1 exit status

Using pin numbering by GPIO or Arduino default gives the same error 

### Sketch

```cpp
Blink.ino
```

### Debug Message

```plain
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\ednie\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\ednie\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\ednie\Documents\Files\Arduino\libraries -fqbn=esp32:esp32:nano_nora:PartitionScheme=default,PinNumbers=default,USBMode=default -ide-version=10819 -build-path C:\Users\ednie\AppData\Local\Temp\arduino_build_507669 -warnings=all -build-cache C:\Users\ednie\AppData\Local\Temp\arduino_cache_411953 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esp32-arduino-libs.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083 -prefs=runtime.tools.esp32-arduino-libs-idf-release_v5.1-442a798083.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083 -prefs=runtime.tools.esp-rv32.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.esp-rv32-2302.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.dfu-util.path=C:\Users\ednie\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\ednie\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.xtensa-esp-elf-gdb-12.1_20231023.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.openocd-esp32.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230921 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20230921.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230921 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-12.2.0_20230208.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-12.2.0_20230208.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.mkspiffs.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.esptool_py-4.6.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.riscv32-esp-elf-gdb-12.1_20231023.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-esp-12.2.0_20230208.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.mklittlefs.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\ednie\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\ednie\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\ednie\Documents\Files\Arduino\libraries -fqbn=esp32:esp32:nano_nora:PartitionScheme=default,PinNumbers=default,USBMode=default -ide-version=10819 -build-path C:\Users\ednie\AppData\Local\Temp\arduino_build_507669 -warnings=all -build-cache C:\Users\ednie\AppData\Local\Temp\arduino_cache_411953 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esp32-arduino-libs.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083 -prefs=runtime.tools.esp32-arduino-libs-idf-release_v5.1-442a798083.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083 -prefs=runtime.tools.esp-rv32.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.esp-rv32-2302.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esp-rv32\2302 -prefs=runtime.tools.dfu-util.path=C:\Users\ednie\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.dfu-util-0.11.0-arduino5.path=C:\Users\ednie\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.11.0-arduino5 -prefs=runtime.tools.xtensa-esp-elf-gdb.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.xtensa-esp-elf-gdb-12.1_20231023.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp-elf-gdb\12.1_20231023 -prefs=runtime.tools.openocd-esp32.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230921 -prefs=runtime.tools.openocd-esp32-v0.12.0-esp32-20230921.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\openocd-esp32\v0.12.0-esp32-20230921 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-esp-12.2.0_20230208.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.xtensa-esp32-elf-gcc-esp-12.2.0_20230208.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\esp-12.2.0_20230208 -prefs=runtime.tools.mkspiffs.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.esptool_py.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.esptool_py-4.6.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.6 -prefs=runtime.tools.riscv32-esp-elf-gdb.path=C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gdb\12.1_20231023 -prefs=runtim [... cut ...]
C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/md5builder.cpp:39: undefined reference to `(anonymous namespace)::HEXBuilder::hex2bytes(unsigned char*, unsigned int, char const*)'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Nano ESP32.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

JAndrassy, 2024-05-30:
it is already fixed https://github.com/espressif/arduino-esp32/pull/9722

ednieuw, 2024-05-30:
Top!