espressif/arduino-esp32 #9732: Use of sodium.h under new 3.0

HomeSpan, 2024-05-30:
### Board

ALL

### Device Description

N/A

### Hardware Configuration

N/A

### Version

v3.0.0

### IDE Name

Arduino IDE

### Operating System

MAC OS 14

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

N/A

### Description

I understand that Espressif has moved libsodium to an optional component when transitioning from IDF 4 to IDF 5.  Since Arduino-ESP32 version 3 uses IDF5, the sodium library is no longer available and `#include <sodium.h>` fails when added to an Arduino sketch.  Is there a recommended method of incorporating sodium into an Arduino sketch when using the new Arduino-ESP32 version 3?   The Espressif web site only includes information for doing so when using the IDF directly.

### Sketch

```cpp
N/A
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-31:
That would require custom libs to be built with sodium in them. @lucasssvaz can help with instructions on how to do that easily

VojtechBartoska, 2024-06-03:
closing as answered, if needed you can reopen the ticket.

HomeSpan, 2024-06-09:
To make sure I fully understand, if I create my own version of the library (with lib sodium) does this result in a custom Board Manager.  In other words, if anyone wants to use the library within the Arduino IDE, do they use the Board Manager functionality and point to where I'm hosting the custom library?  

Also, any reason you did not include sodium in 3.0?  Are there alternatives to the crypto commands that are built into a different library that is included in 3.0?

me-no-dev, 2024-06-10:
> any reason you did not include sodium in 3.0?

We do not have a measure of how much given IDF component is used in Arduino. Sodium was part of IDF prior to v5, so it came for free. If we get enough requests, we can consider adding it to the default libs.

> if anyone wants to use the library within the Arduino IDE, do they use the Board Manager functionality and point to where I'm hosting the custom library?

You would need to create and host your own board manager json, with the libs you built yourself

Nezaemmy, 2024-06-11:
> > any reason you did not include sodium in 3.0?
> 
> We do not have a measure of how much given IDF component is used in Arduino. Sodium was part of IDF prior to v5, so it came for free. If we get enough requests, we can consider adding it to the default libs.
> @me-no-dev if no other crypto commands library built in v3 sodium is great libary to get on v3
> > if anyone wants to use the library within the Arduino IDE, do they use the Board Manager functionality and point to where I'm hosting the custom library?
> 
> You would need to create and host your own board manager json, with the libs you built yourself

PR #9821 (2024-06-13): feat(crypto): Add libsodium to the included components
Fixes: #9732
closed 2024-06-13 by me-no-dev