espressif/arduino-esp32 #9733: How to do cystom partition scheme correctly?

anthonyhoegberg, 2024-05-31:
### Board

T-SIMCAM ESP32s3

### Device Description

ESP32 S3 16MB flash

### Hardware Configuration

Not relevant

### Version

v2.0.17

### IDE Name

ESP-IDF vscode

### Operating System

Windows

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

default

### Description

So in menuconfig i see there are 2 places to define partition schema
and i want to use my own custom schema however i dont know how to set the arduino configuration to use the same schema etc
if i change to my own schema using partition table settings it will give me NVS errors on boot, also does arduino library include its own nvs wifi flash? bc if i have it enabled WiFi dont work but disabled it does (might be a seperate question)

So my main question is how do i correctly let arduino component know my partition format that i want to use that is custom

Using esp-idf in vscode with arduino-esp32 as component

[image]

[image]

### Sketch

```cpp
Mot related
```

### Debug Message

```plain
Not related
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-05-31:
You add the partition table to the sketch folder. Name it `partitions.csv` and will be picked up by the build. You partition scheme must start the same way as ours and the first bootable partition must be at the same address, the rest is up to you.

closed 2024-05-31 by me-no-dev

anthonyhoegberg, 2024-05-31:
Im not using the arduino ide tho, do i leave the file inside the src folder of esp-idf project or in root, and dont i need to set the arudino component configuration aswell?

me-no-dev, 2024-05-31:
It should be in the root of the project and selected in menuconfig. It should still have nvs partition for Arduino to work