espressif/arduino-esp32 #9736: Simultaneous BLE causes WiFiMulti.run to fail on 3.0.0

PaulZC, 2024-05-31:
### Board

ESP32-WROOM-32E but also ESP32-WROVER-IE with PSRAM

### Device Description

SparkFun ESP32 Thing Plus C

### Hardware Configuration

No

### Version

v3.0.0

### IDE Name

Arduino IDE 1.8.19 - but also seen with the CLI

### Operating System

Windows

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

We have been using simultaneous BLE and WiFi successfully for at least two years, on arduino-esp32 <= 2.0.11. We are migrating to 3.0.0 and have noticed that having BLE enabled causes WiFiMulti.run to fail to connect... We've distilled the issue into the following sketch. Change START_BLE to 0 to disable BLE see the code run normally. 

### Sketch

```cpp
#define START_BLE 1

#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti wifiMulti;
#define WIFI_SSID "<ADD_YOUR_WIFI_SSID_HERE>"
#define WIFI_PASSWORD "<ADD_YOUR_WIFI_PASSWORD_HERE>"

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

void setup() {
  delay(1000);
  Serial.begin(115200);

#if (START_BLE)
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic =
[... 481 lines cut ...]
13:33:48.581 -> [ 14865][D][WiFiMulti.cpp:151] run():        2: [6][40:0D:10:48:45:51] VM6184319 (-86) *
13:33:48.581 -> [ 14873][D][WiFiMulti.cpp:151] run():        3: [1][58:CB:52:D8:49:F7] VM6184319 (-90) *
13:33:48.628 -> [ 14881][D][WiFiMulti.cpp:151] run():        4: [1][C0:05:C2:46:C3:59] VM1245360 (-93) *
13:33:48.628 -> [ 14889][D][WiFiMulti.cpp:151] run():        5: [11][D4:35:1D:17:49:49] vodafone174949 (-95) *
13:33:48.628 -> [ 14897][I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 18:35:D1:9D:ED:09 SSID: VM4924742 Channel: 11 (-73)
13:33:48.628 -> [ 14909][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
13:33:48.722 -> [ 14992][V][WiFiGeneric.cpp:356] _arduino_event_cb(): STA Connected: SSID: VM4924742, BSSID: 18:35:d1:9d:ed:09, Channel: 11, Auth: WPA2_PSK
13:33:48.722 -> [ 15005][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
13:33:48.909 -> [ 15159][V][WiFiGeneric.cpp:370] _arduino_event_cb(): STA Got Same IP:192.168.0.88
13:33:48.909 -> [ 15167][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
13:33:48.909 -> [ 15174][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 192.168.0.88, MASK: 255.255.255.0, GW: 192.168.0.1
13:33:48.909 -> [ 15184][I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
13:33:48.909 -> [ 15190][D][WiFiMulti.cpp:175] run(): [WIFI] SSID: VM4924742
13:33:48.909 -> [ 15195][D][WiFiMulti.cpp:176] run(): [WIFI] IP: 192.168.0.88
13:33:48.909 -> [ 15201][D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 18:35:D1:9D:ED:09
13:33:48.956 -> [ 15207][D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 11
13:33:48.956 -> 
13:33:48.956 -> WiFi connected with IP address: 192.168.0.88
13:33:48.956 -> 
13:33:48.956 ->
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

PaulZC, 2024-05-31:
It has to be something really really subtle with the way BLE and WiFiMulti are instantiated and initialized. This code works, even though it should be functionally identical to the non-working example above:

```
#define GLOBAL_WIFIMULTI 1

#define START_WIFIMULTI_BEFORE_BLE 0

#define START_BLE 1

#include <WiFi.h>
#include <WiFiMulti.h>

#if (GLOBAL_WIFIMULTI)
WiFiMulti wifiMulti;
#endif

#define WIFI_SSID "<ADD_YOUR_WIFI_SSID_HERE>"
#define WIFI_PASSWORD "<ADD_YOUR_WIFI_PASSWORD_HERE>"

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

void setup() {
  delay(1000);
  Serial.begin(115200);

#if (START_WIFIMULTI_BEFORE_BLE)
  startWiFiMulti();
#endif

#if (START_BLE)
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic =
    pService->createCharacteristic(CHARACTERISTIC_UUID, BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_WRITE);

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!\r\n");

  delay(1000);
#endif
}

void loop() {
  static int attempt = 0;

  Serial.printf("\r\nLoop start : attempt %d\r\n\r\n", ++attempt);

  startWiFiMulti();
}

void startWiFiMulti() {  
  Serial.println("\r\nLoop start\r\n");
  
  WiFi.mode(WIFI_STA);
[... 405 lines cut ...]
[ 45508][D][WiFiMulti.cpp:233] run():         1: [6][58:CB:52:D8:5E:1E] VM6184319 (-88) (*) (visible)
[ 45517][D][WiFiMulti.cpp:233] run():         2: [1][58:CB:52:D8:49:F7] VM6184319 (-89) (*) (visible)
[ 45527][D][WiFiMulti.cpp:233] run():         3: [6][40:0D:10:48:45:51] VM6184319 (-93) (*) (visible)
[ 45536][D][WiFiMulti.cpp:233] run():         4: [11][7A:83:94:6B:6D:FC] EE WiFi (-94) ( ) (visible)
[ 45545][D][WiFiMulti.cpp:233] run():         5: [11][8C:83:94:6B:6D:FB] BT-XPFH75 (-94) (*) (visible)
[ 45554][D][WiFiMulti.cpp:233] run():         6: [11][98:AA:FC:36:F8:2B] Wi-Fi Socket (-96) ( ) (visible)
[ 45563][V][WiFiMulti.cpp:239] run(): foundCount = 1, failCount = 0
[ 45570][I][WiFiMulti.cpp:249] run(): [WIFI] Connecting BSSID: 18:35:D1:9D:ED:09 SSID: VM4924742 Channel: 11 (-75)
[ 45580][W][STA.cpp:533] disconnect(): STA already disconnected.
[ 45665][V][STA.cpp:204] _onStaEvent(): STA Connected: SSID: VM4924742, BSSID: 18:35:d1:9d:ed:09, Channel: 11, Auth: WPA2_PSK
[ 45677][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 13 - STA_CONNECTED
[ 45685][D][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 13 - STA_CONNECTED
[ 45902][V][NetworkInterface.cpp:66] _onIpEvent(): sta Got New IP: 192.168.0.88 MASK: 255.255.255.0 GW: 192.168.0.1
[ 45913][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 16 - STA_GOT_IP
[ 45920][D][STA.cpp:110] _onStaArduinoEvent(): Arduino STA Event: 16 - STA_GOT_IP
[ 45928][D][STA.cpp:169] _onStaArduinoEvent(): STA IP: 192.168.0.88, MASK: 255.255.255.0, GW: 192.168.0.1
[ 45937][I][WiFiMulti.cpp:272] run(): [WIFI] Connecting done.
[ 45943][D][WiFiMulti.cpp:273] run(): [WIFI] SSID: VM4924742
[ 45949][D][WiFiMulti.cpp:274] run(): [WIFI] IP: 192.168.0.88
[ 45954][D][WiFiMulti.cpp:275] run(): [WIFI] MAC: 18:35:D1:9D:ED:09
[ 45961][D][WiFiMulti.cpp:276] run(): [WIFI] Channel: 11
[ 45967][D][WiFiMulti.cpp:334] resetFails(): [WIFI] Resetting failure flags

WiFi connected with IP address: 192.168.0.88
```

me-no-dev, 2024-05-31:
what actually happens is that for some reason the first sketch takes too long to scan (same on my end) and the second scans in about 8 seconds. Timeout is set to 10 seconds, so it timeouts before results come. I will add a fix for the timeout to WiFiMulti

me-no-dev, 2024-05-31:
@PaulZC you can try this fix: https://github.com/espressif/arduino-esp32/pull/9738/files

PaulZC, 2024-05-31:
@me-no-dev : sincere thanks for the very fast fix!

The fix works for me - both in my simple example and in our much larger product firmware.

Thanks again - and have a great weekend,
Paul

Just for information - for anyone else reading this issue:
* With this fix, the default WiFiMulti.run scan timeout is now 60 seconds - it was 10 seconds and slightly too short
* If needed, you can adjust the timeout with e.g. ```WiFi.setScanTimeout(20000); // Set scan timeout to 20s```

closed 2024-06-03 by me-no-dev