espressif/arduino-esp32 #9739: RTC_CPU_FREQ_80M was not declared in this scope

4ngeliwi, 2024-05-31:
### Board

TTGO T1

### Device Description

HX711

### Hardware Configuration

ttgo, wx711 and a rs485 but this isnt used

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 22

### Flash frequency

not sure

### PSRAM enabled

no

### Upload speed

115200

### Description

Code doesnt run on my pc, i tried running it on other peoples pc and it works, its just mine, i changed versions of the arduino and the libraries but nothing seems to work.

### Sketch

```cpp
rtc_cpu_freq_config_t config;
  rtc_clk_cpu_freq_get_config(&config);
  rtc_clk_cpu_freq_to_config(RTC_CPU_FREQ_80M, &config);
  rtc_clk_cpu_freq_set_config_fast(&config);
  scale.begin(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);
  //scale.set_scale(INSERT YOUR CALIBRATION FACTOR);
  scale.set_scale(213.510);
  scale.tare();               // reset the scale to 0
  scale.power_up();
```

### Debug Message

```plain
exit status 1

'RTC_CPU_FREQ_80M' was not declared in this scope; did you mean 'RTC_XTAL_FREQ_40M'?
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2024-06-03:
@lucasssvaz can you please help with triage? I am not sure, if there is potentially some issue. Thanks

lucasssvaz, 2024-10-09:
Duplicate of https://github.com/espressif/arduino-esp32/issues/7616

closed 2024-10-09 by lucasssvaz