espressif/arduino-esp32 #9740: I2S_Record_to_Wav example error

raprakashvi, 2024-05-31:
### Board

ESP32

### Device Description

ESP-WROOM-32

### Hardware Configuration

SD Card Standard INPUT 

### Version

latest master (checkout manually)

### IDE Name

Arduino 

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Testing the microphone and SD card from the example and everytime the file is created but with 0 bytes. Seems like the record to wav function is not waiting for the process to finish. 

### Sketch

```cpp
/*
  ESP32-S2-EYE I2S record to WAV example
  This simple example demonstrates using the I2S library to record
  5 seconds of audio data and write it to a WAV file on the SD card.

  Don't forget to select the OPI PSRAM, 8MB flash size and Enable USB CDC
  on boot in the Tools menu!

  Created for arduino-esp32 on 18 Dec, 2023
  by Lucas Saavedra Vaz (lucasssvaz)
*/

#include "ESP_I2S.h"
#include "FS.h"
#include "SD_MMC.h"

#define ONE_BIT_MODE   true

const uint8_t I2S_SCK = 21 ;
const uint8_t I2S_WS = 22;
const uint8_t I2S_DIN = 25;

const uint8_t SD_CMD = 15;
const uint8_t SD_CLK = 14;
const uint8_t SD_DATA0 = 2;

void setup() {
  // Create an instance of the I2SClass
  I2SClass i2s;

  pinMode(2, INPUT_PULLUP);
  pinMode(4,INPUT_PULLUP);
  pinMode(12, INPUT_PULLUP);
  pinMode(13,INPUT_PULLUP);
  pinMode(15, INPUT_PULLUP);
    
  // Create variables to store the audio data
  uint8_t *wav_buffer;
  size_t wav_size;

  // Initialize the serial port
  Serial.begin(115200);
  while (!Serial) {
    delay(10);
  }

  Serial.println("Initializing I2S bus...");

  // Set up the pins used for audio input
  i2s.setPins(I2S_SCK, I2S_WS, -1, I2S_DIN);

  // Initialize the I2S bus in standard mode
  if (!i2s.begin(I2S_MODE_STD, 16000, I2S_DATA_BIT_WIDTH_32BIT, I2S_SLOT_MODE_MONO, I2S_STD_SLOT_LEFT)) {
    Serial.println("Failed to initialize I2S bus!");
    return;
  }

  Serial.println("I2S bus initialized.");
  Serial.println("Initializing SD card...");

  // Set up the pins used for SD card access
  if (!SD_MMC.setPins(SD_CLK, SD_CMD, SD_DATA0)) {
    Serial.println("Failed to set SD pins!");
    return;
  }

  // Mount the SD card
  if (!SD_MMC.begin("/sdcard", true)) {
    Serial.println("Failed to initialize SD card!");
    return;
  }

  Serial.println("SD card initialized.");
  Serial.println("Recording 5 seconds of audio data...");

  // Record 5 seconds of audio data
  wav_buffer = i2s.recordWAV(5, &wav_size);

  // Create a file on the SD card
  File file = SD_MMC.open("/test.wav", FILE_WRITE);
  if (!file) {
    Serial.println("Failed to open file for writing!");
    return;
  }

  Serial.println("Writing audio data to file...");

  // Write the audio data to the file
  if (file.write(wav_buffer, wav_size) != wav_size) {
    Serial.println("Failed to write audio data to file!");
    return;
  }

  // Close the file
  file.close();

  Serial.println("Application complete.");
}

void loop() {}
```

### Debug Message

```plain
Standard output with but file is empty.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lucasssvaz, 2024-05-31:
Can you please enable the verbose debug mode and post the output here ?

raprakashvi, 2024-06-03:
Hi @lucasssvaz , I enabled the verbose mode during compiling and upload and below is the output. 

> 
FQBN: esp32:esp32:esp32:UploadSpeed=115200
Using board 'esp32' from platform in folder: /Users/rp/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0
Using core 'esp32' from platform in folder: /Users/rp/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0

/usr/bin/env bash -c "[ ! -f \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/.arduinoIDE-unsaved202452-52285-1m8vjng.xp4k/sketch_jun2a\"/partitions.csv ] || cp -f \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/.arduinoIDE-unsaved202452-52285-1m8vjng.xp4k/sketch_jun2a\"/partitions.csv \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504\"/partitions.csv ] || [ ! -f \"/Users/rp/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/variants/esp32\"/partitions.csv ] || cp \"/Users/rp/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/variants/esp32\"/partitions.csv \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504\"/partitions.csv ] || cp \"/Users/rp/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0\"/tools/partitions/default.csv \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504\"/partitions.csv"
/usr/bin/env bash -c "[ -f \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/.arduinoIDE-unsaved202452-52285-1m8vjng.xp4k/sketch_jun2a\"/bootloader.bin ] && cp -f \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/.arduinoIDE-unsaved202452-52285-1m8vjng.xp4k/sketch_jun2a\"/bootloader.bin \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504\"/sketch_jun2a.ino.bootloader.bin || ( [ -f \"/Users/rp/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/variants/esp32\"/bootloader.bin ] && cp \"/Users/rp/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/variants/esp32\"/bootloader.bin \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504\"/sketch_jun2a.ino.bootloader.bin || \"/Users/rp/Library/Arduino15/packages/esp32/tools/esptool_py/4.6\"/esptool --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504\"/sketch_jun2a.ino.bootloader.bin \"/Users/rp/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32\"/bin/bootloader_qio_80m.elf )"
esptool.py v4.6
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
/usr/bin/env bash -c "[ ! -f \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/.arduinoIDE-unsaved202452-52285-1m8vjng.xp4k/sketch_jun2a\"/build_opt.h ] || cp -f \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/.arduinoIDE-unsaved202452-52285-1m8vjng.xp4k/sketch_jun2a\"/build_opt.h \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504\"/build_opt.h"
/usr/bin/env bash -c "[ -f \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504\"/build_opt.h ] || : > \"/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504\"/build_opt.h"
/usr/bin/env bash -c ": > '/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/arduino/sketches/A9F4B938E4DB0D8DA7FC0C29D7AD1504/file_opts'"
Detecting libraries used...
/Users/rp/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/xtensa-esp32-elf-g++ -c @/Users/rp/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/cpp_flags -w -Og -g3 -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV" -DARDUINO_VARIANT="esp32" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="macosx" -DARDUINO_FQBN="esp32:esp32:esp32:UploadSpeed=115200,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @/Users/rp/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/defines -I/private/var/folders/sn/_2pds5t53sz7gl134mklfs280000gn/T/.arduinoIDE-unsaved202452-52285-1m8vjng.xp4k/sketch_jun2a -iprefix /Users/rp/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/include/ @/Users/rp/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/flags/includes -I/Users/rp/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-442a798083/esp32/qio_qspi/include -I/Users/rp/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32 -I/Users/rp/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/variants/esp32 @/private/var/folders/sn/_2pds5t53sz7gl134mklfs [... cut ...]
Writing at 0x00055929... (84 %)
Writing at 0x0005e511... (92 %)
Writing at 0x00065763... (100 %)
Wrote 372752 bytes (212826 compressed) at 0x00010000 in 19.0 seconds (effective 157.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Serial Monitor: 
23:26:26.580 -> I2S bus initialized.
23:26:26.580 -> Initializing SD card...
23:26:26.645 -> SD card initialized.
23:26:26.678 -> Recording 5 seconds of audio data...
23:26:26.678 -> Writing audio data to file...
23:26:26.678 -> Application complete.

me-no-dev, 2024-06-03:
we need debug enabled in the core (for your board) not for compiling and uploading in the IDE

VojtechBartoska, 2024-06-19:
@raprakashvi can you please provide us more information as stated above? thanks.

raprakashvi, 2024-06-19:
@VojtechBartoska I am having trouble figuring out the steps to enter debug mode and hence the delay, would appreciate pointers

lucasssvaz, 2024-06-20:
@raprakashvi select this option in the menu and upload the code to the board again

[image]

VojtechBartoska, 2024-06-25:
any updates on this @raprakashvi?

raprakashvi, 2024-06-25:
Hi @VojtechBartoska thanks a lot for your patience. Here is the code and verbose output. Seems buffer size issue. 

> Initializing I2S bus...
I2S bus initialized.
Initializing SD card...
SD card initialized.
Recording 5 seconds of audio data...
Writing audio data to file...
Application complete.
an.c:160] perimanSetPinBus(): Pin 15 successfully set to type SDMMC_CMD (42) with bus 0x3ffc1150
[  1065][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 14 successfully set to type SDMMC_CLK (41) with bus 0x3ffc1150
[  1076][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 2 successfully set to type SDMMC_D0 (43) with bus 0x3ffc1150
SD card initialized.

Recording 5 seconds of audio data...
[  1087][D][ESP_I2S.cpp:965] recordWAV(): Record WAV: rate:16000, bits:32, channels:1, size:320000
[  1101][E][ESP_I2S.cpp:969] recordWAV(): Failed to allocate WAV buffer with size 320044
Writing audio data to file...
Application complete.
[  1116][E][esp32-hal-periman.c:122] perimanSetPinBus(): Invalid pin: 255
[  1124][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 21 successfully set to type INIT (0) with bus 0x0
[  1134][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 22 successfully set to type INIT (0) with bus 0x0
[  1144][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 25 successfully set to type INIT (0) with bus 0x0
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   379728 B ( 370.8 KB)  
 Free Bytes        :   318308 B ( 310.8 KB)
  Allocated Bytes   :    53684 B (  52.4 KB)
  Minimum Free Bytes:   301768 B ( 294.7 KB)
  Largest Free Block:   110580 B ( 108.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     1 : UART_TX[0]
     2 : SDMMC_D0
     3 : UART_RX[0]
     4 : GPIO
    12 : GPIO
    13 : GPIO
    14 : SDMMC_CLK
    15 : SDMMC_CMD
============ After Setup End =============

me-no-dev, 2024-06-25:
You need PSRAM to record that much data with this function. It's trying to allocate 320KB buffer and that is more than you can do on internal RAM

raprakashvi, 2024-06-25:
I see. Then it won't work with my current ESP Wroom 32 boards as they only have up to 4 MB flash and the PSRAM shows 0. Seems like S3 boards might be better for testing this function. Meanwhile, I did get the recording and writing part working where I wrote a custom function to record and save the Wav file in real time to SD card. Might be a good idea to add that as a function option in the future in the official implementation as the MMC is already being used,

VojtechBartoska, 2024-09-16:
Hello @raprakashvi, can I consider this ticket as solved?

raprakashvi, 2024-09-16:
Yes. This part is solved. I am happy to help if anyone has any questions on
this or related issue.

closed 2024-09-17 by VojtechBartoska