espressif/arduino-esp32 #9743: some keys cannot use with ESP32-S3

53175ddd, 2024-06-01:
### Board

ESP32-Dev-Module

### Device Description

My original Keyboard prototype with ESP32-S3-WROOM-1

[image]

[image]

### Hardware Configuration

In my Keyboard, GPIO 10, 11, 12, 9, 46, 3 are used as Scan lines and GPIO 42, 41, 40, 39, 38, 37, 36, 35, 45, 48, 47, 21, 14, 13, 8, 2, 1 are used as Read lines. Read lines are pulluped.

- Left Ctrl key is located between GPIO 42 and 3
- Left Shift key is located between GPIO 42 and 46
- Left Win key is located between GPIO 41 and 3
- Left Alt key is located between GPIO 40 and 3
- Right Ctrl key is located between GPIO 13 and 3
- Right Shift key is located between GPIO 13 and 46
- Right Alt key is located between GPIO 14 and 3

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.3.2

### Operating System

Windows 11

### Flash frequency

80 MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Both Ctrl, Alt, Shift, Win keys on my keyboard do not work independently.
Like this : When I pressing the Ctrl key, did not show any response on the keyboard test

Keyboard test site : https://www.onlinemictest.com/ja/keyboard-test/

But, when I use this keyboard with BLE connection, they worked.

I would expect all keys to respond when pressed independently of each other.

### Sketch

```cpp
https://gist.github.com/53175ddd/0d95b3d494cd21595637ed0d233a8b87
```

### Debug Message

```plain
https://gist.github.com/53175ddd/7e271aee10666b8f2d6727d983181fdc
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

53175ddd, 2024-06-01:
I have mistake
I don't use PSRAM

53175ddd, 2024-06-01:
This is a schematic
[HT-Keys-TKL_v2.pdf](https://github.com/user-attachments/files/15520746/HT-Keys-TKL_v2.pdf)

me-no-dev, 2024-06-01:
it's not clear exactly which type of S3 you are using. If it has on-board OPI PSRAM, then GPIOs 35-37 are not to be used. Please enable debug to level Verbose and post the output of UART

53175ddd, 2024-06-01:
Thank you for your reply.
In my sketch, each PSRAM option is disabled. (image below)

[image]

this is an output with Verbose level debug
```
[   483][V][BLEDevice.cpp:76] createServer(): >> createServer
[   489][V][BLEServer.cpp:291] registerApp(): >> registerApp - 0
[   495][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegisterAppEvt (0x3fcac5e0), owner: <N/A> for registerApp
[   505][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegisterAppEvt (0x3fcac5e0), owner: registerApp
[   515][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: RegisterAppEvt (0x3fcac5e0), owner: registerApp for registerA
15][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_REG_EVT
[   537][V][BLEUtils.cpp:1530] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
[   545][V][BLEUtils.cpp:1673] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
[   553][V][BLEServer.cpp:144] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
[   562][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: RegisterAppEvt (0x3fcac5e0), owner: registerApp
[   571][V][BLEServer.cpp:281] handleGATTServerEv[e n t (5)7:1 ]<[<V ]h[aFnrdeleeRGTAOTST.Sceprpv:e7r7Etv ewnati
 (): << wait: Semaphore released: name: RegisterAppEvt (0x3fcac5e0), owner: <N/A>
[   589][V][BLEServer.cpp:295] registerApp(): << registerApp
[   594][V][BLEDevice.cpp:83] createServer(): << createServer
[   600][V][BLEServer.cpp:67] createService(): >> createService - 0000180a-0000-1000-8000-00805f9b34fb
[   609][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3fcac884), owner: <N/A> for createService
[   619][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3fcac884), owner: createService
[   629][V][BLEService.cpp:60] executeCreate(): >> executeCreate() - Creating service (esp_ble_gatts_create_service) service uuid: 0000180a-0000-1000-8000-00805f9b34fb
[   643][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3fcac988), owner: <N/A> for executeCreate
[   654][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3fcac988), owner: executeCreate
[   663][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: CreateEvt (0x3fcac988), owner: executeCreate for executeCreate[
                                 664][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_CREATE_EVT
[   686][V][BLEUtils.cpp:1530] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CREATE_EVT
[   694][V][BLEUtils.cpp:1599] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 40 0x28, service_id: [uuid: 0000180a-0000-1000-8000-00805f9b34fb, inst_id: 0]]
[   709][V][BLEServer.cpp:144] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
[   718][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: CreateEvt (0x3fcac884), owner: createService
[   728][V][BLEService.cpp:194] setHandle(): >> setHandle - Handle=0x28, service UUID=0000180a-0000-1000-8000-00805f9b34fb)
[   739][V][BLEService.cpp:200] setHandle(): << setHandle
[   744][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: CreateEvt (0x3fcac988), owner: executeCreate
[   753][V][BLEServer.cpp:281] handle[G A T T7S5e3r]v[eVr]E[vFernete(R)T:O S<.<c phpa:n7d7l]e GwAaTiTtS:e)r:v e<r<E vweanitt
  Semaphore released: name: CreateEvt (0x3fcac988), owner: <N/A>
[   770][V][BLEService.cpp:76] executeCreate(): << executeCreate
[   776][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: CreateEvt (0x3fcac884), owner: <N/A> for createService
[   787][V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: CreateEvt (0x3fcac884), owner: <N/A>
[   797][V][BLEServer.cpp:83] createService(): << createService
[   803][V][BLEServer.cpp:67] createService(): >> createService - 00001812-0000-1000-8000-00805f9b34fb
[   812][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3fcac884), owner: <N/A> for createService
[   822][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3fcac884), owner: createService
[   832][V][BLEService.cpp:60] executeCreate(): >> executeCreate() - Creating service (esp_ble_gatts_create_service) service uuid: 00001812-0000-1000-8000-00805f9b34fb
[   846][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: CreateEvt (0x3fcad24c), owner: <N/A> for executeCreate
[   857][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: CreateEvt (0x3fcad24c), owner: executeCreate
[   866][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: CreateEvt (0x3fcad24c), owner: executeCreate for executeCreate
  867][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_CREATE_EVT
[   889][V][BLEUtils.cpp:1530] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CREATE_EVT
[   897][V][BLEUtils.cpp:1599] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 55 0x37, service_id: [uuid: 00001812-0000-1000-8000-00805f9b34fb, inst_id: 0]]
[   912][V][BLEServer.cpp:144] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
[   921][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: CreateEvt (0x3fcac884), owner: createService
[   931][V][BLEService.cpp:194] setHandle(): >> setHandle - Handle=0x37, service UUID=00001812-0000-1000-8000-00805f9b34fb)
[   942][V][BLEService.cpp:200] setHandle(): << setHandle
[... cut ...]
[  8483][V][GeneralUtils.cpp:323] hexDump(): 0000 64                                               d
[  8492][V][BLECharacteristic.cpp:500] notify(): << notify: No connected clients.
[  8499][D][BLECharacteristic.cpp:796] onStatus(): >> onStatus: default
[  8506][D][BLECharacteristic.cpp:797] onStatus(): << onStatus
[  8511][D][BleKeyboard.cpp:137] begin(): [] Advertising started!
[  8517][D][esp32-hal-tinyusb.c:569] tinyusb_load_enabled_interfaces(): Load Done: if_num: 1, descr_len: 41, if_mask: 0x4
```

me-no-dev, 2024-06-01:
there were some prints before and after setup, showing your chip, config, etc. We need those :)

53175ddd, 2024-06-02:
So, what should I do ?
Do I have to add some codes ?

SuGlider, 2024-06-04:
>Both Ctrl, Alt, Shift, Win keys on my keyboard do not work independently.
>Like this : When I pressing the Ctrl key, did not show any response on the keyboard test
>But, when I use this keyboard with BLE connection, they worked.

if the sketch executes `usbkbd.pressRaw(HID_KEY_SHIFT_LEFT);`, does it show up in the keyboard test?

https://github.com/hathach/tinyusb/blob/master/src/class/hid/hid.h#L578-L586
```
// RESERVED					                        0xDE-DF
#define HID_KEY_CONTROL_LEFT                0xE0
#define HID_KEY_SHIFT_LEFT                  0xE1
#define HID_KEY_ALT_LEFT                    0xE2
#define HID_KEY_GUI_LEFT                    0xE3
#define HID_KEY_CONTROL_RIGHT               0xE4
#define HID_KEY_SHIFT_RIGHT                 0xE5
#define HID_KEY_ALT_RIGHT                   0xE6
#define HID_KEY_GUI_RIGHT                   0xE7
```

SuGlider, 2024-06-04:
What exactly Arduino Core version is this project using?
There is a FIX to USB Keyboard that must be in place for the Key Modifiers (SHIFT, CTRL, ALT etc)
https://github.com/espressif/arduino-esp32/pull/9473
This Fix is available in the Arduino Core 3.0.0-RC1 and forward.

53175ddd, 2024-06-04:
I'm using 3.0.0 (not RC).

53175ddd, 2024-06-04:
@SuGlider 
I tried this test.
> if the sketch executes usbkbd.pressRaw(HID_KEY_SHIFT_LEFT);, does it show up in the keyboard test?

I added this code at the end of `setup()`. 
```c
  while(true) {
    usbkbd.pressRaw(HID_KEY_SHIFT_LEFT);
    delay(1000);
    usbkbd.releaseRaw(HID_KEY_SHIFT_LEFT);
    delay(1000);
  }
```

In the keyboard test, the left shift key is being pressed and released.

Should I replace `usbkbd.press()` to `usbkbd.pressRaw()` ?

SuGlider, 2024-06-04:
>Should I replace usbkbd.press() to usbkbd.pressRaw() ?

Yes, it is possible, but the parameters are different for `press()` and `pressRaw()`.
`usbkbd.pressRaw(HID_Key_Code); // takes a HID Key Code Key as parameter.`
`usbkbd.press(ASCII_Char); // takes an ASCII code as parameter.`

HID Key Codes can be seen here: 
https://github.com/hathach/tinyusb/blob/master/src/class/hid/hid.h#L366-L587
or here:
https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2

The sketch you have presented uses ASCII as key code.
You may need to change it in https://gist.github.com/53175ddd/0d95b3d494cd21595637ed0d233a8b87#file-sketch-ino-L130-L218

More information in USB ORG documentation (HID Codes in Chapter 10)
https://usb.org/sites/default/files/hut1_21.pdf

53175ddd, 2024-06-04:
Thank you so much. I understood.
I change my code like this.

```cpp
#define Key_L_Shift     0xE1
#define Key_R_Shift     0xE5
#define Key_L_Ctrl      0xE0
#define Key_L_Win       0xE3
#define Key_L_Alt       0xE2
#define Key_R_Alt       0xE6
#define Key_R_Win       0xE7
#define Key_R_Ctrl      0xE4
```

```cpp
void keyOut(const uint8_t key, const bool mode) {
  if(key == 0) {
  }else {
    if(mode) {
      if(digitalRead(MODE_SELECT) == HIGH) {
        if((0xE0 <= key) && (key <= 0xE7)) {
          usbkbd.pressRaw(key);
        }else {
          usbkbd.press(key);
        }
      }else {
        blekbd.press(key);
      }
    }else {
      if(digitalRead(MODE_SELECT) == HIGH) {
        if((0xE0 <= key) && (key <= 0xE7)) {
          usbkbd.releaseRaw(key);
        }else {
          usbkbd.release(key);
        }
      }else {
        blekbd.release(key);
      }
    }
  }
}
```

Then my keyboard works well !!

VojtechBartoska, 2024-06-04:
Marking as solved and closing :)

closed 2024-06-04 by VojtechBartoska

53175ddd, 2024-06-04:
Thank you so much everyone !