espressif/arduino-esp32 #9744: Error compiling, works good on 2.0.14 but cannot compile and terminates on 3.0.0

TobbeG, 2024-06-01:
### Board

ESP32-S3 (Elecrow ESP32-Terminal)

### Device Description

Elecrow ESP32-Terminal

### Hardware Configuration

Standard device Elecrow ESP32-S3 with TFT display and Touch

### Version

v3.0.0

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Cannot compile - Compilation terminates

### Sketch

```cpp
The error message indicates that the compiler is unable to find the bits/c++allocator.h header file, which is part of the standard C++ library implementation.
```

### Debug Message

```plain
cmd /c type nul > "C:\\Users\\THORBJ~1\\AppData\\Local\\Temp\\arduino_build_956191/file_opts"
Detecting libraries used...
"C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++" -c "@C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/flags/cpp_flags" -w -Os -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10819 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32S3_DEV\"" "-DARDUINO_VARIANT=\"esp32s3\"" -DARDUINO_PARTITION_default_8MB "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=opi,FlashMode=qio,FlashSize=16M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default_8MB,CPUFreq=240,UploadSpeed=921600,DebugLevel=none,EraseFlash=none,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/flags/defines" "-IC:\\Users\\Thorbjörn Geiser O\\Dropbox\\MAS\\Ohmigo.server_3.1.1" -iprefix "C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/include/" "@C:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/flags/includes" "-IC:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/qio_opi/include" "-IC:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\cores\\esp32" "-IC:\\Users\\Thorbjörn Geiser O\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0\\variants\\esp32s3" "@C:\\Users\\THORBJ~1\\AppData\\Local\\Temp\\arduino_build_956191/build_opt.h" "@C:\\Users\\THORBJ~1\\AppData\\Local\\Temp\\arduino_build_956191/file_opts" "C:\\Users\\THORBJ~1\\AppData\\Local\\Temp\\arduino_build_956191\\sketch\\Ohmigo.server_3.1.1.ino.cpp" -o nul
Alternatives for bits/c++allocator.h: []
ResolveLibrary(bits/c++allocator.h)
  -> candidates: []
In file included from c:\users\thorbj�rn geiser o\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208\xtensa-esp32s3-elf\include\c++\12.2.0\unordered_map:40,
                 from c:\users\thorbj�rn geiser o\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208\xtensa-esp32s3-elf\include\c++\12.2.0\functional:61,
                 from C:\Users\Thorbj�rn Geiser O\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/HardwareSerial.h:49,
                 from C:\Users\Thorbj�rn Geiser O\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/Arduino.h:200,
                 from C:\Users\Thorbjörn Geiser O\Dropbox\MAS\Ohmigo.server_3.1.1\Ohmigo.server_3.1.1.ino:209:
c:\users\thorbj�rn geiser o\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208\xtensa-esp32s3-elf\include\c++\12.2.0\bits\allocator.h:46:10: fatal error: bits/c++allocator.h: No such file or directory
   46 | #include <bits/c++allocator.h> // Define the base class to std::allocator.
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board ESP32S3 Dev Module
```

### Other Steps to Reproduce

Everthing worked fine on 2.0.14

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-01:
You are hitting a rather nasty problem with Windows, the new toolchains and the length of your username. Windows has a limit of path length of 240/256 chars and when your username is longer, combined with the required path for some headers in the toolchain, you exceed this limit. Resulting error is like this, which does not really point to the issue.
Your only current way is to shorten your Windows username to `Thorbjörn` or shorter

TobbeG, 2024-06-01:
I thought so.... any info on how to change username on windows 11 accounts?

...or do I need to create a new user account?

engperini, 2024-06-02:
> You are hitting a rather nasty problem with Windows, the new toolchains and the length of your username. Windows has a limit of path length of 240/256 chars and when your username is longer, [...]

Hi. I had the same issue,  so I am going to try this solution,  about changing the installation for a portable folder. 
https://forum.seeedstudio.com/t/portable-arduino-ide/274046

me-no-dev, 2024-06-02:
Portable install could also do. Since this is the first time that we hit this issue with xtensa toolchain (we have already done what we can for RiscV) for future versions we will shorten the package name to allow for longer usernames.

closed 2024-06-04 by me-no-dev