espressif/arduino-esp32 #9750: Controlling RGB LED Builtin before setup() for ESP32S3

HamzaHajeir, 2024-06-02:
### Board

ESP32-S3-DevKitC-1-N8R2

### Device Description

None

### Hardware Configuration

None

### Version

v3.0.0

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

.

### Description

The system crashes when `digitalRead()` is called before `setup()`, the main reason is it being addressable Builtin LEDs.

### Sketch

```cpp
#include <Arduino.h>
struct Init {
  Init() {
    digitalWrite(LED_BUILTIN, LOW);
  }
};
Init i;
void setup() {
}

void loop() {
}
```

### Debug Message

```plain
ELF file SHA256: 2170e19b1060522f

Rebooting...
�
assert failed: xQueueGenericSendFromISR queue.c:1122 (pxQueue)

Backtrace: 0x4037777e:0x3fceae30 0x4037ad19:0x3fceae50 0x4038077d:0x3fceae70 0x4037b222:0x3fceafa0 0x4037d47f:0x3fceafc0 0x4200244b:0x3fceaff0 0x40375a85:0x3fceb020 0x40377d19:0x3fceb060 0x403785cd:0x3fceb080 0x400559dd:0x3fceb140 0x4037d95a:0x3fceb150 0x4037f823:0x3fceb170 0x4037f988:0x3fceb190 0x403779fe:0x3fceb1b0 0x40377a11:0x3fceb1e0 0x4037df94:0x3fceb200 0x4037bb99:0x3fceb220 0x4200e3dd:0x3fceb260 0x4200e437:0x3fceb290 0x4200a829:0x3fceb2b0 0x4200a91e:0x3fceb2e0 0x40376fa3:0x3fceb310 0x403cd852:0x3fceb340 0x403cdafe:0x3fceb380 0x403c98d5:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
  #0  0x4037777e in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:466
  #1  0x4037ad19 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
  #2  0x4038077d in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:81
  #3  0x4037b222 in xQueueGenericSendFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1123 (discriminator 2)
  #4  0x4037d47f in xTimerPendFunctionCallFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/timers.c:1119
  #5  0x4200244b in _rmt_tx_done_callback at C:/Users/Hamza/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-rmt.c:100
  #6  0x40375a85 in rmt_isr_handle_tx_done at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt/rmt_tx.c:888     
      (inlined by) rmt_tx_default_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt/rmt_tx.c:1017
  #7  0x40377d19 in shared_intr_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/intr_alloc.c:438    
  #8  0x403785cd in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240
  #9  0x4037d95a in vPortClearInterruptMaskFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:568
      (inlined by) vPortExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:532
  #10 0x4037f823 in multi_heap_internal_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:167 (discriminator 1)
  #11 0x4037f988 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:262  
      (inlined by) multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:240   
  #12 0x403779fe in heap_caps_malloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:176
  #13 0x40377a11 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:197
  #14 0x4037df94 in pvPortMalloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/heap_idf.c:83
  #15 0x4037bb99 in xTaskCreatePinnedToCore at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:946
  #16 0x4200e3dd in init_timer_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:531      
      (inlined by) esp_timer_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:555        
  #17 0x4200e437 in __esp_system_init_fn_esp_timer_startup_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:577
  #18 0x4200a829 in do_system_init_fn at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:228 (discriminator 15)
  #19 0x4200a91e in do_secondary_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:429
      (inlined by) start_cpu0_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:490
  #20 0x40376fa3 in call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/cpu_start.c:706
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2024-06-02:
It can not work, because the driver requires FreeRTOS to be fully booted, but that is not the case before app_main (which creates the task that will run setup and loop).

closed 2024-06-02 by me-no-dev

HamzaHajeir, 2024-06-02:
It was fully working in previous versions (3.0.0-alpha and earlier).

me-no-dev, 2024-06-02:
Are you sure you have tested with the same exact board and code? The error comes from the IDF driver itself and I doubt they added that functionality only recently (I will doublecheck). Because you have an RGB LED, it's not doing `digitalWrite`, it's rather using RMT to send the neopixel color

HamzaHajeir, 2024-06-02:
I've just tried it with core release 2.0.11 with no error. 

Also I can ensure that it was the case for my custom builds, both projects controls the LED on Globals initializations. But behavior seems changed in very recent builds (Haven't made a successful build for [~8 months](https://github.com/HamzaHajeir/arduino-esp32/releases/tag/lwip-tls-3.0.0-rc2)) which I had tested in both projects with.

Using alpha1 produces similar behaviour but just after turning the LED ON: 

```
00:11:01.029 > assert failed: xQueueGenericSendFromISR queue.c:1122 (pxQueue)
00:11:01.034 >
00:11:01.034 >
00:11:01.034 > Backtrace: 0x4037775a:0x3fceae10 0x4037ac41:0x3fceae30 0x40380665:0x3fceae50 0x4037b14a:0x3fceaf80 0x4037d367:0x3fceafa0 0x42001ec7:0x3fceafd0 0x40375a81:0x3fceb000 0x40377cdd:0x3fceb050 0x40378599:0x3fceb070 0x400559dd:0x3fceb130 |<-CORRUPTED
00:11:01.051 >
00:11:01.051 >
00:11:01.051 >
00:11:01.057 > 
00:11:01.057 > ELF file SHA256: 23ce96923314d0fd
00:11:01.057 >
00:11:01.057 > Rebooting...
00:11:01.057 > �ESP-ROM:esp32s3-20210327
00:11:01.068 > Build:Mar 27 2021
00:11:01.068 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
00:11:01.086 > Saved PC:0x403772c1
00:11:01.086 >   #0  0x403772c1 in esp_restart_noos at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158 (discriminator 1)
00:11:01.086 >
00:11:01.086 > SPIWP:0xee
00:11:01.086 > mode:DIO, clock div:1
00:11:01.086 > load:0x3fce3818,len:0x498
00:11:01.086 > load:0x403c9700,len:0x4
00:11:01.086 > load:0x403c9704,len:0xad0
00:11:01.086 > load:0x403cc700,len:0x2a84
00:11:01.097 > entry 0x403c9880
```

VojtechBartoska, 2024-06-05:
Hi @HamzaHajeir, can you please retest on 3.0.1 version? We are not sure where should be a problem. Thanks!

HamzaHajeir, 2024-06-05:
Hi @VojtechBartoska, I've just tested the sketch provided and it yet crashed with this log:

```
ELF file SHA256: ceefe85b54968faf

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403772e5
  #0  0x403772e5 in esp_restart_noos at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880

assert failed: xQueueGenericSendFromISR queue.c:1122 (pxQueue)

Backtrace: 0x4037777e:0x3fceae30 0x4037ad19:0x3fceae50 0x4038077d:0x3fceae70 0x4037b222:0x3fceafa0 0x4037d47f:0x3fceafc0 0x4200245f:0x3fceaff0 0x40375a85:0x3fceb020 0x40377d19:0x3fceb060 0x403785cd:0x3fceb080 0x400559dd:0x3fceb140 0x4037d95a:0x3fceb150 0x4037f823:0x3fceb170 0x4037f988:0x3fceb190 0x403779fe:0x3fceb1b0 0x40377a11:0x3fceb1e0 0x4037df94:0x3fceb200 0x4037bba8:0x3fceb220 0x4200e3f1:0x3fceb260 0x4200e44b:0x3fceb290 0x4200a83d:0x3fceb2b0 0x4200a932:0x3fceb2e0 0x40376fa3:0x3fceb310 0x403cd852:0x3fceb340 0x403cdafe:0x3fceb380 0x403c98d5:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
  #0  0x4037777e in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:466
  #1  0x4037ad19 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
  #2  0x4038077d in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:81
  #3  0x4037b222 in xQueueGenericSendFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1123 (discriminator 2)
  #4  0x4037d47f in xTimerPendFunctionCallFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/timers.c:1119
  #5  0x4200245f in _rmt_tx_done_callback at C:/Users/Hamza/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-rmt.c:100
  #6  0x40375a85 in rmt_isr_handle_tx_done at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt/rmt_tx.c:888
      (inlined by) rmt_tx_default_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rmt/rmt_tx.c:1017
  #7  0x40377d19 in shared_intr_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/intr_alloc.c:438
  #8  0x403785cd in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240
  #9  0x4037d95a in vPortClearInterruptMaskFromISR at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:568
      (inlined by) vPortExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:532
  #10 0x4037f823 in multi_heap_internal_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:167 (discriminator 1)
  #11 0x4037f988 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:262
      (inlined by) multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:240
  #12 0x403779fe in heap_caps_malloc_base at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:176
  #13 0x40377a11 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:197
  #14 0x4037df94 in pvPortMalloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/heap_idf.c:83
  #15 0x4037bba8 in xTaskCreatePinnedToCore at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:951
  #16 0x4200e3f1 in init_timer_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:531
      (inlined by) esp_timer_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:555
  #17 0x4200e44b in __esp_system_init_fn_esp_timer_startup_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer.c:577
  #18 0x4200a83d in do_system_init_fn at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:228 (discriminator 15)
  #19 0x4200a932 in do_secondary_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:429
      (inlined by) start_cpu0_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:490
  #20 0x40376fa3 in call_start_cpu0 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/cpu_start.c:706
```

me-no-dev, 2024-06-05:
@VojtechBartoska there is nothing to fix here. IDF Driver (now) requires FreeRTOS to be started. Nothing wrong with that.
@HamzaHajeir can overwrite `void init(){}` to start it before `setup()` is called (and after FreeRTOS is up)

HamzaHajeir, 2024-06-05:
I've already made a [workaround](https://github.com/HamzaHajeir/h4plugins/commit/db1c9815ac7946406c76b7b5c953d670d720db63#diff-13ca5704f98946ea54b8394bf8741518d0a4d86ddb6f8cf43db43c3561f5d899) for that, `H4::.queueFunction()` just excutes the given funtion on the next "loop iteration".