espressif/arduino-esp32 #9753: Error Compiling Library ESPAsyncWebServer

Zhu-jiatong, 2024-06-02:
### Board

Any

### Device Description

Any

### Hardware Configuration

Any

### Version

v3.0.0

### IDE Name

Arduino IDE with VSCode

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Cannot compile sketch that uses ESPAsyncWebServer. It seems to be an issue with breaking changes to core library.

### Sketch

```cpp
#include <ESPAsyncWebServer.h>
#include <SD.h>

AsyncWebServer server(80);

void setup()
{
	Serial.begin(115200);
}

void loop()
{
}
```

### Debug Message

```plain
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp: In function 'bool getMD5(uint8_t*, uint16_t, char*)':
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?
74 |   mbedtls_md5_starts_ret(&_ctx);
|   ^~~~~~~~~~~~~~~~~~~~~~
|   mbedtls_md5_starts
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:75:3: error: 'mbedtls_md5_update_ret' was not declared in this scope; did you mean 'mbedtls_md5_update'?
75 |   mbedtls_md5_update_ret(&_ctx, data, len);
|   ^~~~~~~~~~~~~~~~~~~~~~
|   mbedtls_md5_update
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:76:3: error: 'mbedtls_md5_finish_ret' was not declared in this scope; did you mean 'mbedtls_md5_finish'?
76 |   mbedtls_md5_finish_ret(&_ctx, _buf);
|   ^~~~~~~~~~~~~~~~~~~~~~
|   mbedtls_md5_finish
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp: In member function 'void AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*)':
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp:188:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
188 |       ets_printf("ERROR: Too many messages queued\n");
|       ^~~~~~~~~~
|       vswprintf
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp: In member function 'void AsyncWebSocketClient::_queueMessage(AsyncWebSocketMessage*)':
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp:551:7: error: 'ets_printf' was not declared in this scope; did you mean 'vswprintf'?
551 |       ets_printf("ERROR: Too many messages queued\n");
|       ^~~~~~~~~~
|       vswprintf
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp: In member function 'IPAddress AsyncWebSocketClient::remoteIP()':
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp:832:28: error: call of overloaded 'IPAddress(unsigned int)' is ambiguous
832 |         return IPAddress(0U);
|                            ^
In file included from C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/Arduino.h:196,
from C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp:21:
C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/IPAddress.h:115:3: note: candidate: 'IPAddress::IPAddress(const ip_addr_t*)'
115 |   IPAddress(const ip_addr_t *addr);
|   ^~~~~~~~~
C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/IPAddress.h:73:3: note: candidate: 'IPAddress::IPAddress(const IPAddress&)'
73 |   IPAddress(const IPAddress &address);
|   ^~~~~~~~~
C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/IPAddress.h:72:3: note: candidate: 'IPAddress::IPAddress(const char*)'
72 |   IPAddress(const char *address);
|   ^~~~~~~~~
C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/IPAddress.h:69:3: note: candidate: 'IPAddress::IPAddress(const uint8_t*)'
69 |   IPAddress(const uint8_t *address);
|   ^~~~~~~~~
C:\Users\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/IPAddress.h:67:3: note: candidate: 'IPAddress::IPAddress(uint32_t)'
67 |   IPAddress(uint32_t address);
|   ^~~~~~~~~
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp: In constructor 'AsyncWebSocketResponse::AsyncWebSocketResponse(const String&, AsyncWebSocket*)':
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp:1262:3: error: 'mbedtls_sha1_starts_ret' was not declared in this scope; did you mean 'mbedtls_sha1_starts'?
1262 |   mbedtls_sha1_starts_ret(&ctx);
|   ^~~~~~~~~~~~~~~~~~~~~~~
|   mbedtls_sha1_starts
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp:1263:3: error: 'mbedtls_sha1_update_ret' was not declared in this scope; did you mean 'mbedtls_sha1_update'?
1263 |   mbedtls_sha1_update_ret(&ctx, (const unsigned char*)key.c_str(), key.length());
|   ^~~~~~~~~~~~~~~~~~~~~~~
|   mbedtls_sha1_update
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp:1264:3: error: 'mbedtls_sha1_finish_ret' was not declared in this scope; did you mean 'mbedtls_sha1_finish'?
1264 |   mbedtls_sha1_finish_ret(&ctx, hash);
|   ^~~~~~~~~~~~~~~~~~~~~~~
|   mbedtls_sha1_finish
Error during build: exit status 1
IntelliSense configuration updated. To manually rebuild your IntelliSense configuration run "Ctrl+Alt+I"
[Error] Verifying sketch 'TestServer.ino': Exit with code=1
```

### Other Steps to Reproduce

Compile the sketch using the 3.0.0 version.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Zhu-jiatong, 2024-06-02:
Same issue: me-no-dev/ESPAsyncWebServer#1410

lbernstone, 2024-06-02:
This is a fairly easy fix, but the issue belongs in the library, not here.

me-no-dev, 2024-06-02:
will fix it these days

ctodor, 2024-06-02:
guys, heads up, ESPAsyncWebServer is bugy and it will do a mem. corruption sooner or later.  There is a big "desing" flaw on that lib.  ESPAsyncWebServer uses AsynTcp lib, onDisconnect callaback delete an AsyncClient pointer allocated by AsyncServer .
my app. crashed random, sometime after few hours, sometime after few days.

lbernstone, 2024-06-02:
Please don't hijack issues.  Open an issue in the appropriate forum.  Post a complete breakdown of the problem, along with code that demonstrates the issue.

ctodor, 2024-06-03:
@lbernstone sorry for my prev. post. it seems the procedure is more important than the issue it self. should I delete my prev. comment?

lbernstone, 2024-06-03:
Your question is not going to be answered here, since this is not the repository for ESPAsyncWebServer. You can interpret that as a focus on process if you like,

ctodor, 2024-06-03:
@lbernstone  sorry, but you said "your question....", about what question are you talking about?

VojtechBartoska, 2024-06-03:
Hello, as mentioned above, this problem need to be solved in ESPAsyncWebServer library. There are no action points in this repo, closing.

Relevant ticket: https://github.com/me-no-dev/ESPAsyncWebServer/issues/1410

closed 2024-06-03 by VojtechBartoska

Innocent546, 2024-06-28:
For anyone having trouble with this now, I made the compiler's suggested changes manually in library files and got it to work. Though this might not be the best solution, it will work.

For example, for this error:
C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp:74:3: error: 'mbedtls_md5_starts_ret' was not declared in this scope; did you mean 'mbedtls_md5_starts'?

1. Open C:\Users\...\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp in a text editor
2. Go to line 74
3. Change error: 'mbedtls_md5_starts_ret' to 'mbedtls_md5_starts'

Do this for all compiler suggestions. 

### Here's all the changes I made. 

Arduino\libraries\ESPAsyncWebServer\src\WebAuthentication.cpp
```
#include <mbedtls/md5.h>  // Add this include at the top

//Changes in this function
bool getMD5(uint8_t* data, uint16_t len, char* output) {
  mbedtls_md5_context _ctx;
  unsigned char _buf[16];

  mbedtls_md5_init(&_ctx);
  mbedtls_md5_starts(&_ctx);  // Change this line
  mbedtls_md5_update(&_ctx, data, len);  // Change this line
  mbedtls_md5_finish(&_ctx, _buf);  // Change this line
  mbedtls_md5_free(&_ctx);

  for (int i = 0; i < 16; i++) {
    sprintf(output + (i * 2), "%02x", _buf[i]);
  }

  return true;
}
```

In all code files where ets_printf() exists.
Use the find tool to locate all ets_printf and change to printf. (Eg: in AsyncWebSocket.cpp and AsyncEventSource.cpp)
```
printf("ERROR: Too many messages queued\n");  // Change this line (Example)
```

rlbrinkman, 2024-07-13:
I made the changes shown by rashmithachamikara above, but there was no change in the error messages generated when I compiled my sketch.  Possible differences in my environment that may have rendered the changes ineffective:
1) My target device is "Al Thinker ESP32-CAM"
2) I am using Arduino IDE 1.8.19, not IDE 2.x
3) I tried both #include <ESPAsyncWebServer.h> and #include "ESPAsyncWebServer.h" but got no change in results

Any suggestions?

This problem has been around for a few months now; is a fix on the horizon?

sumitbasumallick, 2024-08-01:
Thanks @rashmithachamikara. Worked for me for ESP32 project.

assiro, 2024-08-06:
Thanks rashmithachamikara, after the changes suggested the library is working on IDE 2.3.2

kalwinskidawid, 2024-08-14:
Is this already fixed? 
I am trying to build a project with this library and I have the same problem.
In particular, there is a problem when I execute `request->send()`.

josep11, 2024-08-15:
Thanks @rashmithachamikara! It worked like charm for me as well.

Zeca-man, 2024-08-20:
Many thanks to @rashmithachamikara ...Issue solved  !

bartdereu, 2024-09-01:
Thanks, this worked. Why not commit these changes?

xiaohe9527, 2025-01-09:
> For anyone having trouble with this now, I made the compiler's suggested changes manually in library files and got it to work. Though this might not be the best solution, it will work.
>
> For example, [...]

thank u i fixed it