espressif/arduino-esp32 #9756: PSRAM not initialized?

anthonyhoegberg, 2024-06-03:
### Board

ESP32s3

### Device Description

ESP32s3

### Hardware Configuration

FLASH 16MB DIO 80MHZ
OPI PSRAM

### Version

v3.0.0

### IDE Name

ESP-IDF vscode

### Operating System

Windows 11

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

default

### Description

So i cant figure out how to enable psram for arduino components 
My sdkconfig has been setup correctly with all the correct settings, however i feel like arduino components are ignoring some settings?

Like i cant get serial to work or my device unless i add definitions for USB CDC and USB MODE
But i think that should be automatic based on sdkconfig no? 
Also in older versions of arduino component the BOARD_HAS_PSRAM definition made it enable psram for me
However in this version 3.0.0 it no longer works and gives me the error 
project_folder/managed_components/espressif__arduino-esp32/cores/esp32/esp32-hal-psram.c:22:10: fatal error: esp_psram.h: No such file or directory      
   22 | #include "esp_psram.h"

Is arduino components somehow ignoring my sdkconfig and not applying settings correctly since i have to add definitions or is something wrong here? (When i remove the definition for BOARD HAS PSRAM I get no error but sketch shows no psram)

Project root cmake config:
```cmake
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

add_definitions(-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue)
add_definitions(-DARDUINO_USB_CDC_ON_BOOT=1)
add_definitions(-DARDUINO_USB_MODE=1)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(amcamiot3)
```
[sdkconfig](https://github.com/user-attachments/files/15527561/sdkconfig.txt)

### Sketch

```cpp
#include "Arduino.h"

void setup() {
    Serial.begin(115200);
    Serial.println("Serial works!");

    Serial.printf("Total heap: %li\n", ESP.getHeapSize());
    Serial.printf("Free heap: %li\n", ESP.getFreeHeap());
    Serial.printf("Total PSRAM: %li\n", ESP.getPsramSize());
    Serial.printf("Free PSRAM: %li\n", ESP.getFreePsram());
}

void loop() {
}
```

### Debug Message

```plain
if definition is removed for psram, i get this output
Serial works!
Total heap: 401636
Free heap: 368860
Total PSRAM: 0
Free PSRAM: 0
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

anthonyhoegberg, 2024-06-03:
I found out about the error
It looks like the CMakeLists.txt file in arduino-esp32 does not include esp32-psram component at all?
https://github.com/espressif/arduino-esp32/blob/c16a325439fe5a8a3160701d9601634155bb1443/CMakeLists.txt#L290
adding **esp_psram** to the end of that will make PSRAM work with our -DBOARD_HAS_PSRAM definitions

Looks like we need to do some kind of check to see if psram config is enabled and include it in the requires 

Based on https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s3/migration-guides/release-5.x/5.0/system.html#psram
it is required to include esp_psram component i think?

me-no-dev, 2024-06-03:
We do not need to specify this so far on IDF v5.1. I will add it to requires and see if it will work ok on targets without PSRAM

PR #9762 (2024-06-03): feat(idf): Require esp_psram when as component
Fixes: #9756
closed 2024-06-03 by me-no-dev