espressif/arduino-esp32 #9757: ESP32 compatability with windows11

sai-charan1, 2024-06-03:
### Board

ESP 32 Dev Module

### Device Description

ESP 32  C106

### Hardware Configuration

Cable connected to port 

### Version

v3.0.0

### IDE Name

Aurdino IDE

### Operating System

Windows 11

### Flash frequency

Cannot find it

### PSRAM enabled

yes

### Upload speed

115200

### Description

Cannot see the port

### Sketch

```cpp
void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ;
  }

  Serial.println("Hello Arduino!");
}

void loop() {}
```

### Debug Message

```plain
avrdude: ser_open(): can't open device "\\.\COM3": The system cannot find the file specified.

Failed uploading: uploading error: exit status 1
```

### Other Steps to Reproduce

I also tried by downloading [CP210x Windows Drivers with Serial Enumerator]

(https://www.silabs.com/documents/public/software/CP210x_Windows_Drivers_with_Serial_Enumeration.zip)

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

weoiss1998, 2024-06-03:
Are you sure that you have selected the ESP devices in the Arduino IDE?

SuGlider, 2024-06-03:
A @weoiss1998  pointed, the issue seems to be related to selecting ESP32 as board in the Arduino IDE.

>avrdude: ser_open(): ....

avrdude is **Arduino UNO** tool...

SuGlider, 2024-06-06:
Closing this issue as it seems to be an IDE board misconfiguration. Please feel free to open it again, if necessary.

closed 2024-06-06 by SuGlider